arendst/Tasmota #3217: Downtime counter for failure monitoring

Sandominion, 2018-07-14:
Hi.
I would like to use a Sonoff POW R2 for failure monitoring of a drainage pump. The pump is allways voltage supplied (PowerOnState 4) and switches itself on/off dependend on the water level. During rainy periods this might happen a couple of times an hour or during dry periods only once a month. If the pump is working the Sonoff recognize the used energy.
I would like to measure the time without energy flow and alert somehow (e.g. blinking led & buzzer) when the periode without power drain lasts more than double compared to the last power gap (to avoid flooding the cellar due to a broken pump).
I would be happy about a hint what kind of timer or counter could be the best to use in this case.
Kind Regards
Sandominion

Frogmore42, 2018-07-14:
I would do that with NodeRED.  I don't believe timers or rules were designed with that use case in mind.  If you want complex control, you really have to use a tool that enables it.

roguestreak, 2018-07-15:
Seems like an overly complicated method, especially considering the vary large variance in running periods.  What if you just used another sonoff (or other ESP8266 board) attached to a float switch or something that triggered the alarm?   I believe due to the potentially live nature of the inputs on the POW it's not recommended to connect other sensors, but using a different version you could perhaps monitor whatever is supposed to be switching the pump?
Maybe that's an option that could be done onboard; if the float switch has triggered, but the pump is not drawing power?  Might be able to achieve that via rules, don't know 'cause I haven't used onboard rules yet (mainly using NodeRed)

Sandominion, 2018-07-16:
Thanks for your answers, but can't see a reason to use extra Hardware (2nd ESP/Raspberry) when there is a way to do it in Software. (BTW: The fluid level switch is integrateted in the pump and under water.)
I tried the new arithmetic funktions but have some trouble to copy / compare values in Var1 and Var2. Any solution?

rule on System#Boot do Backlog Var1 0; Var2 6 endon on Time#Minute do Add1 1 endon on Energy#Power>10 do Backlog Var2 %Var1%; Mult2 2 endon on Energy#Power<10 do Var1 0 endon on Var1 > %Var2% do ALARM endon

In Wiki there is a hint in chapter Rules: "Currently no if directive or nested rules are supported."
Does it mean there is no solution yet? Is it planed?

ascillato, 2018-07-16:
Hi,

on Var1 > %Var2%

Is not going to work.

Instead do this

On time#minute do backlog add1 1; event check=%var1% endon
On event#check>%var2% do power1 1 endon

Sandominion, 2018-07-17:
Tanks very much for the hint about events. I guss this should solve my problem. I tried to use them for compare & copy values but it seems they never get executed. What did I do wrong?

rule
 On System#Boot do Backlog Var1 0; Var2 6 endon
 On Time#Minute do Backlog Add1 1; event check=%Var1% endon
 On event#check>%Var2% do Power1 1 endon
 On Button1#State>0 do Backlog event setVar2=%Var1% endon
 On event setVar2 do Backlog Var2 %value%; Mult2 2; Var1 0 endon

ascillato, 2018-07-17:
Hi

I put your rule:

`rule On System#Boot do Backlog Var1 0; Var2 6 endon On Time#Minute do Backlog Add1 1; event check=%var1% endon On event#check>%var2% do Power1 1 endon On Button1#State>0 do event setVar2=%var1% endon On event#setVar2 do Backlog Var2 %value%; Mult2 2; Var1 0 endon`

and works for me in Tasmota v6.1.1b. Which Tasmota version are you using?

Try copy paste the rule I posted.

ascillato, 2018-07-17:
The output:

_(I did a var2 2 just to accelerate the test)_

```
00:00:00 Proyecto sonoff Cocina (Topic cocina, Fallback DVES_F7A6DD, GroupTopic sonoffs) Versión 6.1.1b-2_3_0
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como cocina-1757...
00:00:03 WIF: Conectado
00:00:03 HTP: Servidor web activo en cocina-1757 con dirección IP 192.168.1.32
00:00:05 MQT: Intentando conectar...
10:38:17 MQT: Conectado
10:38:17 MQT: tele/cocina/LWT = Online (Grabado)
10:38:17 MQT: cmnd/cocina/POWER = 
10:38:17 MQT: tele/cocina/INFO1 = {"Module":"Generic","Version":"6.1.1b","FallbackTopic":"DVES_F7A6DD","GroupTopic":"sonoffs"}
10:38:17 MQT: tele/cocina/INFO2 = {"WebServerMode":"Admin","Hostname":"cocina-1757","IPAddress":"192.168.1.32"}
10:38:17 MQT: tele/cocina/INFO3 = {"RestartReason":"External System"}
10:38:17 RUL: SYSTEM#BOOT performs "Backlog Var1 0; Var2 6"
10:38:17 MQT: stat/cocina/RESULT = {"Var1":"0"}
10:38:18 MQT: stat/cocina/RESULT = {"Var2":"6"}
10:38:19 MQT: stat/cocina/RESULT = {"POWER1":"OFF"}
10:38:19 MQT: stat/cocina/POWER1 = OFF
10:38:27 MQT: tele/cocina/STATE = {"Time":"2018-07-17T10:38:27","Uptime":"0T00:00:17","Vcc":2.784,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":52,"APMac":"18:D6:C7:80:38:2C"}}
10:39:16 RUL: TIME#MINUTE performs "backlog Backlog Add1 1; event check=0"
10:39:16 MQT: stat/cocina/RESULT = {"Add1":"1.00"}
10:39:16 MQT: stat/cocina/RESULT = {"Event":"Done"}
10:40:00 RUL: TIME#MINUTE performs "backlog Backlog Add1 1; event check=1.00"
10:40:00 MQT: stat/cocina/RESULT = {"Add1":"2.00"}
10:40:00 MQT: stat/cocina/RESULT = {"Event":"Done"}
10:40:50 CMD: var2 2
10:40:50 MQT: stat/cocina/RESULT = {"Var2":"2"}
10:41:00 RUL: TIME#MINUTE performs "backlog Backlog Add1 1; event check=2.00"
10:41:00 MQT: stat/cocina/RESULT = {"Add1":"3.00"}
10:41:00 MQT: stat/cocina/RESULT = {"Event":"Done"}
10:42:00 RUL: TIME#MINUTE performs "backlog Backlog Add1 1; event check=3.00"
10:42:00 MQT: stat/cocina/RESULT = {"Add1":"4.00"}
10:42:00 MQT: stat/cocina/RESULT = {"Event":"Done"}
10:42:00 RUL: EVENT#CHECK>%VAR2% performs "Power1 1"
10:42:00 MQT: stat/cocina/RESULT = {"POWER1":"ON"}
10:42:00 MQT: stat/cocina/POWER1 = ON
10:42:20 RUL: BUTTON1#STATE>0 performs "backlog event setVar2=4.00"
10:42:20 MQT: stat/cocina/RESULT = {"Event":"Done"}
10:42:20 RUL: EVENT#SETVAR2 performs "Backlog Var2 4.00; Mult2 2; Var1 0"
10:42:21 MQT: stat/cocina/RESULT = {"Var2":"4.00"}
10:42:21 MQT: stat/cocina/RESULT = {"Mult2":"8.00"}
10:42:21 MQT: stat/cocina/RESULT = {"Var1":"0"}
```

Sandominion, 2018-07-17:
Cool that it works (on your device). Thanks for trying!!! I have Version 6.1.1-2_3_0 (newest version from Web-Server via OTA 2 or 3 days ago) on my Sonoff Pow R2. Where do I get 6.1.1b from?

ascillato, 2018-07-17:
I just compile last github version.

Please, try to copy and paste the rules I post in your sonoff. I think that I add some # that were missing in your rules.

Sandominion, 2018-07-17:
Can't compile at the moment (because my Laptop run out of battery and I didn't expect to use it that much during this holiday, so that I would have to recharge it ;-). I can only use precompiled files for update. Under OTA-URL (http://sonoff.maddox.co.uk/tasmota/sonoff.bin) still seams to be the Version 6.1.1. Any idea where I can get a pre-compiled newer version?

Sandominion, 2018-07-19:
@ascillato: Now I managed to test the rule with your correction and it works! Thanks a lot!
Do you know how to switch on the LEDs for alarm instead of the relays?
Do you think, other users could be interested in this rule? Is it worth to put it somehow as an example into the WiKi?

ascillato, 2018-07-19:
Hi,

Great to know that it is working for you! Please close this issue. Thanks.

For turning the leds on and off use the command LedPower. Please, read the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main

Thanks :+1:

closed 2018-07-19 by Sandominion