arendst/Tasmota #3219: Buzzer on Sonoff POW2

Sandominion, 2018-07-14:
I want to use a 3.3V buzzer on a GPIO of a Sonoff. Is it easier to use an active buzzer (and configure it e.g. as an extra relay) or an passive buzzer and configure the GPIO as PWM? Why I can't configure any extra GPIO's after selecting Sonoff POW2?
Kind regards
Sandominion

curzon01, 2018-07-14:
The Sonoff Pow ESP is not isolated from 230V~ power so it's very dangerous to connect external devices to a Pow GPIO. That's also the reason why you can't define any GPIO within Tasmota

ascillato, 2018-07-14:
Sorry. Use another device for connecting things. Sonoff Pow and sonoff por R2 are not prepared for that, resulting in a very DANGEROUS situation. You can be electrocuted. All its electronics are connected directly to mains AC, so everything you connect will be also connected directly to AC.

Please, do not use sonoff pow for that. Use another device. 

Thanks

ascillato2, 2018-07-14:
DANGEROUS. Closing

closed 2018-07-14 by ascillato2