arendst/Tasmota #3221: [REQUEST FOR CHANGE] Remove WPS from Tasmota

arendst, 2018-07-14:
The current versions of Tasmota allow for initial wifi configuration using either:
- serial console
- Wifi Smartconfig using a mobile phone and the ESP8266 Smartconfig App
- Wifi Manager providing a local Access Point and a webbrowser
- Wifi Protected Setup (WPS) using the WPS functionality of a router/switch

Following on to issue #4889 (https://github.com/esp8266/Arduino/pull/4889) AND the fact that the latest core stage version allows for 4k more heap when WPS is removed from Tasmota I hereby ask if anyone is desperate to keep WPS as an initial wifi configuration tool available in Tasmota.

Some may argue that they can never have enough configuration tools but others seem to hate WPS by lack of it's functionality on their routers.

Removing WPS saves some code space and in the future saves 4k precious RAM.

So please provide a good reason to keep WPS available within Tasmota.

Rgrds, Theo.

Jason2866, 2018-07-14:
Kick WPS -> 4k more space for features 👍😃

ascillato, 2018-07-14:
I also agree. Wifi manager as default. Thanks :+1:

arendst, 2018-07-14:
Best of both world...

Made it optional in v6.1.0b and default OFF saving 33k code and 1k mem with current core versions. +4k mem starting with core v2.4.2 to be released some day...

Thx.

closed 2018-07-14 by arendst

digiblur, 2018-07-14:
Awesome change!  I don't run WPS on anything due to its security risks.

ascillato, 2018-07-14:
Theo, as a lot of flash space has been freed up with this, can now be set TasmotaMQTT as default also to eliminate the hungs when mqtt got disconnected? I have been using this TasmotaMQTT driver since you released it and works flawlessly without problems.

arendst, 2018-07-14:
Good idea. I'll start tests too.

Jason2866, 2018-07-14:
@ascillato 
I made difference experiences, with TasmottaMQTT i never reached the "endless" uptimes as before.
Not bad at all, but after 30 - 40 days there is a restart (exception). 
I am back, because my wifi and mqtt is very reliable 😃

digiblur, 2018-07-14:
@ascillato been using #define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT  in my user_config since it was added and it has been working great on an array of ESP8266 devices (Sonoff Basic/S20/S31/Dual, NodeMCU, Wemos D1).

Jason2866, 2018-07-14:
@arendst How many space could be freed if wifi config is disabled completly? 
I ask because user who are compiling there own version dont need this at all...
Maybe this whole part could be made optional as WPS.

digiblur, 2018-07-14:
@Jason2866 I compile my own but I do like the WiFi config in the build so I don't have to hard code any SSIDs, password, etc.

Jason2866, 2018-07-14:
@digiblur That is the reason i asked to make it (if possible and save space) configurable.

Jason2866, 2018-07-15:
@arendst THX!! now my v.6.1.1b is only 438.2k