arendst/Tasmota #3222: Can't enter serial config commands in 6.1.0-minimal

gittyhubbyfrankybobby, 2018-07-14:
was running 5.14.0a. did OTA upgrade from http://sonoff.maddox.co.uk/tasmota/sonoff.bin
ESP boot into 6.1.0-minimal with no config.
If I enter "ssid" or "password" commands into the serial interface I get
00:00:51 CMD: ssid
00:00:51 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}

But POWER and STATUS commands work.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues and in the wiki
- [ x] Hardware used :  Wemos D1 v3.1.0
- [ ] Provide the output of command ``status 0`` :
```
00:00:04 CMD: status 0
00:00:04 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:04 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:04","StartupUTC":"","Sleep":0,"BootCount":1,"SaveCount":0,"SaveAddress":"FB000"}}
00:00:04 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.0-minimal","BuildDateTime":"2018-07-06T21:02:12","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:04 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:00:04 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":398,"Free":604,"Heap":29,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","00082580","0C000002","21021702","00000000"]}}
00:00:04 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3212","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"DC:4F:22:1C:8C:8C","Webserver":2,"WifiConfig":3}}
00:00:04 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1C8C8C","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:04 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:04 1970","Local":"Thu Jan 01 00:00:04 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}
00:00:04 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:00:04 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:04"}}
00:00:04 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:04","Uptime":"0T00:00:04","Vcc":2.821,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}

```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-07-14:
Sonoff-minimal does NOT store any config change so trying to configure SSID and/or Password would not work anyway.

Apparently the sonoff-minimal firmware image for v6.1.0 fails to connect to wifi which in turns defeats the OTA process.

I've just released v6.1.1 solving the sonoff-minimal wifi problem.

Within a few minutes the new firmware images should be available allowing OTA to proceed again.

gittyhubbyfrankybobby, 2018-07-14:
Well. I now get to 6.1.1 minimal, but It's not connecting to my wifi, and after 1 round of failures it boot loops. Doesn't boot after reset button either.

`
17:45:32 MQT: stat/ACMon1/RESULT = {"OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin"}
17:45:32 MQT: stat/ACMon1/RESULT = {"Upgrade":"Version 5.14.0a from http://sonoff.maddox.co.uk/tasmota/sonoff.bin"}
17:45:35 HTP: Web server stopped
17:45:36 HTP: Web server active on ACMon1.local with IP address 192.168.99.215
17:45:38 HTP: Web server stopped
17:46:12 HTP: Web server active on ACMon1.local with IP address 192.168.99.215
17:46:13 MQT: Attempting connection...
17:46:13 MQT: Connected
17:46:13 MQT: tele/ACMon1/LWT = Online (retained)
17:46:13 MQT: cmnd/ACMon1/POWER = 
17:46:14 MQT: stat/ACMon1/UPGRADE = Successful. Restarting
17:46:15 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
@cp:0
ld
system param error
<bunch of hex data>

00:00:00 CFG: Use defaults
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_1C8C8C, GroupTopic sonoffs) Version 6.1.1-minimal-2_3_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-3212...
00:00:22 WIF: Connect failed with AP timeout
00:00:22 WIF: Connecting to AP1  in mode 11N as sonoff-3212...
00:00:44 WIF: Connect failed with AP timeout
00:00:44 WIF: WPSConfig active for 3 minutes
00:01:03 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff27e4
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff27e4
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff27e4
~ld
{l

`

PerfectCrime, 2018-07-16:
Same here, Tasmota version 6.1.1, Sonoff Basic.
Does not connect to WiFi and reboots after every try.
Using Arduino 2.4.1 / 2.2.1

I run ping -t to the device. Led starts to flash quickly, halts for a second and than flashes twice. Ping than gets 1 packet replied to and sequence starts from the beginning.

ascillato, 2018-07-16:
try using esp8266 core lib v2.3.0 instead of 2.4.1 that is unstable

PerfectCrime, 2018-07-17:
Thank you @ascillato, that did the trick.

closed 2018-07-21 by ascillato2