arendst/Tasmota #3223: Buzzer on Sonoff TH

Sandominion, 2018-07-14:
I want to use a 3.3V buzzer on a GPIO of a Sonoff. Is it easier to use an active buzzer (and configure it e.g. as an extra relay) or an passive buzzer and configure the GPIO as PWM?
Kind regards
Sandominion

arendst, 2018-07-14:
Just do it and let us know if it works.

ascillato, 2018-07-14:
I think that using a passive will not work without extra hardware (transistors, etc)
Better should be an active one.

Sandominion, 2018-07-15:
So I ordered active ones and will test. Any suggestion which GPIO and related commands to use?

ascillato2, 2018-07-16:
Any gpio different than 0, tx and rx. Configure as a relay. Commands. Power on, power off. Check if the current and voltage required by your buzzer are compatible with the esp8266.

ascillato, 2018-07-23:
Depending on the resistance of the buzzer, sometimes it prevent the ESP8266 of booting up. But if it is working now for you, leave it like that because seems that your buzzer is not interfering.

Good to know. Thanks a lot for sharing :+1:

Sandominion, 2018-07-23:
Got the active 3.3V-buzzer, connected it to GND & GPIO3=RX
and configured the GPIO1 in the settings to "18 Relay 2". Works fine!
But now I read again your hint to use any other GPIO than 0, TX or RX.
What's the  disadvantage about using RX?

Sandominion, 2018-07-24:
I have booting problems as well when I try to use GPIO1=TX for the buzzer.
But my new problem is: Together with PowerOnState 4 (Turn relay(s) on and disable further relay control) it seems that it is not possible to configure the GPIO as Relay2 and control it via commands because its always on.
Which other GPIO-configuration can be used for the buzzer and easily controlled (e.g. LED2 or LED2i)? What command should be used?

ascillato, 2018-07-24:
Yes, leds should work. Commands in the wiki.

Sandominion, 2018-07-24:
But how? In the Wiki I can find only the one Command "LedState on/off/..." which seems to influence only the 1st LED. What is the command e.g. for LED2 to switch on/off?

Sandominion, 2018-07-24:
Unfortunately I can't use LED1 because it seems impossible to switch it totally off (LedState "off" only does: Disable use of LED as much as possible). That's not enogh when the buzzer is doing alarm on each WiFi message!
The PowerOnState 4 looks for me the only way to prevent an accidently switching relay off (needs to be allways on). Any other suggestion?

ascillato, 2018-07-24:
Hi,

Too much restrictions in your wanted configuration. I don't understand your idea. So, now, I'm not sure what is what you are aiming.

You say:

1 - Sonoff TH (so you have 1 relay and 1 temperature monitoring)
2 - Your output is all the time ON and you don't want control over it.
3 - You have a buzzer and you have boot problems connecting it to TX or RX
4 - You want to control that buzzer

So,

1, 2 - If you do not have control, so, why not connecting that directly to mains instead and leave the relay free?
3 - See restrictions (https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs)
4 - See available GPIOs (https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations) and set any of those pins as relay to control it

So, what is what you are trying to control or set ?

Sandominion, 2018-07-31:
Item 3 is solved. There are no boot problems or any other restrictions when connecting the buzzer to GND & GPIO3=RX on the Sonoff TH. Only for information I added the boot problems I realized when buzzer is connected to to GND & GPIO1=TX, but that’s no problem for me.

Item 4 is no problem either. I just choose GPIO’s which can easily accessed like RX on Sonoff TH (and GPIO4 on Sonoff Pow R2). My concern is more about configuring. When I assign it as an relay it is under influenced by PowerOnState. When I assign it as an LED it is influenced by minimal LED signals like activity on WiFi.

Item 1-2 is a good solution for Sonoff TH. On the other hand I still have in mind to transfer this functionality to a Sonoff POW. There I have to use the always-on relay in order to measure the power. But you mentioned already in the issue #3219 that I will not get any help to connect anything on a Sonoff Pow… even though it works fine on my module (just needed a small SW-change in sonoff_template.h to configure under Sonoff_POW the GPIO04 as “GPIO_USER”. To meet your concerns I let the buzzer inside the Box of Sonoff POW – so no risc in my opinion and still loud enough).
Looks like I fail just in front of my aim… maybe I need to switch to another SW like ESPEasy.
Anyway thanks for your help so far!