arendst/Tasmota #3224: Wifi problems

mcc05, 2018-07-14:
Just upgraded a 5.14 version to the new 6.1.1 20180714.

Same issue with Wifi, not connecting, or after 4 presses smart manager not advertising an AP to connect to.

R4C3R, 2018-07-14:
I have the same issue. First I tried installing 6.1.1. via Arduino IDE but the Sonoff Basic doesn't connect to the Wifi.

Installing 5.14 worked like a charm.

trying OTA from 5.14 to 6.1.1. failed, the Sonoff Basic doesn't connect to the Wifi after the OTA upgrade.

ascillato, 2018-07-14:
Hi, I have just tried last firmware and works fine for me, but I was upgrading from 6.0.0b, on a Sonoff Basic and others.

If you need the new features of Tasmota and you want to upgrade, you can try erase all flash using `esptool.py --port COM5 erase_flash` and then flash the last firmware.

mcc05, 2018-07-14:
@ascillato tried what you suggested erase_flash and then write_flash with new firmware still the same with v 6.1.1. Erased 6.1.1 and back to 5.14 ok again.

ascillato, 2018-07-14:
After you flashed 6.1.1 did you connect to the AP? or did you set your credentials and compile by yourself?

ezar, 2018-07-14:
Same here! Sonoff Basic using 6.1.1 doesn't have wifi :(

ascillato, 2018-07-14:
Hi,

I could not reproduce that problem. Sorry.

I erase the flash as you did and then I upload the firmware from github to a Sonoff Basic and works.

This is the serial output:

```
> pio device monitor --port COM31 --baud 115200
--- Miniterm on COM31  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
{$␀l��|␀�d�|␃␌␄␌�␌l�␌#|��␃�␓�s�b�␌c��'o�$g'���␄c␜p��dsdsd8�g�␐␃␄␄�␄l␄Ĝ␄␌␄c␌n�|␃$�␄␌�c��o'�␀l��d`␃�␓␛gn␌d`␂␇␃gs�ۓn␌␄#␌�␎d␎;��g␄␌c␄�␏l
�sĜ��␜�ćd`␂��'�␂�
00:00:00 Proyecto sonoff Sonoff (Topic sonoff, Fallback DVES_F7A6DD, GroupTopic sonoffs) Versión 6.1.1a-2_3_0
00:00:00 WIF: Connectando a AP1  en modo 11N como sonoff-1757...
00:00:23 WIF: Falló Conexión, timeout de AP
00:00:23 WIF: Connectando a AP1  en modo 11N como sonoff-1757...
00:00:46 WIF: Falló Conexión, timeout de AP
00:00:46 WIF: WifiManager activo por 3 minutos
00:00:47 HTP: Servidor web activo en sonoff-1757 con dirección IP 192.168.4.1

00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:01:45"}
00:02:29 WIF: Hostname %s-%04d, SSId1 NetWireless, SSId2 NetWireless
00:02:31 APP: Reiniciando

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3360
~ld
�
00:00:00 Proyecto sonoff Sonoff (Topic sonoff, Fallback DVES_F7A6DD, GroupTopic sonoffs) Versión 6.1.1a-2_3_0
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como sonoff-1757...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en sonoff-1757 con dirección IP 192.168.1.110
00:00:05 MQT: Intentando conectar...
20:41:40 MQT: Intentando conectar...
```

Please, note the times that Tasmota uses BEFORE it start showing the AP for setup. After power up it took 47 seconds to start showing the AP.
After that, I connect to a wifi called _sonoff-1757_ and I put the SSID and PASSWORD from my wifi network and it connects to it.

Please, try again but waiting these times.

@arendst 
Theo, What about adding a little check in the code that if the SSID1 and SSID2 are equal to nothing (as the actual user_config.h file have), Tasmota did not try to connect to them and instead goes directly to the `WIFI_CONFIG_TOOL`?
Or can be added another key like `EMPTY_SSID_WIFI_TOOL` to choose which `WIFI_CONFIG_TOOL `to use at first use?

```
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"        // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.1.1"    // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"  // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.1.1"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              ""                // [Ssid1] Wifi SSID
#define STA_PASS1              ""                // [Password1] Wifi password
#define STA_SSID2              ""                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WAIT         // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
                                                 //   *** NOTE: When WPS is disabled by USE_WPS below, WIFI_WPSCONFIG will execute WIFI_MANAGER ***
#define EMPTY_SSID_WIFI_TOOL   WIFI_MANAGER      // [WifiConfig] Default tool if STA_SSID1 and STA_SSID2 are EMPTY
                                                 //   (WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG)
```

arendst, 2018-07-14:
Will investigate the empty ssids. Good idea.

mcc05, 2018-07-14:
@ascillato @arendst, All I've followed the process again from scratch.
erase_flash and then write_flash with new firmware  6.1.1.

Its states in the log that the sonoff has gone into WifiManager mode, but its not broadcasting an AP, I have scanned for it using two devices but its just not there.

This is the comport trace:
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_7C8F2A, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-3882...
00:00:20 WIF: Connect failed with AP timeout
00:00:20 WIF: Connecting to AP1  in mode 11N as sonoff-3882...
00:00:40 WIF: Connect failed with AP timeout
00:00:40 WIF: WifiManager active for 3 minutes
**00:00:41 HTP: Web server active on sonoff-3882 with IP address 192.168.4.1**
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:03:43 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

R4C3R, 2018-07-14:
I use the SSID + PW in the user_config.h The sonoff basic won't connect to the wifi (5.14 works). and get the following output on the serial monitor.

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_812B50, GroupTopic sonoffs) Version 6.1.1-2_4_1
00:00:00 WIF: Connecting to AP1 my_ap_thuis in mode 11N as sonoff-2896...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP1 my_ap_thuis in mode 11N as sonoff-2896...
00:00:18 WIF: Connect failed as AP cannot be reached
00:00:18 WIF: WPSConfig active for 3 minutes
00:00:40 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

ethan4, 2018-07-15:
Same issue here, updated 5 devices (3x Sonoff Basic and 2x Magichome type controllers) and one of the Sonoff Basics failed to connect when coming from the update, but did connect under 6.1.1 minimal.

Rolled back to 5.1.4 for now and working correctly again.

Mikeinnc, 2018-07-15:
I'm having the same problems as seen above. Two new 'Basic' devices - flashed initially with version 6.1.0 and a whole environment of pain and frustration!! Wifi is intermittent - at best; can't login to 'internal' network (using wifi_manager as wifi_config_tool in user_config.h); one device will sometimes connect to router and then won't....the other just will not connect...and it goes on and on. Doesn't appear at all stable. Have other devices that are rock-solid (and fantastic!) with v5.14.0 and so I have reverted to that. And guess what? Yep - rock solid on ALL devices. Seems like there's a problem........?

pascalsaul, 2018-07-15:
I've upgraded a Basic from 5.10.e directly to 6.1.1a (2.3.0):
`stat/basic_kerstboom/STATUS2 {"StatusFWR":{"Version":"6.1.1a","BuildDateTime":"2018-07-15T11:36:45","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}} => Webserver works
stat/basic_kerstboom/STATUS2 {"StatusFWR":{"Version":"6.1.1a","BuildDateTime":"2018-07-15T11:40:08","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}} => Webserver gone
stat/basic_kerstboom/STATUS2 {"StatusFWR":{"Version":"6.1.1a","BuildDateTime":"2018-07-15T11:44:13","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}} => Webserver gone
`

No WiFi problems and MQTT is working but only Webserver issues...

arendst, 2018-07-15:
Considering the change in power supply design on the Sonoff Basic R2 I get the feeling the new design cannot deliver enough power to keep the ESP8266 running more powerful with Tasmota than the default firmware. 

As newer versions of Tasmota are using more resources of the ESP8266 (rules, timers, sensors checked more often) the ESP8266 will draw more current. Other people already noticed lost sensors over time which might be caused by overloaded power supply too.

This might explain the differnet results people encounter using the latest firmware.

Sealight10, 2018-07-15:
I update to tasmota 6.1.1 from the 5.14 version in my Wemos D1 Mini and no issues this time. Good job!!!

mcc05, 2018-07-15:
All mine sonoffs are th-v1.1, just appears no wifi manager broadcast with full firmware, seems to connect on minimal. 5.14 rock solid

JF62, 2018-07-15:
If I use 6.1.1.b and compile with my own credentials and use #define WIFI_CONFIG_TOOL with the value WIFI_RETRY do I need to use one of them?:
// -- Wifi Config tools ---------------------------
//#define USE_WPS                                  
//#define USE_SMARTCONFIG

I won't loos the connection to my WLAN ;-)

My compiled version of 6.0.0a is 487k - 6.1.1b is now 416k  Does this look reliable?

Jason2866, 2018-07-15:
@JF62 Yes, this looks correct disabling WPS and Wifi Manager saves a lot of space :-))
Mine droped from 512.9k v.6.1.0a  to 439.1k v.6.1.1b !! -> 73,8k
[image]

pascalsaul, 2018-07-15:
Impressive decrease of firmware size!

`-rw-r--r--  1 root root 427680 Jul 15 11:38 sonoff_6.1.1a_2.3.0.bin
-rw-r--r--  1 root root 452560 Jul 15 11:41 sonoff_6.1.1a_2.4.0.bin
-rw-r--r--  1 root root 454224 Jul 15 11:45 sonoff_6.1.1a_2.4.1.bin
-rw-r--r--  1 root root 405088 Jul 15 21:12 sonoff_6.1.1b_2.3.0.bin
-rw-r--r--  1 root root 430064 Jul 15 21:15 sonoff_6.1.1b_2.4.0.bin
-rw-r--r--  1 root root 431696 Jul 15 21:19 sonoff_6.1.1b_2.4.1.bin`

JF62, 2018-07-15:
I still use 2.3.0 Do you think 2.4.0 or .1 is now ready for daily use?

Jason2866, 2018-07-15:
I am on 2.3.0 and will stay 2 reasons
- need Alexa -> HueEmulation 
- 2.4.x wifi never worked reliable for me 
- and nice to have -> smallest firmware size

pascalsaul, 2018-07-15:
I agree with Jason2866 at the moment.. I had WiFi/Webserver problems with 5.x and switched to 2.4.1. 

With the latest 6.x release I need to switch back to 2.3.0 where the Webserver is stable...

mike2nl, 2018-07-16:
I had wifi issues too and had posted one thing i had tested in an other issue posting.
Link: [https://github.com/arendst/Sonoff-Tasmota/issues/3150#issuecomment-405021805](url)
Possible a hint about issues? I don't know i can say only that i works like a charm with
Arduion IDE: 1.8.5
Firmware: Sonoff-Tasmota-6.1.0 latest
ARDUINO ESP8266 Release: v2.4.1

Mikeinnc, 2018-07-16:
A couple more comments as this appears to be an ongoing problem. Both my Sonoffs Basic are marked TH_V1.1 with a date of 2017-5-5. There's also a code - 1744 - etched 'sideways'. I don't know if this helps? When flashed with v6.1.0, as I commented, wifi was flaky on one device and totally 'non-available' on the other. I did wonder if the problem was a lack of power, as Theo suggested. In both cases, I was powering the board via a USB-FDTI adapter. To make sure that wasn't the issue, I also tried powering from a 3v3 adapter via a 12v 2A 'power brick'. It was the same. I _didn't_ try powering via mains - it wasn't something I wanted to do at that time, nor did I think it would make a difference. Since reverting to 5.14.0, both devices connected to my network **immediately** (I flash a fixed IP address in the code). I've had them powered up for the best part of 24 hours now - just as I have found with other devices now incorporated in my home network, they are absolutely rock-solid. I have used Arduino IDE; ESP8266 v2.4.1 and carefully followed Theo's assistance on the wiki entry. I see there is a newer version (v6.6.1b) and as I have one more device to 'fix', I will try this and comment on it. Thanks to everyone for their help!

mcc05, 2018-07-16:
@arendst, I got the same results as @pascalsaul, with 6.1.1b, and 2.4.1, flashed ok, advertised an AP, but would not take wifi settings, webserver very slow to respond and timeouts and i failed to get it to connect to wifi. 

Re-built using 2.3.0 and its seems to work great, connected to wifi and webserver is rapid and responsive.

Mikeinnc, 2018-07-17:
Reading the other comments has led me to check the combinations I have on my currently installed and working Sonoff devices. So I have a Basic with v5.14.0a using core 2.3.0 / sdk 1.5.3 and - more interesting! - a S20 'Smart WiFi Socket' that I upgraded to v6.1.0 with core 2.3.0 / sdk 1.5.3. And it works - perfectly! So it does look like the v6.x.x and core 2.4.1 combination is the fundamental problem? Does this help?

motaz9k, 2018-07-17:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_974834, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-2100...
00:00:14 WIF: Connect failed with AP timeout
00:00:14 WIF: Connecting to AP1  in mode 11N as sonoff-2100...
00:00:15 RSL: stat/sonoff/RESULT = {"WifiConfig":"WifiManager selected"}
00:00:15 WIF: WifiManager active for 3 minutes
00:00:16 HTP: Web server active on sonoff-2100 with IP address 192.168.4.1

Exception (0):
epc1=0x40215757 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff4b90 end: 3fff4d70 offset: 01a0

>>>stack>>>
3fff4d30:  00000000 0000002c 00000010 40216fc5
3fff4d40:  3fffdad0 00000000 3fff3d35 4022ac03
3fff4d50:  3fffdad0 00000000 3fff3d35 4020c900
3fff4d60:  feefeffe feefeffe 3fff3d50 4020e8b4
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3b38
~ld
�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_974834, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-2100...
00:00:14 WIF: Connect failed with AP timeout
00:00:14 WIF: Connecting to AP1  in mode 11N as sonoff-2100...
00:00:28 WIF: Connect failed with AP timeout
00:00:28 WIF: WifiManager active for 3 minutes
00:00:29 HTP: Web server active on sonoff-2100 with IP address 192.168.4.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3b38
~ld
�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_974834, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-2100...
```

iandronowicz, 2018-07-17:
Same here, but happened to me on 3 out of 4 Sonoff's when upgrading from 5.14 the the latest 6.1.1 (after pivoting succesfully on the minimal version.

The strange thing is that on one device worked ok, it failed on devices that I had set up rules...

iandronowicz, 2018-07-17:
no how can I go back to 5.14 without taking the device off the wall socket 👎

timstanley1985, 2018-07-17:
@iandronowicz to downgrade I published an upgrade message on mqtt with a retain flag and placed a 5.14 firmware on my ota server. It took about half hour of WiFi reconnections but it eventually held on long enough to download the whole firmware and downgrade

jumblies, 2018-07-17:
Just reporting the same things as other.  WiFi manager won't allow connections and on my android tablet, it will try to connect to the SSID but never gets an IP address assigned.  Lots of exception restarts when it is installed and trying to configure.  Can change SSID and PASSWORD over serial but it still won't connect to network.

iandronowicz, 2018-07-17:
Too late for me! I had to take out 3 sonoff basic from within the walls to re flash them!! Never again will upgrade to a new firmware version so soon...

pReya, 2018-07-19:
Same problem here (https://github.com/arendst/Sonoff-Tasmota/issues/3056). This has been going on for a while now, and seems to affect many users with Sonoff Basic modules. I'd love to get to the bottom of this.

Mikeinnc, 2018-07-21:
It's a bit glib to state that 'it seems that the issue has been solved'. Where? With what? Perhaps you'd like to share what the solution is for the many people with this problem before arbitrarily closing the problem? Thanks.

meingraham, 2018-07-21:
@Mikeinnc (& @maciboy @sreknob, et. al.),

I think Adrian closed it because the solution (workaround?) is to use the 2.3.0 core (mentioned by several in this track and several other tracks on this forum and in the Google Groups SonoffUsers group). I know that reverting to 2.3.0 back from 2.4.1 worked for me.

Perhaps @ascillato2  should have summarized that this seems to be the general approach when he closed the track to give people that feedback. Otherwise those that don't read the posts on the thread might jump to the conclusion you guys reached. I know that @ascillato2 is just trying to moderate this forum to keep it as cleaned up and up to date as possible.

Having said that, if your experience is that reverting to 2.3.0 did not work for you, that would be good information for others to have at their disposal. Otherwise, until a new, stable, core (e.g., 2.4.2) is available, using 2.3.0 seems like the only option. I did see that Theo is already preparing for 2.4.2 based on entries in the release notes/changelog.

So, as soon as that eventuality (2.4.2) occurs, it should relieve your frustration about using this **fantastic** **free** product which brings along with it support provided by a **community** of hackers... all trying to do their best.

Regards.

Mike

jumblies, 2018-07-21:
@Mikeinnc 
I think your explanation is correct and the issue cannot be replicated because there seems to be diversity of Sonoff Basic Hardware that gives rise to the issue itself.  I'm not really clear how that is, but it also isn't important at this point.  

The thing that is important is the precompiled binaries are pseudo-bricking Sonoff Basic units and there is a widely held belief that you can bury these in your walls and never have to worry about it again.  The OTA facitlity in Tasmota is pulling these precompiled binaries to cause the problems.  

There are already a lot of versions of precompiled bins and I'd hate to add one more called *flaky sonoff basic.bin*

As with any and all open source projects, it is incumbent on the user to test and know what they are installing.  

I think the big rub is that the sonoff basic units are probably the highest prevalence (citation needed-nod to XKCD) and Tasmota has been mostly trouble free up until 6.1.0.  It is still a phenomenal piece of software stuffed into a tiny package on a cheap device and I am grateful for the enormous expertise poured into it by @arendst and crew.  I'm not a very good programmer and I'd estimate I spent about 8 hours building and debugging my Sonar and Garage Door Relays when I could have thrown Tasmota on them and been up in 15 minutes.

pascalsaul, 2018-07-21:
It's not only the basic but basically all Sonoff hardware as far as I know ;) I've here the multiple basic, th, pows, b1, touches, dual and 4ch with and without extra sensors and external connected switches :)

jumblies, 2018-07-21:
> @jumblies <https://github.com/jumblies> yes, but I start testing a new
> firmware first at a basic ;) And if it doesn't work good enough I try it at
> a spare touch or pow to replicate the issue but don't post these results
> online since it doesn't make sense. They all use more or less the same ESP
> model and share the same firmware.

Understood.  I almost feel that your approach should be canonized in the
wiki.  It might eliminate some of the angst this problem has caused.  Maybe
on the upgrade path page?

ascillato, 2018-07-21:
Hi,

Please, if anyone think that this issue should be reopen, just ask. No problem at all.

This issue was closed because **mcc05** (the owner of this thread) had explained:

> Re-built using 2.3.0 and its seems to work great, connected to wifi and webserver is rapid and responsive.

R4C3R, 2018-08-24:
This is just to let you know that the 6.1.1. release from 2018-07-31 solved the issue I had of the webserver not being available on the sonoff devices. I have flashed all my devices and they have performed steadily during the last weeks, no issues at all. A big THANK YOU! to @arendst 

I have installed the software on 
- 5x sonoff basic 
- 2x sonoff TH16
- 1x sonoff RF Bridge 433

Program Version	6.1.1
Build Date & Time	2018-07-31T17:59:09
Core/SDK Version	2_3_0/1.5.3(aec24ac9)

Naqua, 2018-08-24:
Tha't's awkward, that is the version i still have these wifi-problems with.

pReya, 2018-08-24:
I can confirm that erasing the flash completely worked for me.

Grhosso, 2018-09-12:
Hi there,

Anyone can help me?

I have been updating my 10 Sonoff basic via OTA, 9 of them perefectly. Now I have 6.2.1. But 1 of them started to give me problem, I did not get to upload it, always was on 5.12.0. I tried to do it uploading the file and after click on update, I leave it some second, and then I restarted, but now is a brick. No power up, no reset, NOTHING.

How can I wake up the device? Is dead or there is a way to wake it up? I ahve been reading some answer and tryinh with the short push on the black button without sucess. 

Anyone can help me?  Or maybe I have to buy another device? :/
thank youu

lalo-uy, 2018-09-12:
You can always flash it via cable.

Grhosso, 2018-09-12:
Sorry I tried to edit the message and I deleted by mistake.

Yes, I have upload with cable, and I have recovered again.. Thank you!!!

betyar, 2018-09-13:
I have the same WiFi problem as everyone else using an already compiled 6.1.1 on a Basic RF. However I had absolutely no problems on a POW, TH16, and Dual.

ascillato, 2018-09-13:
@betyar 

Hi,

This is an old closed issue. Please see the issue, troubleshooting and solution in https://github.com/arendst/Sonoff-Tasmota/issues/3678

Thanks