arendst/Tasmota #3225: firmware upgrade sonoff B1

coolbombom, 2018-07-14:
Hi guys

I was upgrading all my sonoff units and everything was running smoothly until i came to my B1 bulbs. I had 5.12.0 on my units already, and i wanted to update first to 5.14.0 and then 6.1.0 as suggested on the release page. however i could not just upgrade my other units by uploading the sonoff.bin file first, because they where giving me an error if i tried to do that. So i first had to upload the sonoff-minimal.bin and then the sonoff.bin afterwards to get everything to update. I also uploaded the sonoff-minimal.bin to my first B1 bulb and after reboot it never connected to the network again. :-( ... i thought that it might had been a one time problem and tried my other B1 bulb, but the exact same thing happened. Now non of my sonoff B1 bulbs are visible on my network :-(

I just wanted to let you guys know about this issue and i was wondering if there is anyway of reviving the units without getting out the soldering iron? remember i can't see the units on the network, not even if i try to ping there IP addresses :-(

kind regards
c_bb

ascillato, 2018-07-14:
Sorry,

If you can not see them on your network you will need to flash by cable. Sorry. Please, remember to erase all flash before uploading to start fresh and avoid further problems (erase can be done with `esptool.py --port COM31 erase_flash`)

mike2nl, 2018-07-16:
BTW: 
- i have a changed image for the interface pins.
- interested to add it to the wiki?

[image]

ascillato, 2018-07-16:
Yes, add it please. Thanks :+1:

ascillato2, 2018-07-21:
If you need more help, please, reopen this issue. Thanks

closed 2018-07-21 by ascillato2