arendst/Tasmota #3230: Initiate (re)discover after reboot Home Assistant

pascalsaul, 2018-07-15:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
When you restart HA all auto discovered devices are gone. You need to reboot every single device to get it added again to HA.

**Describe the solution you'd like**
An option to trigger the auto discovery for every device which has Option19 enabled.
For the power state you can use "cmnd/sonoffs/powerX" by HA start-up.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Trigger a restart of every HA auto discovered device on start-up.

ascillato, 2018-07-16:
Hi,

You can configure your device as explained in the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#power-state-on-start-up-without-retain-option

pascalsaul, 2018-07-16:
You're not understanding the probleem and quoting my own text which is an example.

ascillato, 2018-07-16:
Hi, 

I understood your issue. You want Tasmota to say to Home Assistant all its config in the case your sonoff is up before your HA.

I gave you a guide of the wiki, where you can configure your devices and also retrieve Tasmota status when HA boot up. This configuration will work everytime. Discover sometimes did not work in Home assistant.

So. That was a more reliable solution.

If you don't like that, you can make an automation in HA to trigger a rule to the grouptopic that all your Tasmotas have, to restart all your sonoffs.

Hope this helps. 

(I did not understood what you said about a quote? I did not quote anything.)

pascalsaul, 2018-07-16:
You just don't want to reboot all the devices since that's a poor mans solution. We need the option to trigger the auto discover again on request when the device is still online and HA is rebooted or upgraded.

ascillato, 2018-07-16:
Have you tried the following command?

`Backlog setoption19 0; setoption19 1`

pascalsaul, 2018-07-16:
Than the device will reboot.. Backlog triggers a reboot..

arendst, 2018-07-16:
See #1630 and #1849 

Use command state to get HASS in sync again.

ascillato, 2018-07-16:
> setoption19 1 alone triggers the configuration allready... I'll test it tonight.

The command `Backlog setoption19 0; setoption19 1` will make the autodiscover to autodiscover again. Tested and works!

> Only sending this command to a group topic is dangerous. You want only trigger it when auto discovery is enabled on a device

So, change the grouptopic to all the devices you want to use this autodiscover

> I guess Theo need to built in this option to trigger it only when the device is in auto HA mode.

How Tasmota can know that your HA is rebooted? You know that you can reboot HA without rebooting your MQTT broker. And also, your MQTT Broker can be rebooted (or disconnected) but your HA still up.
So, Tasmota MUST not re-send the autodiscover if Tasmota reconnects to the MQTT Broker. May be the case that your HA stills up and you will have 2 entries (or more) with the same device.

So, summerizing. If your HA restarts, your HA need to ask Tasmota again all the configuration.

So, as the command `Backlog setoption19 0; setoption19 1` works, you will need to make an automation at HA startup to call to a rule in your Tasmotas to execute this command.

As this HA autodiscover configuration is important and after you make it work, please, add it to the wiki. Thanks a lot :+1:

pascalsaul, 2018-07-16:
What I did:
stopped HA
restarted MQTT
updated HA

Discovered items are not back after triggering "state" and "powerX" on HA start-up...

mihalski, 2018-07-17:
I'm not sure whether this is related but I thought I would mention that there is a zigbee2mqtt bridge with support for Home Assistant that has dual way communication when either the bridge or Home Assistant is restarted. Relevant info at:

https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant

Similarly the bridge also notifies via topic zigbee2mqtt/bridge/state that it is "online".

Hope this helps.

ascillato, 2018-07-17:
Hi,

@pascalsaul 

After Stopping and restarting HA, have you done in the Tasmota console the command: `Backlog setoption19 0; setoption19 1` ?

arendst, 2018-07-17:
@pascalsaul I'm not a Hass guru but what I thought how it works is:

- Hass needs to receive a MQTT message containing device information for a new device
- Tasmota has the option to send a RETAINED mqtt discover message when setoption19 1 is executed
- This discover message will configure Hass and make the device visible
- A Hass reboot looses any config as there is no persistence within Hass
- A Hass reboot will always receive RETAINED messages from the MQTT server and will re-configure devices

So in this scenario Hass still should have the once discovered sensors.

Just rebooted my HassIO v0.73.2 many times any never lost a device...

I do not understand the problem.

pascalsaul, 2018-07-17:
I also rebooted the MQTT broker also so al information is lost in my case since it isn't persistent ;) Maybe my topic name was a bit confusing, sorry :(

I can just trigger "SetOption19 1" only to a device and then it's added offline to HASS again, without reboot. The power doesn't get updated automatically after adding through Option19. When I reboot HASS it does get updated because of the on start-up get triggered.

arendst, 2018-07-17:
So SetOption19 1 solves the lost device.

Power should be synced using the state command. I guess you have to tell Hass to read the state result.

After that I guess your problem is solved. Just make sure to execute SetOption19 1 and state when you restart your mqtt server. This can also be done using the grouptopic so you only need one mqtt command line like:
``cmnd/grouptopic/backlog setoption19 1; state``

pascalsaul, 2018-07-17:
Wow! I enabled the SetOption 19 on my "sonoffs" group topic.  All my Sonoffs are using AD now. I fixed all automation/scene/friendly names to correspond again. I managed to get everything online again (with sensors). Then I only do: payload: "SetOption19 1;State" on start-up but then there is a race condition. All devices are back but power status is not. When I manually trigger a "state" everything gets updated accordingly. 

p.s. Since I enabled HA AD on all devices I've an extra and stupid BTN for my softswitch in HA :(

ascillato, 2018-07-19:
Great!

Can you add this information to the wiki at the Home Assistant?

If your issue is solved, please close it. Thanks :+1:

closed 2018-07-21 by ascillato2

pascalsaul, 2018-07-21:
Nah, I wasn't fully done with all the tests yet and fixing minor issues...

Nephiel, 2019-10-28:
This issue bit me today.

For future reference, my workaround was an automation on Home Assistant start, that publishes `backlog = "SetOption19 1; Delay 100; State"` to the topic **`sonoffs/cmnd/`**. The 10 second delay should help against the race conditions.

Note that with autodiscovery enabled, Tasmota switches the order of prefix and topic, and **that goes for the group topic as well**. Sonoffs with HA autodiscovery disabled (Option19 0) listen for group commands at **`cmnd/sonoffs`** so this should not affect them.

gorootde, 2020-05-29:
@Nephiel Could you please share the automation yaml? Somehow my tasmotas are not responding to my inputs.

Nephiel, 2020-05-29:
> @Nephiel Could you please share the automation yaml? Somehow my tasmotas are not responding to my inputs.

See _Sync Power State_ example here: https://tasmota.github.io/docs/Home-Assistant/#useful-automations 

¿Are you using `sonoffs` as a group command topic? If so, that may be the problem. It was changed to `tasmotas` on newer versions.