arendst/Tasmota #3231: 1 Light 2 Switches

timota, 2018-07-15:
Hi All, 

Can someone explaine/show how the GroupTopic works or following can be achived with GroupTopic. 

I have 2 way multi-switches in my house. Similar diagram http://azoudange.info/preview/premium-1-light-2-switches-wiring-diagram

My idea to change standart switches on Sonoff T1 UK. As this is not possible to build similar electrical schema with T1, so my idea to use GroupTopic to group some relays (in 2-3 gang T1) in one GroupTopic.

as example:
1-st floor - 2 gang T1, where 
    - one relay without GroupTopic and control light on 1st floor only
    - second relay with GroupTopic like 2Floor light and can control light on second floor

2-d floor - 2 gang T1, where
   - one relay with GroupTopic like 2Floor light and can control light on second floor
   - second relay can control something else. 

when you switching 2-d relay on 1st floor - it will change status for the relay on 2-d floor, so they can work in pair as one switch.

Main question: is each gang in T1 can use different group topic? Similar thing has been impelented in ESPUrna  - reference https://github.com/xoseperez/espurna/issues/300

Thanks in advance

ascillato, 2018-07-16:
Please, read the wiki at:

- https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#2-execute-any-mqtt-message-when-a-button-is-pressed
and
- https://github.com/arendst/Sonoff-Tasmota/wiki/Tips#topic-grouptopic-and-fallback-topic

Thanks

Frogmore42, 2018-07-16:
The diagram shows 2 three-way switches.  When you have two standard switches (non electronic) that control a light (one or more).  The switches are called three way, probably because they have three terminals on them.  If you have three way switches, it will be challenging replacing them with Sonoffs.  How do you plan on wiring them?  What wires do you have in each of the boxes?

ascillato, 2018-07-16:
Also in Tutorials in the Tasmota wiki, there are some YouTube channels where this connection with a sonoff is very well explained in a video.

timota, 2018-07-17:
Sorry for late response, busy days for me. 

@ascillato - thanks for pointing me in right direction. Rules look promised. 

@Frogmore42 - i planned to wire only one of the gangs of T1 to actual load - light, another in the same "group" will stay not connected and will act as a dummy. It can only send/receive status OFF/ON, so it will control first T1 that connected to actual load. 

> What wires do you have in each of the boxes?
to be honest - a lot of wires inside box

Another question about Rules and T1 - is it possible to send OFF/ON signal from one T1 to another via RF.? Idea is - to omit using WiFi, so sonoffs can communicate even if network/controller is down.

ascillato, 2018-07-17:
> Another question about Rules and T1 - is it possible to send OFF/ON signal from one T1 to another via RF.? Idea is - to omit using WiFi, so sonoffs can communicate even if network/controller is down.

No, Sonoff T1 have just a RF Reception. Can not transmit anything. The device that have both (RF Transmitter and Receiver is the Sonoff Bridge).

ascillato, 2018-07-19:
Hi

Any update? 

Have you managed to solve your issue?

timota, 2018-07-20:
hi, 
not yet. Digging your links. 

Found interesting one
https://www.youtube.com/watch?v=l-CM8JQmges&t=2s

looking current wiring. Will update thread with more info later. 

Thanks

rhavin42, 2018-07-31:
Here's how I did it.
https://community.home-assistant.io/t/putting-a-sonoff-in-your-wall-switch-box/33202/117?u=rhavin42

ascillato2, 2018-08-18:
Hi,

Closing this issue as there is no feedback. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2

Stevehans, 2018-12-29:
Use knx, there's an example in the wiki.

Just for reference as this issue is closed.