arendst/Tasmota #3233: Tunable parameters for white channel(s)

ethan4, 2018-07-16:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
My RGB(Warm)W strip produces strange colours in some parts of the colour spectrum.

**Describe the solution you'd like**
I would like to be able to set the colour temperature and the intensity/output of the white channel so the firmware can automatically compensate for a white channel which is not neutral and/or one which is brighter/darker than the other channels.

**Describe alternatives you've considered**

Controlling the white channel separately, but this is not very user friendly.

I've tried using PWM5 and skipping PWM4 as this seems to be for the 'warm white' in a 5 channel strip however the white channel did not come on at all when setting colours via the console.

**Additional context**
There seems to be a wide variance in colour temperature and output of the white channel on the RGBW strips available from the likes of Aliexpress (which can be useful depending on what you want) and using these with Tasmota (via a Magichome device or variant) produces unexpected results as Channel 4 seems to expect a neutral white.

[Example 1](https://www.aliexpress.com/item/DC12V-LED-strip-5050-RGBW-RGBWW-96LEDs-m-5M-lot-Super-Bright-5050-LED-Flexible-Strip/32833506454.html) has a warm white channel which seems to be about as bright as the three colour channels combined
[Example 2](https://www.aliexpress.com/item/5M-5IN1-LED-Strip-Light-RGB-CCT-RGBW-5050-SMD-Dual-White-Temperature-adjustable-RGB-Led/32847113360.html) has two white channels which appear to be about as bright as the three colour channels combined each
I've also seen strips where the ratio of LEDs RGB to white is not equal so this further complicates things.

**(Please, remember to close the issue when the problem has been addressed)**

renne, 2018-07-17:
I suggest a closed-loop brightness/color temperature control with the [APDS-9960](https://github.com/arendst/Sonoff-Tasmota/wiki/APDS-9960) sensor.

1. Lux -> set APDS-9960 thresholds
2. Kelvin -> RGB -> set APDS-9960 thresholds
3. Listen to soft-IRQ for exceeded thresholds
4. increase/decrease brightness and RGB values

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

ethan4, 2018-08-27:
Hi there,

I have not solved this issue.

ascillato, 2018-08-27:
Ok, reopening...

Could you tried as renne suggested?

ascillato, 2018-08-27:
Ok,

I understand. It is complicated. The routines that are doing that calculation are in the file _xdrv_04_light.ino_ from _line 411_.

In the code there is this comment:

> Color Temperature (https://developers.meethue.com/documentation/core-concepts)

Hope you can dig into this to find where can be implemented your request.

ethan4, 2018-08-29:
Hi @ascillato,

CT doesn't seem to work for me as this is an RGBW strip and I think that's only for CW/WW strips.

```
21:55:25 CMD: CT 200
21:55:25 MQT: stat/sonoff_6/RESULT = {"Command":"Unknown"}
21:55:41 CMD: CT
21:55:41 MQT: stat/sonoff_6/RESULT = {"Command":"Unknown"}
```

Jason2866, 2018-10-01:
Should be solved with https://github.com/arendst/Sonoff-Tasmota/commit/6a1a21f03a1d9fcaab3bfe81162a8eb1b575c778
Please close issue. THX

closed 2018-10-01 by ascillato2