arendst/Tasmota #3234: BlitzWolf BW-SHP2 wiring

WladyX, 2018-07-16:
Hi, 
I saw some wiring schema for blitzwolf sp1 here: https://github.com/arendst/Sonoff-Tasmota/issues/2223
Is it the same for BlitzWolf BW-SHP2?
I want to order one, but I don't find much information about how to connect the pins in order to flash it.
Thank you very much for your support and for your nice firmware!

arendst, 2018-07-16:
Yes as far as I know.

My Blitzwolf BW-SHP2 ordered from Banggood Canada is still waiting for release for transport so I can't confirm yet.

WladyX, 2018-07-16:
Then how do you know it is supported? :))
Thanks, I ordered one also, it will take about 1-2month to get my hands on it, hopefully you will get yours first.

wilfredsmit, 2018-08-10:
Issue can be closed. Answered in wikipage [BlitzWolf SHP2](https://github.com/arendst/Sonoff-Tasmota/wiki/BlitzWoIF-SHP2).

WladyX, 2018-08-10:
cool, thank you very much!

closed 2018-08-10 by WladyX

arendst, 2018-08-13:
Verified too.

jobhh, 2018-10-05:
wilfredsmit made a small typo, wiki page: https://github.com/arendst/Sonoff-Tasmota/wiki/BlitzWolf-SHP2

kovacsm, 2018-11-08:
Hi!

I have ordered a SHP2 from bangood, and it seems it is a new version v2.3, which use a new wifi adapter(?).
And because of this it has a different pins.
As far as i saw, it still uses BL0937 chip, so i think it could run tasmota firmware.
I'm just don't know how to wire it. 3.3V and GND is clear, but don't know the others.
I have attached some pictures, I hope You can help me! :)
[image]
[image]
[image]
[image]

arendst, 2018-11-08:
See #4297 and latest development release.