arendst/Tasmota #3235: Can I get the Status of Switch1 with MQTT

MYLE-01, 2018-07-16:
I have
[image]

Work Great 
think this is right i did a
07:56:34 CMD: switchmode1
07:56:34 MQT: stat/office/RESULT = {"SwitchMode1":0}

How can I get Tasmota to MQTT the Switch Status of switch1 

I can turn ON/OFF with MQTT Cool
I can turn ON/OFF with the Switch on the wall Cool and it sends the MQTT of the relay cool

Would like to Know If i Turn it on/off by the wall I get Switch1 ON/OFF in a MQTT mess but still keep the MQTT option working what I already have 

hope this make cents

arendst, 2018-07-16:
Search wiki. Use switchtopic.

MYLE-01, 2018-07-17:
thanks 
But thats not what I wanted 
BUT reading it more

What I End up Doing is

Setting the TelePeriod to 10  

then in HA

```
  - platform: mqtt  
    name: Heater Switch
    state_topic: "tele/SONOFF7/SENSOR"
    value_template: "{{value_json['Switch1']}}"
```
So know I know the Status of the switch 

Thanks

closed 2018-07-17 by MYLE-01