arendst/Tasmota #3237: Upgrade Issue Sonoff Basic 6.1.1

helmar74, 2018-07-16:
**Describe the bug**

I upgraded successful Sonoff SC, S20, S26, 4CH, 433Bridge, B1 to version 6.1.1
The only Device that failed is a Sonoff Basic. 
The OTA failed, the device wasn't responding.
Than I flashed via serial connection, which worked. 
After a reboot the device didn't response to ping or webinterface.
I flashed again, after next reboot Device didn't connect to wifi.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues and in the wiki
- [X] Hardware used : Sonoff Basic R2, V1.0, 2017-10-11
- [ ] Provide the output of command ``status 0`` :
```
3:38:33 CMD: status 0
13:38:33 MQT: stat/sonoffBasic_1/STATUS = {"Status":{"Module":1,"FriendlyName":["SonoffBasic1"],"Topic":"sonoffBasic_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"xxxxxxxxxxxx","OtaUrl":"http://192.168.xxx.xxx:8080/static/firmware/xxxxx.ino.bin","RestartReason":"Power on","Uptime":"0T00:01:07","StartupUTC":"2018-07-16T11:37:26","Sleep":0,"BootCount":15,"SaveCount":32,"SaveAddress":"F9000"}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018.07.16 13:36:23","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"192.168.xxx.xxx","LogPort":514,"SSId":["xxxxx","sonoffap"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS4 = {"StatusMEM":{"ProgramSize":554,"Free":448,"Heap":9,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","1FDAE594","00000010","23B6179E","00000000"]}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS5 = {"StatusNET":{"Hostname":"SonoffBasic1","IPAddress":"192.168.xxx.xxx0","Gateway":"192.168.xxx.xxx","Subnetmask":"255.255.255.0","DNSServer":"192.168.xxx.xxx","Mac":"EC:FA:BC:86:A7:53","Webserver":2,"WifiConfig":3}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xxx.xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86A753","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 16 11:38:33 2018","Local":"Mon Jul 16 13:38:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:24","Sunset":"21:06"}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS10 = {"StatusSNS":{"Time":"2018-07-16T13:38:33"}}
13:38:33 MQT: stat/sonoffBasic_1/STATUS11 = {"StatusSTS":{"Time":"2018-07-16T13:38:33","Uptime":"0T00:01:07","Vcc":3.361,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxx","RSSI":74,"APMac":"9C:C7:A6:F7:72:A2"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-07-16:
Hi,

you are using "Core":"2_4_0"

That is the problem. That library is unstable.

Please use 2.3.0

Thanks

helmar74, 2018-07-16:
Hi, tried with 2.3.0. The Basic is running well.
Thanks a lot.

But, why are the other sonoff devices (SC, S20, S26, 4CH, 433Bridge, B1) running well with 2.4.0?

I'm quite unsure which libary to use... should I use 2.3.0 for all devices, or 2.4.0 for all (except of sonoff basic) or does 2.4.1 work for every device?

Thanks a lot

Helmar

ascillato, 2018-07-16:
> But, why are the other sonoff devices (SC, S20, S26, 4CH, 433Bridge, B1) running well with 2.4.0?

well, 2.4.0 is unstable, so you were lucky.

> I'm quite unsure which libary to use... should I use 2.3.0 for all devices, or 2.4.0 for all (except of sonoff basic) or does 2.4.1 work for every device?

by now, the most stable is 2.3.0 for all devices, until 2.4.2 or higher is released by Arduino/esp8266

ascillato, 2018-07-16:
Hi

If your issue is solved, please close it. Thanks :+1:

ascillato, 2018-07-23:
Hi,

Using your STATUS 0
```
{"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://xxx/tasmota/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:52","StartupUTC":"2018-07-23T11:18:13","Sleep":0,"BootCount":1,"SaveCount":4,"SaveAddress":"F8000"},"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018.07.22 18:55:38","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.6.5","LogPort":514,"SSId":["IoT"],"TelePeriod":300,"SetOption":["00008009","55818000"]},"StatusMEM":{"ProgramSize":474,"Free":528,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","0C000000","B7B61F9E","00000000"]},"StatusNET":{"Hostname":"sonoff-7267","IPAddress":"192.168.6.62","Gateway":"192.168.6.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.6.15","Mac":"5C:CF:7F:72:1C:63","Webserver":2,"WifiConfig":5},"StatusMQT":{"MqttHost":"192.168.6.47","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_721C63","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Mon Jul 23 11:27:06 2018","Local":"Mon Jul 23 12:27:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:43","Sunset":"20:19"},"StatusSNS":{"Time":"2018-07-23T12:27:06"},"StatusSTS":{"Time":"2018-07-23T12:27:06","Uptime":"0T00:08:53","Vcc":3.187,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","RSSI":64,"APMac":"12:9F:C2:A4:00:9B"}}}
```

with the tool _decode-status.py_, we can see:

```
Decoding information for device Sonoff from status report taken at 2018-07-23T12:27:06

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) Rules until 5.14.0b
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led

Features
  Language LCID = 1031
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_BH1750
  USE_BME680
  USE_BMP
  USE_DHT
  USE_DISCOVERY
  USE_DOMOTICZ
  USE_DS18B20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HOME_ASSISTANT
  USE_HTU
  USE_I2C
  USE_IR_RECEIVE
  USE_IR_REMOTE
  USE_LM75AD
  USE_MHZ19
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_RULES
  USE_SDM120
  USE_SDM630
  USE_SENSEAIR
  USE_SERIAL_BRIDGE
  USE_SGP30
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_TM1638
  USE_WEBSERVER
  USE_WS2812
  VTABLES_IN_FLASH
  WEBSERVER_ADVERTISE
```

So, please, try to recompile changing MQTT_PUBSUBCLIENT to TASMOTA_MQTT and test again.

ChipStewart1, 2018-07-23:
Um, how do I use decode-status.py?

ascillato, 2018-07-23:
@ChipStewart1

> Um, how do I use decode-status.py?

Instructions are inside that python script:

please see https://github.com/arendst/Sonoff-Tasmota/blob/development/tools/decode-status.py