arendst/Tasmota #3239: 1-10v pwm dimming, meanwell driver, d1 mini

regoobgib, 2018-07-16:
first just wanna say i really appreciate all the time and effort that has gone Tasmota.  

i have tasmota up and running on a d1 mini pro.  the d1mini is connected to a Sharp PC817 optoisolator (phototransistor) and i am using it to control the dimming on a Meanwell HLG led driver.  The HLG has a few options for dimming and Im using PWMi ( inverted PWM) and its working good. 

what im trying to accomplish is having muliple dimming sliders, the dark to bright kind.  if i select pwmi 1 and pwmi 2 i get dark to bright and cold to warm sliders. ive tried all the different pins, and tried searching through all the previous posts, couldnt find any info related to what i want.  

Is it possible to have multiple sliders, 3 seperate pins on the a single d1mini each hooked up to a different meanwell HLG driver and each controlled separately??  sorry if im posting this in the wrong section.  

thank you for any help anyone can offer.

digiblur, 2018-07-16:
I noticed this issue as well when I selected a second PWM for a LED this weekend. Not sure of the intended behavior but I couldn't figure out how to have two separate sliders for the two LEDs on different GPIO pins.

arendst, 2018-07-17:
Not supported

regoobgib, 2018-07-17:
oh well. thank you

closed 2018-07-17 by regoobgib