arendst/Tasmota #3244: ESPTOOL issue with <argument #2>

mike2nl, 2018-07-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : 3 different USB-TTL adapter one is original from Adafruit 
- [x ] Provide the output of command ``status 0`` : Not needed

Yes, i searched if there any information about it.
i used external sources from Ch. Klippel <ck at atelier-klippel dot de>

Issue with the esptool which is not correct documented in:
-> https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool <-

There you can find the information:
-> Optional: Erase firmware
-> Ensure the device is in firmware upload mode.
-> Erase the complete flash memory holding the firmware with the following command:
-> 
-> esptool.py --port COM5 erase_flash

The tool is to find in the Arduino IDE package. In my case under:
-> C:\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.13

But it does'nt work so as the information suggest!

--------------------------------------------------------
Open the cmd.exe as administrator, goto to your folder
where you can find your esptool and then type: esptool -h then use 'Enter'

Here is a short list of commands:

esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
ESP8266/ESP32 build and upload helper tool
Maintained by Ivan Grokhotkov: https://github.com/igrr/esptool-ck

The program interprets arguments given on the command line, and in the order
they are given.

-cp <device>
    Select the serial port device to use for communicating with the ESP.
    Default is /dev/ttyUSB0 on Linux, COM1 on Windows, /dev/tty.usbserial on Mac OS.

-cb <baudrate>
    Select the baudrate to use, default is 115200.

-ce
    Erase flash.

Commands tested and they are working 100% (COM18 in my case, yours can be different):
-> esptool -cp COM18 -cb 19200 -ce
-> esptool -cp COM18 -cb 115200 -ce

Possible Theo can fix the issue, Thank You.

ascillato, 2018-07-17:
Hi

esptool is different than esptool.py explained in the wiki. 

It is good to add also this information to the wiki. 

Thanks

arendst, 2018-07-17:
Pls do NOT updage the esptool info in the wiki.

There  is apples and pears, both fruit but...

The esptool from the wiki is the original tool as provided from the espressif github. The one noticed here is a stripped down version used by the arduino ide. The latter is NOT to be used as documented in the wiki.

mike2nl, 2018-07-17:
arendst,
Thanks for the information and it's good to know that it is a stripped down version.
Ok will take a deeeeeeper look in it ;-). I mean the phyton version.

closed 2018-07-17 by mike2nl