arendst/Tasmota #3245: Wemos Mini not starting when BH1750 is connected

helmar74, 2018-07-17:
Hi, 

I connected a BH1750 and a WS2812 to my Wemos Mini.
WS2812 is connected to D4, I2C SCL to D7 and I2C SDA to D8
The Wemos Mini doesn't start, when the BH1750 Sensor is connected. The Blue LED of Wemos is only flashing once, that's all what's happening.
The Ws2812 LED is working fine
When I disconnect the BH1750 sensor, restart Wemos and then connect the sensor the Illuminance is shown correctly. It couldn't be a power issue, I'm running on a 5V/10A Power supply. The BH1750 and the WS2812 are directly connected to power supply

[image]

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Wemos Mini V3.0.0
- [X] Provide the output of command ``status 0`` :
```
19:15:02 CMD: status 0
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoffWemosMini-5"],"Topic":"sonoffWemosMini_5","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffBasicStandardGruppe","OtaUrl":"http://192.168.xxx.xxx/firmware/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:00:53","StartupUTC":"2018-07-17T17:14:09","Sleep":0,"BootCount":43,"SaveCount":165,"SaveAddress":"F9000"}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018.07.17 15:52:38","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXX","hs.net_Keller"],"TelePeriod":300,"SetOption":["00028009","55818000"]}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS4 = {"StatusMEM":{"ProgramSize":539,"Free":464,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000407","0FDAE794","00000000","23B6179E","00000000"]}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-5","IPAddress":"192.168XXx.XXX","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"DC:4F:22:1D:88:E5","Webserver":2,"WifiConfig":3}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D88E5","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 17 17:15:02 2018","Local":"Tue Jul 17 19:15:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:25","Sunset":"21:05"}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS10 = {"StatusSNS":{"Time":"2018-07-17T19:15:02","BH1750":{"Illuminance":25}}}
19:15:02 MQT: stat/sonoffWemosMini_5/STATUS11 = {"StatusSTS":{"Time":"2018-07-17T19:15:02","Uptime":"0T00:00:53","Vcc":2.814,"POWER":"ON","Dimmer":40,"Color":"102,102,0,0","HSBColor":"60,100,40","Channel":[40,40,0,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"XXXXX","RSSI":64,"APMac":"CC:CE:1E:AB:0A:3F"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-07-17:
Dont Use GPIO 15. This Pin is critical at startup since ist has to be low at this time.
Use for example GPIO 12, 13 or 14
Gpio 2 is maybe a proble´m too. This pin has to be high at startup. Better use Gpio 4 or Gpio 5

I am pretty sure after this changes it will work as you want

helmar74, 2018-07-18:
Hi thanks a lot. Will try and give you feedback. 
Is there a list (suggestion) which GPIOs should be used for what sensor, and which gpios shouldn't be used?
Would be helpful for future projects....

Thanks a lot

Helmar

helmar74, 2018-07-18:
With this settings it's working now:
[image]

Thanks a lot !

closed 2018-07-18 by helmar74