arendst/Tasmota #3246: Sonoff s20 not flashing

SSkull995, 2018-07-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

gertmenke, 2018-07-17:
And, since this is certainly not a code issue - is there a policy what qualifies as (GitHub-)issue and what should rather be discussed on the mailing list? (I'm unsure myself - it feels to me like GitHub should be mainly about coding stuff, but OTOH I loathe Google Groups and would be happy to discuss stuff here.)

SSkull995, 2018-07-17:
Wait, what do you mean about coding? I saw many discussions just about the flashing problem.
I really don't know why it does not switch up, I will try using the pin 3.3v on the board itself

gertmenke, 2018-07-17:
My bad - carry on, then :-)

What do you mean by "the pin 3.3v on the board itself"?
What other pin would you use to supply the power?

SSkull995, 2018-07-17:
I tried to use the other 3.3v pin I have on the fdti, but nothing works. Really don't know. the plug works if connected to AC

Jason2866, 2018-07-17:
**FTDI often fails due to the fact the dont deliever not enough current on the 3.3V pin.**
Use a CH340 USB serial adapter OR Power the device with a extra sufficent 3.3 V power supply
There are many issues about that....

Frogmore42, 2018-07-17:
Yes, the issues are Littered with Many people having issues with flashing.  I keep telling people that the manufacturer of the FTDI part says they can't deliver enough current to flash an esp8266.  But, it works often enough that people seem loathe to believe it.

Part of the problem is there are many variants and copies of those.  So you never really know what you are getting.  I discovered that my FTDI sourced converter actually supplies VBUS (typically 5V) on the VCC line.  It actually worked fine on several Sonoffs till I discovered the issue when I was having problems with a Dual.  I then added a regulator to drop the voltage, but the damage was already done to the dual.

Finally, generally, Github issues are not supposed to be used to get help on how to use things or discuss things.  That is the purpose of forums.  But, making that happen for this project appears to be much harder than flashing an esp8266 with an FTDI 😀

SSkull995, 2018-07-18:
### 

-

SSkull995, 2018-07-18:
Hi, so I bought a ch340 wemos ttl adapter. I will try with it, I hope it will work. This adapter has a dedicated 3.3v output, let's see

SSkull995, 2018-07-19:
ok so I tried the wemos serial ttl, 3.3v set. Nothing works! It doesn't switch up!

Jason2866, 2018-07-19:
@SSkull995 
Wemos???

SSkull995, 2018-07-19:
@Jason2866 
Yeah it’s a board I found at a shop here in Eindhoven. I post the link:
https://www.tinytronics.nl/shop/nl/arduino/wemos/wemos-ch340-3.3v-5v-ttl-usb-serial-port-adapter?search=Wemos

SSkull995, 2018-07-19:
It has a regulator separated form the main board, which I had to solder to 3.3v. However, nothing appears to work. I tried sonOTA, but I get err0 after connectin back to my home WiFi

SSkull995, 2018-07-19:
The plug works with its own app

Frogmore42, 2018-07-19:
How do you have it hooked up?  Take a picture and post it.  Take multiple to show both sides of the board(s).

Jason2866, 2018-07-19:
As far as i can see this is a variant of a usual ch340 usb serial adapter.
You need no app... you have to install the ch340 drivers on your pc and connect it via usb cable to it.
The connect to the sonoff is made via the black connector. 
As Frogmore42 said we need pictures of your setup...

Jason2866, 2018-07-19:
At banggood i found this for the adapter:

Customize R/C port:
RTS:Connect R/C port to RTS, it can flash ESP8266 module(such as: ESP-01 02 ... 12F)
CTS:Connect R/C port to CTS, it can flash Arduino boards(such as: Arduino pro mini)

Operating voltage:

5V (The max current depends on your USB port, usually 500mA)
3.3V (Using LDO, 300mA MAX)

SSkull995, 2018-07-19:
Hi as soon as I reach home here I post a pic of the connections and everything. Actually with this module I could flash the sonoff basic I have here, not the s20 DAMNNN ahahahah. I will post the figure of everything ASAP, thanks for replies!

SSkull995, 2018-07-20:
[image]

SSkull995, 2018-07-20:
[image]

SSkull995, 2018-07-20:
This is another adapter I found in the house. I am sorry if I reply just know buy yesterday I landed late and reached home even later

SSkull995, 2018-07-20:
[image]

SSkull995, 2018-07-20:
[image]

SSkull995, 2018-07-20:
this is the demos I talked about. As it is possible to see, with just the power and the end cables the s20 does not switch on at all

SSkull995, 2018-07-20:
[image]

SSkull995, 2018-07-20:
tisi the back of the s20, with the soldered pins

Jason2866, 2018-07-20:
It is normal that relais is not switching when powered with 3.3 Volt (via adapter or 3.3 power supply)
If flashing is successfull everything is fine. Disconnect the wires and reasemble the Sonoff.
Put it in the Power Socket an configure it....

gertmenke, 2018-07-20:
I understood it as "the ESP does not power on at all", not just the relais.
Have you checked that your soldered connections are really okay?

SSkull995, 2018-07-20:
@gertmenke @Jason2866 what I meant is that if I connect the adapter to the basic sonoff the led starts to blink. While if I connect it to the s20, no led blinks

SSkull995, 2018-07-20:
I will check the soldered pins and @Jason2866 if I try to flash it the error about upload memory comes up with all the other ESPs and thus no flash succeeds

gertmenke, 2018-07-20:
I think they should blink, but nevertheless, could you get a connection with esptool? (Or whatever tool you use to flash)

SSkull995, 2018-07-20:
no nothing @gertmenke, I use the platformio based atom

SSkull995, 2018-07-20:
this damn sonoff is driving me crazy

gertmenke, 2018-07-20:
So you get the same output as when you don't connect the thing at all, correct?

SSkull995, 2018-07-20:
yes like no possibility to connect to the s20, no connection at all

gertmenke, 2018-07-20:
My best bet would be the soldering, those joints don't look very good, TBH.

gertmenke, 2018-07-20:
Do you have a continuity tester?

SSkull995, 2018-07-20:
yeah actually I saw it when I was taking the pictures. I don't know why but the tin does not really keep sticked to the pins

SSkull995, 2018-07-20:
Yes, I will try to see the voltage output from the adapter. Just a minute

SSkull995, 2018-07-20:
ok so i re-soldered the pins. I also tried some components on the boards and the current is always 3.3 everywhere. However, no led blinks

gertmenke, 2018-07-20:
What does platformio/atom say when you try to flash? Any difference to before?

SSkull995, 2018-07-20:
I maaaaaaaaade it, I just disconnected the gnd soldered pin and I used a male jumper. Done!!! 
Thank you very much to both of you, really thank you!

gertmenke, 2018-07-20:
I'm happy to hear that.
Cheers!

gertmenke, 2018-07-20:
If it works now, you should probably close the issue.
Have fun! :-)