arendst/Tasmota #3249: Flashed tasmota can't connect to wifi (wrong password?)

amore75, 2018-07-18:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff and USB to TTL adapter
- [x] Provide the output of command ``status 0`` :
```
00:01:48 CMD: status 0
00:01:48 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:48 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:47","StartupUTC":"","Sleep":0,"BootCount":3,"SaveCount":7,"SaveAddress":"F5000"}}
00:01:48 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-07-17T20:50:29","Boot":31,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
00:01:48 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FiOS-17CL1","FiOS-17CL1"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:01:48 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":452,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:01:48 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4414","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"B4:E6:2D:25:31:3E","Webserver":2,"WifiConfig":2}}
00:01:48 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25313E","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:48 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:48 1970","Local":"Thu Jan 01 00:01:48 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:48 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:48"}}
00:01:48 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:48","Uptime":"0T00:01:47","Vcc":4.466,"POWER":"OFF","Wifi":{"AP":2,"SSId":"FiOS-17CL1","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}
****
```
The sonoff shows up on my wifi list and "connects" but when I try to open the webUI it doesn't open and termite shows it's not connected.  I'm coming to the conclusion it's my router/network (Verizon Fios, router is Fios Quantum Gateway) b/c when I tested it at work (Cox with TPLink router), it worked fine.

Tried turning off the 5g, changed SSID with only letters/numbers, and just about all the "fixes" I could find online.  So far, nothing worked.  I'd appreciate any help you guys can provide.  Thanks!

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-07-18:
You are using a core that is known to have problems.  Search the issues list and you will using core 2.4.x is frequently problematic.  It works fine for some people/devices/environments and fails unpredictably for others.

Search again for core 2.4.0.  There are issues, and help in the wiki on how to change it.

amore75, 2018-07-18:
Frogmore42:  Thank you for the info!  I'm totally new to this whole Sonoff/Tasmota so was only limited to searching for issues with key words that showed (ie incorrect password).

Frogmore42, 2018-07-18:
You don't need to turn off 5GHz on your router, but the esp8266 is 2.4GHz only, so you have to have that on.  Yes, core refers to the core software libraries of the esp8266.  Please build the software with 2.3.0 and try again.  Collect logs showing what is happening from startup and also status 0.

amore75, 2018-07-19:
I've downgraded to 2.3.0 and did 2 flashes.  1st flash is without my SSID/PW and the 2nd flash is with. 
Both shows up on my wifi list and "connects" but webUI won't load.

00:01:31 CMD: status 0
00:01:31 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:31 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:30","StartupUTC":"","Sleep":0,"BootCount":11,"SaveCount":12,"SaveAddress":"F8000"}}
00:01:31 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-18T21:33:49","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:01:31 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:01:31 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
00:01:31 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4414","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"B4:E6:2D:25:31:3E","Webserver":2,"WifiConfig":2}}
00:01:31 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25313E","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:31 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:31 1970","Local":"Thu Jan 01 00:01:31 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:31 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:31"}}
00:01:31 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:31","Uptime":"0T00:01:30","Vcc":4.094,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}
-----------------------------------------------------------------------------------------------------------------

2nd flash with SSID/PW

00:01:22 CMD: status 0
00:01:22 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:22 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:22","StartupUTC":"","Sleep":0,"BootCount":3,"SaveCount":4,"SaveAddress":"F8000"}}
00:01:22 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-18T22:12:45","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:01:22 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxx",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:01:22 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
00:01:22 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4414","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"B4:E6:2D:25:31:3E","Webserver":2,"WifiConfig":2}}
00:01:22 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25313E","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:22 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:22 1970","Local":"Thu Jan 01 00:01:22 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:22 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:22"}}
00:01:22 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:22","Uptime":"0T00:01:22","Vcc":4.094,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxx","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}

Frogmore42, 2018-07-19:
How are you powering it?  The VCC is listed as being 4.094V.  The only time I have seen it that high is when I powered it from my FTDI that has VCC directly connected to VBUS.  Over 3V6 can damage the esp8266, you need to fix that.

You also did not show the rest of the log where it is trying to connect.

Frogmore42, 2018-07-19:
Does ping work?
How are you trying to connect, by name or by IP address?

amore75, 2018-07-19:
@Frogmore42 

I'm powering through the USB/TTL adapter as shown in the tutorials.  I've connected 3v-3v, RX-TX, TX-RX, GND-GND.  I got the status 0 using Termite app and that's all that showed. Is there another app that would show more info?  And I've been trying to connect via name since I can't connect to the wifi properly, I have no way of finding the IP.  So can't ping it either...

@pReya 

It's driving me nuts b/c I can get THIS particular sonoff to work at my office but not at home.  But I have another sonoff (bought from different vendor) that I couldn't get to work either at home OR office.

Frogmore42, 2018-07-19:
If you have a meter, you should check the actual voltage.  While the ADC in the esp8266 is not very accurate, I have never seen it read 4V when connected to 3V3.  I have seen it do that when it was connected to VBUS through a diode.

You can find the IP by looking at the serial output, it lists the IP address it is using.  You can also find it by looking at the management page of you router/AP.  But, looking closer at you status 0, I see that it doesn't have an IP address, so it may not be connected.

I would double check the voltage, because the level the ADC is seeing can cause a problem with the chip.  You might also have a defective one.  I had one of those.  If it had been my only one, I would have been very frustrated.  I have a NodeMCU board probably has a defective regulator on it and it won't stay running from more than a day or sometimes a few min, but it always restarts and then works for awhile (so unless I am looking at it very closely it appears to be working).

amore75, 2018-07-20:
I found something wrong with the wifi on my pc so I reinstalled the driver.  Now I got it to the point where the 192.168.4.1 comes up.  I saved the ssid/pw and rebooted.  But I still can't get it to show on my network.  

@Frogmore42   I didn't realize I had to change the jumper to get to the 3.5v.  I fixed that and now get 3v according to the meter.

~~~
00:01:34 CMD: status 0
00:01:34 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:34 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:34","StartupUTC":"","Sleep":0,"BootCount":10,"SaveCount":22,"SaveAddress":"F6000"}}
00:01:34 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-19T21:48:25","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:01:34 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxx","xxxxxxxxx"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:01:34 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
00:01:34 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1978","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"DC:4F:22:4B:67:BA","Webserver":2,"WifiConfig":2}}
00:01:34 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B67BA","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:34 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:34 1970","Local":"Thu Jan 01 00:01:34 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:34 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:34"}}
00:01:34 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:34","Uptime":"0T00:01:34","Vcc":2.979,"POWER":"OFF","Wifi":{"AP":2,"SSId":"xxxxxxxxx","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
~~~

Frogmore42, 2018-07-20:
It's possible your esp8266 is damaged.  I have one that I can't flash, perhaps because it got damaged by too high voltage.

But, your IP settings look strange.  It has no IP address,  but does have other settings.  Are you using DHCP or static?

amore75, 2018-07-20:
Using DHCP.  I have 4 sonoffs.  I tried flashing 3 with same result.  The last one, I used ewenlink to test if there was a problem with my router but that one connected on the 2nd try.

Frogmore42, 2018-07-20:
I would try the basic Wi-Fi example from esp8266 Arduino and see if that works.  There seems to be a low level issue here that is probably not Tasmota related.

motaz9k, 2018-07-20:
@amore75 in user_config.h change `#define MODULE SONOFF_BASIC`
to
`#define MODULE WEMOS` and see that everything will work

amore75, 2018-07-21:
@Frogmore42 Will try that one.

@motaz9k  Tried and failed.  Same result.

Frogmore42, 2018-07-21:
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/readme.html

Try the most basic example there, since you are not even getting an IP address it seems.

amore75, 2018-07-21:
@Frogmore42 I went through the basic wifi connection and it works! I can see the IP as it shows in the example.

Frogmore42, 2018-07-21:
The basic WiFi sample does not have a web server so you can connect to it that way, but I would expect it to respond to a ping.

There should be an example web server in the esp8266 Arduino samples.  You could try that and see if you can connect to that.

But, I don't understand what you mean by ESP_xxxxxx shows up but then disappears and something else shows up.  Where?

In all of the logs you showed with Tasmota, the DHCP process was not completed correctly, since it always had no IP address, but the rest of it looked like it could be correct.  I have never seen that before and can't explain how that might happen.

If you have other esp8266 devices that work correctly, it likely means there is something wrong with this device and I would just get another one.

If you have multiple esp8266 devices and they all behave badly, that probably means you still have something wrong with the way you are building/installing/configuring the Tasmota image.

ascillato, 2018-07-23:
Hi, please check https://github.com/arendst/Sonoff-Tasmota/issues/3289. Thanks

amore75, 2018-07-26:
I got it to work. Flashed as ascillato directed. Totally changed the name of my SSID/PW, rebooted.
Flashed again using Dr. Zzzs newest method (uploaded 07/25/18). Copy/pasted SSID/PW in termite.

Thanks to everyone for all the suggestions and advice!

closed 2018-07-26 by amore75