arendst/Tasmota #3250: DHT issue Wemos D1 Mini

robgardien, 2018-07-18:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Wemos D1 mini
- [ x] Provide the output of command ``status 0`` :
```
 07:51:16 CMD: Status 0
07:51:16 RSL: Ontvangen topic /Status, Data lengte 1, Data 0
07:51:16 RSL: Groep 0, Index 1, Opdracht STATUS, Data 0
07:51:16 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["WemosD1"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:51:16 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:04:07","StartupUTC":"2018-07-18T06:47:09","Sleep":0,"BootCount":16,"SaveCount":29,"SaveAddress":"F7000"}}
07:51:16 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-07-18T08:46:35","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:51:16 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FamilieGardien","Indebuurt1"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
07:51:16 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":554,"Free":448,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
07:51:16 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5229","IPAddress":"192.168.43.196","Gateway":"192.168.43.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.43.1","Mac":"68:C6:3A:D6:34:6D","Webserver":2,"WifiConfig":3}}
07:51:16 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.43","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D6346D","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:51:16 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Jul 18 06:51:16 2018","Local":"Wed Jul 18 07:51:16 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:06","Sunset":"20:46"}}
07:51:16 DHT: Te lang wachten op laag start signaal signaal
07:51:16 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-07-18T07:51:16"}}
07:51:16 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-07-18T07:51:16","Uptime":"0T00:04:07","Vcc":2.698,"Wifi":{"AP":2,"SSId":"Indebuurt1","RSSI":100,"APMac":"BA:D7:AF:68:49:65"}}}
```

When I try to connect my DHT11 to my Wemos D1 mini I receive an error:

DHT: Te lang wachten op laag start signaal signaal
(DHT: To long wait on low start signal signal)

ascillato, 2018-07-18:
Hi,

This issue is duplicated and has been addressed several times. Please search in issues for DHT11 and you will have a lot of solutions. (https://github.com/arendst/Sonoff-Tasmota/issues/1745) and [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-DHT11-Shield-Humidity-&-Temperature).

The message you receive means that your DHT Sensor has a wiring problem. Please check that. Do you have a 4.7kohm resistor on DHT11 between 3.3V and DATA pin ?
(https://i0.wp.com/www.electrodummies.net/wp-content/uploads/2017/02/dht11-library-and-schematic-1.png)

ascillato2, 2018-07-21:
Please, reopen this issue if you need more help. Thanks

closed 2018-07-21 by ascillato2