arendst/Tasmota #3253: Tasmota, Home Assistant, mqtt e switch config?

umibozu83, 2018-07-18:
Hi I'm just using tasmofa firmware with my wemos mini and the command with Home assistent through mqtt. all working just some info:

I'm using this code to control the switch:

`switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: false`

on the wiki he tells me to use this with the latest versions of Home Assistant:

`- platform: mqtt
  name: "On/Off Sonoff"
  state_topic: "stat/sonoff/RESULT"
  value_template: '{{ value_json["POWER"] }}'
  command_topic: "cmnd/sonoff/POWER"
  availability_topic: "tele/sonoff/LWT"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  retain: false`

what's the difference?

with the first code with a small change I can command more relays with wemos, how can I do it with the second code? what do I change?

Jason2866, 2018-07-18:
Hello @umibozu83 
this is not a forum it is for issues that belongs to Tasmota.
You better ask her for help for HomeAssistant
https://community.home-assistant.io/

umibozu83, 2018-07-18:
sorry delete him

closed 2018-07-18 by umibozu83