arendst/Tasmota #3254: after update not showing temp

Dmxsir, 2018-07-18:
after update from var. 5.12 to 6.1.1b through Arduino IDE the temp from sensor (GPIO14) not Not displayed , not even zero.
i did choose from setting the right sensor type.

ascillato, 2018-07-18:
Please, complete the troubleshooting template

ascillato, 2018-07-18:
Hi,

Thanks for completing the template. :+1: 

I have used your `status 0` output with the tool **decode-status.py** that Theo has posted.

**status.json:**
```
{"Status":{"Module":1,"FriendlyName":["Sonofftemp"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://10.0.0.32:9541/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:32","StartupUTC":"2018-07-18T10:45:15","Sleep":0,"BootCount":33,"SaveCount":108,"SaveAddress":"FA000"},"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-14T15:57:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["maoz","indebuurt2"],"TelePeriod":100,"SetOption":["00000009","55818000"]},"StatusMEM":{"ProgramSize":487,"Free":516,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0052A794","0C000008","00261796","00000000"]},"StatusNET":{"Hostname":"sonofftemp","IPAddress":"10.0.0.17","Gateway":"10.0.0.1","Subnetmask":"255.0.0.0","DNSServer":"10.0.0.1","Mac":"EC:FA:BC:13:B5:81","Webserver":2,"WifiConfig":2},"StatusMQT":{"MqttHost":"10.0.0.32","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_13B581","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Wed Jul 18 11:20:47 2018","Local":"Wed Jul 18 12:20:47 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1},"StatusSNS":{"Time":"2018-07-18T12:20:47"},"StatusSTS":{"Time":"2018-07-18T12:20:47","Uptime":"0T00:35:32","Vcc":3.090,"POWER":"OFF","Wifi":{"AP":1,"SSId":"maoz","RSSI":60,"APMac":"CC:2D:E0:3E:D9:1F"}}}
```

**Output:**

```
Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (OFF) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) Rules until 5.14.0b
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group

Features
  Language LCID = 2057
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
  USE_ADC_VCC
  USE_BH1750
  USE_BMP
  USE_CLASSIC
  USE_DHT
  USE_DISCOVERY
  USE_DOMOTICZ
  USE_DS18B20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HOME_ASSISTANT
  USE_HTU
  USE_I2C
  USE_IR_REMOTE
  USE_MHZ19
  USE_PMS5003
  USE_SHT
  USE_SHT3X
  USE_WEBSERVER
  USE_WS2812
  VTABLES_IN_FLASH
  WEBSERVER_ADVERTISE
```

So, the driver for the DS18B20 is there.

All the wirings are ok?

Dmxsir, 2018-07-18:
I don't think that it is related to wiring.
Because when i choose the DS18B20 in a another  sonoff, It shows the header for the selected sensor temperature with 0%.
Now I don't even have that. As i post in the screenshot.

Frogmore42, 2018-07-18:
Do you have a pull-up on the data line?
What size?

Dmxsir, 2018-07-18:
Yes, the same one that was before the update.
but, again i dont think its related to that .
when you go to any sonoff-tasmota configuration>Configure Module>GPIO14 and select from the menu '04-DS18x20' . on the main page you should see above the on/off ,a title like `` DS18B20 temperature    0%`` with out to connect anything .

after the update there is nothing show up.

Dmxsir, 2018-07-18:
this one, dont have any sensor attach.
[image]
as you can see its still show the ``DS18B20 temperature 0% ``
[image]

Frogmore42, 2018-07-18:
I use the one that supports multiple sensors (USE_DS18X20).  It has to discover the sensors and it won't show any unless it finds at least one.  I believe the sensor code was updated relatively recently, so maybe it is not working as well for you.  Double check your resistor value is correct.  It is easy to make a mistake there that will cause it to work sometimes but not always and the new code may be doing something different.

Dmxsir, 2018-07-19:
The problem was found. wiring was mixed.
Apparently in the new version, only when there is a sensor connected, the information is displayed
thanks for you help .
At least I learned how to debug. :100:

closed 2018-07-19 by Dmxsir

fbbfnc, 2018-12-27:
Hey! 
After the update, from 5.13 to 6.4.1 the DS18B20 does not work anymore. Before the update there was the temperature measurement on the webserver. Immediately after the update there is no measurement, maybe because the Sonoff Basic is not able to read the sensor. The fact is that i didn't change anything on wires. I am using a 10kOhm resistor as a pull up, between data wire and Vcc wire. 

`11:02:40 CMD: status 0
11:02:40 SRC: WebConsole from 192.168.113.113
11:02:40 RSL: Received Topic /status, Data Size 1, Data 0
11:02:40 RSL: Group 0, Index 1, Command STATUS, Data 0
11:02:40 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:02:40 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.4.1/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:25","StartupUTC":"2018-12-27T09:59:15","Sleep":50,"BootCount":770,"SaveCount":2296,"SaveAddress":"F8000"}}
11:02:40 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(basic)","BuildDateTime":"2018-12-24T14:43:30","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:02:40 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["************************"],"TelePeriod":60,"SetOption":["00008009","558180C0","00000001"]}}
11:02:40 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0F082780","000083A0","21021706","000001C0"]}}
11:02:40 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-8144","IPAddress":"192.168.113.117","Gateway":"192.168.113.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.113.1","Mac":"84:F3:EB:AF:BF:D0","Webserver":2,"WifiConfig":2}}
11:02:40 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.113.119","MqttPort":1883,"MqttClientMask":"Sonoff1","MqttClient":"Sonoff1","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:02:40 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 27 10:02:40 2018","Local":"Thu Dec 27 11:02:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
11:02:40 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-12-27T11:02:40"}}
11:02:40 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-12-27T11:02:40","Uptime":"0T00:03:25","Vcc":3.543,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"*****************","BSSId":"***********************","Channel":11,"RSSI":28}}}
11:02:44 MQT: tele/sonoff/STATE = {"Time":"2018-12-27T11:02:44","Uptime":"0T00:03:29","Vcc":3.542,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"*****************","BSSId":"*****************","Channel":11,"RSSI":28}}
11:02:44 SRC: MQTT`