arendst/Tasmota #3255: After update HTTP server not working.

Chopin85, 2018-07-18:
After upgrade tasmota basic from web interface 5.14 -> 6.1.1 the web interface not working but mqtt work.

Help.

ascillato, 2018-07-18:
May be your device changed IP. Have you searched in your router?

Chopin85, 2018-07-18:
Yes i have 9 sonoff basic, 8 sonoff touch and 4 sonoff dual, all upgrade to 6.1.1 and now all sonoff web interface not work.

mike2nl, 2018-07-18:
Have you tested it with reconnect the USB-TTL adapter !without! main power or directly with main power supply in the housing?
With the USB-TTL adapter you can open the Serial Monitor in Arduion IDE to see what's going on.
Or you are a novice in serial monitoring?

brainnex, 2018-07-19:
Hello,
I have a similar problem. after upgrade from 5.12 ->6.1.1b or even 5.12 -> 5.14 -> 6.1.1b the Webserver is not reachable, because it wont connect to WiFi anymore. It is reproducible( 2 of 2 devices)  . The serial debug is showing:
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff3888
~ld
⸮
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_05D26D, GroupTopic sonoffs) Version 6.1.1-2_3_0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-4717...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 Machete_braucht_kein_WLAN in mode 11N as sonoff-4717...
00:00:16 WIF: Connect failed as AP cannot be reached
00:00:16 WIF: WPSConfig active for 3 minutes
00:00:35 APP: Restarting
```
AP 1 and even AP2 are working but the sonoff fails to find them. After restoring 5.12 the module config is lost but WiFi is working again.

ascillato2, 2018-07-21:
Please, see #3224

Chopin85, 2018-07-21:
Hi thanks you, so I need to downgrade at 5.14.

ascillato, 2018-07-23:
Hi, please check https://github.com/arendst/Sonoff-Tasmota/issues/3289. Thanks

ascillato2, 2018-08-18:
Hi,

Closing this issue as there is no feedback. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2