arendst/Tasmota #3257: Option to CloudMQTT

upendrawww9, 2018-07-18:
What is the other options for MQTT Broker 
With  CloudMQTT. I have connection problem and its not consistent.
Can I have name of any other free Broker works with 

 Regards
Upendra

**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-07-18:
I don't believe anyone who is serious about home automation, uses a broker in the cloud.  I have not heard of one that works well.  If you are just trying out things, running mosquitto is really easy on a laptop.  Many people use a Raspberry Pi to run it and their home automation software of choice.  Other people choose to have no broker at all and turn off MQTT.

You didn't provide any of the information in the template so it is hard to be more helpful.  It's possible you are using a known bad combination of things that is contributing to the problem.

curzon01, 2018-07-20:
Hint - the following is philosophical :wink:

I'd like to consider that's not in black/white thinking: For example a Rasperry Pi at home is (without additional steps) unreachable from outside - we can not look for e.g. temperature in our garden or something like that. Connecting a VPN before using some HA features has a WAF (**W**oman **A**cceptance **F**actor) near 0 - and if you are married you don't want having WAF <= 0.9 :smirk:

Differentiate several secnarios and not saying home automation on public broker is unsafe at all:
Examples:
- There is nothing to be said against e.g. bridging sensor values into the cloud by restrict the bridge access to specific topics. This may require planing of your topic hierarchy.
- Secondly it's also always a question what's unsafe depending of the HA environment if - let's say - a hacker got the 32 char user/password or found a security vulnerability: What's the result is he/she could see the humidty in the bathroom and control two shutter (because there is nothing more in this scenario) - so what.

What I finally want to say is looking beyond one's own nose - there is no bad or good decision at all, it depends in general.
That applies also to the question using public broker for HA.

@upendrawww9: There is not enough info to consider somthing. May be your problem is not broker based you got the same with other too. So what exact is the problem with current broker?

upendrawww9, 2018-07-20:
Can I have your personal email  I can share some details

Thanks
Upendra

curzon01, 2018-07-21:
@upendrawww9 - click on alias "curzon01" of my [post](https://github.com/arendst/Sonoff-Tasmota/issues/3257#issuecomment-406516932)

ascillato2, 2018-07-25:
Hi,

Have you managed to solve your issue?

upendrawww9, 2018-07-28:
Sorry the delay.I am still living with the problem.I am trying with others
brokers as cloud mqtt offers only 5 connections for free. Other way is
install mqtt in your own cloud space which is again $ cost.

Best way is to tweek keepAlive time in code

Regards
Upen

ascillato2, 2018-07-28:
CloudMQTT is not realiable for home automation. Sorry, better is to have it locally.

ascillato2, 2018-08-25:
Hi

Any news?

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2