arendst/Tasmota #3258: Predefine GPIO sensor in firmware

Loreman01, 2018-07-18:
Hey guys,

short question (Sonoff TH10):
I want to predefine DS18B20 to an specific GPIO for example GPIO14 in the firmware before compiling. Where can I find that?

I know that I can do the settings with the webpanel or with mqtt. I got 100 of these TH10 and I don't want to open a webpage for every single one of them. They not have the same grouptopic so mqtt is also not an option.

Thanks for any advice.

Loreman01, 2018-07-18:
I want to define it before flashing the device so I need to do the settings in the code before compiling.

Jason2866, 2018-07-18:
Pragmatic way. Compile a version you want. Flash it to ONE device. Do the GPIO setting you need.
Read the whole flash with Esptool. Write this image to all your devices with Esptool.
Dont use static IP ;-)

Loreman01, 2018-07-18:
Thanks, that will do it for now. Why did I not come to this by myself? :/

Still looking for an solution in the code.

andrethomas, 2018-07-18:
Or configure one device, export settings and import settings on all the others...

reloxx13, 2018-07-18:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff_template.h#318

Frogmore42, 2018-07-18:
Be aware that you will probably want to have unique topic names, which means you need to set that up too.  I would not want to do that in the code, since that would mean compiling 100 unique copies.

I don't know how/where you are planning on deplopying 100 devices that are all the same.  There is typically some more configuration that must be done.  I would do it over serial immediately after each one is flashed and automate that operation.  I have a checklist I use, since I typically do one or two every few months as I add stuff.  If I knew I was going to have to deploy 100 in a short period, I would take the time to automate it.

roguestreak, 2018-07-19:
I haven't tried http commands so don't know if it would work, but just throwing it out there; could you just flash them as per normal and come up with a batch that targets a range of IP addresses and sends the modified config backlog command via http?  ie. a backlog containing the standard sensor setup plus variable commands based on the device IP you are up to?
192.168.0.20 set as sonoff20
192.168.0.21 set as sonoff21...

Loreman01, 2018-07-19:
>Or configure one device, export settings and import settings on all the others...

That will keep me touching every single one device twice.

>Be aware that you will probably want to have unique topic names, which means you need to set that up too. I would not want to do that in the code, since that would mean compiling 100 unique copies.

Good Point but this process is already automated. I have already 500+ POWs in use. I don't want them to be alle the same I just want to do the most settings in the code instead of doing it after flashing because in some cases sonoff devices resetting by themselfs without warning. In this case they keep all the settings. In my case the  topic is mac address related so that will fit it.

>I would do it over serial immediately after each one is flashed and automate that operation.

I will give that a try.

Thanks to you all.

closed 2018-07-19 by Loreman01

Frogmore42, 2018-07-19:
Now I have to know, what you are doing with 500 POWs?
Are they all at the same location?
On the same network?
Have the same network name, even if at different location?

Loreman01, 2018-07-19:
>Are they all at the same location?

Yes.

>On the same network?

Yes.

>Have the same network name, even if at different location?

No. They are only different in the topic and the grouptopic.
I use them to measure the power consumption and as reboot function for cryptoservers. Topic just to identify every machine and grouptopic to shut down an entire section. I also use them to control an 600KW chilling unit (obvioulsy with some other stuff :D).