arendst/Tasmota #3260: SOnOff-SC, friendly name missing

mike2nl, 2018-07-18:
**Describe the bug**
_A clear and concise description of what the bug is._
When i open the Information web page for the SONOFF_SC i miss the entry for the Friendly Name
Please see screenshot Figure 2.

- [x ] Searched the problem in issues and in the wiki
- [ ] Hardware used :  SONOFF_SC
- [ ] Provide the output of command ``status 0`` :
Status 0:
00:04:53 CMD: status 0
00:04:53 RSL: STATUS = {"Status":{"Module":21,"FriendlyName":["envsensor"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:04:53 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:04:50","StartupUTC":"","Sleep":0,"BootCount":7,"SaveCount":10,"SaveAddress":"FA000"}}
00:04:53 RSL: STATUS2 = {"StatusFWR":{"Version":"6.1.0","BuildDateTime":"2018-07-18T13:20:58","Boot":0,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
00:04:53 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Linksys22267_2GHz",""],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
00:04:53 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F804394","00000000","23B6179E","00000000"]}}
00:04:53 RSL: STATUS5 = {"StatusNET":{"Hostname":"envsensor","IPAddress":"192.168.2.181","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"5C:CF:7F:68:9D:C9","Webserver":2,"WifiConfig":3}}
00:04:53 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:04:53 1970","Local":"Thu Jan 01 00:04:53 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:45","Sunset":"15:38"}}
00:04:54 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:04:53","SonoffSC":{"Temperature":1.0,"Humidity":1.0,"Light":80,"Noise":50,"AirQuality":90},"TempUnit":"C"}}
00:04:54 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:04:54","Uptime":"0T00:04:50","Vcc":3.519,"Wifi":{"AP":1,"SSId":"Linksys22267_2GHz","RSSI":100,"APMac":"C0:56:27:61:5E:06"}}}

**To Reproduce**
_Steps to reproduce the behavior:_
First you need the SONOFF_SC with PCB version 2.0 !  
Change the configuration based on my Status.
Then compile and upload it.
Next open the Information web page for the sOnOff_SC.

**Expected behavior**
_A clear and concise description of what you expected to happen._
I would expect to see the Friendly Name in the 1st block of information.

**Screenshots**
_If applicable, add screenshots to help explain your problem._
Figue 1, screenshot from a SONOFF_BASIC:
[image]

Figure 2, screenshot from the SONOFF_SC, missing Friendly Name:
[image]

I have done some tests with the FRIENDLY_NAME variable in the source code and i can get
the friendly name visible but also a button to toggle. The funny part is when is hit the toggle
button i can switch on the green LED near the USB power connector.

  else if (SONOFF_SC == Settings.module) {
    Settings.flag.mqtt_serial = 0;
    devices_present = 0;	
    baudrate = 19200;
  }

Hope that helps otherwise please ask me.

ascillato, 2018-07-18:
If you don't have any relay or output defined, you will not have a friendly name to be used by Alexa u other.

Hostname is different than friendly name. The first one is the wifi name of the device, the second one is the name of the Output defined.

No output, no friendly name

mike2nl, 2018-07-18:
I know that because i engineer networks in HighSpeed EthernetPowerLink EPL.
So there is something missing in the code.
I had taken a deep look into the code from tasmota and found serveral places where 
changings can be done to fix it.

reloxx13, 2018-07-18:
again:

`If you don't have any relay or output defined, you will not have a friendly name to be used by Alexa`

show us your device config page with a relay set or confirm that you set a relay, saved, waited for restart and there is still no friendly name input.

mike2nl, 2018-07-18:
relox,
i have one output to switch the green LED near the power USB connector.
In my other post you can find it think.
I can swicht also the extra LED connector on and off.
Is that output enough?
I'm not a beginner in coding and must read unknown code too on daily base.

A friendly name is one part but i will discusse that not right here because it's not the right place.
I have shown something and when Theo say no, no stop for me i code it and can publish the code. Thats it.

mike2nl, 2018-07-18:
ascilloato,
it was no my intention to do so, really. I understand the way of working in these types of groups and i'm happy when people are willing to help. I help where i can and that a lot when i look in Android and industrial coding.

And i'm more then happy that you had found it ;-). THANK YOU

I hope we could clear it up because i wish to work a longer time with tasmota i had tested many other distributions around sonoff and tasmota is one of the best, when not the best so far i have seen.

mike2nl, 2018-07-19:
ascilloato,
to be honest: 
i'm working for many years now here on github for posting and addtions to many different wiki's,
but never ever i had done a PR (PullRequest). I hope i get it otherwise i will call for help.

Solution is here quick and slappy ;-), because i have not much time over these weeks.
Tools:
- Notepad++

- changes in file:
 -> xdrv_02_webserver.ino

- search for: 
-> void HandleInformation()
-> in line: 1213

- search for: 
-> if (SONOFF_IFAN02 == Settings.module) { maxfn = 1; }
-> in line: 1244

- action: set this line in comment by adding two slashes in front of the line '//'

- action: add a new line and fill it with: 
-> if ((SONOFF_IFAN02 == Settings.module) || (SONOFF_SC == Settings.module)) { maxfn = 1; }
-> in line 1245

- verrify it
- compile and upload
- test it

Figure 1:
[image]

Next step the PR, hope it works.

What would very nice if one or two users can do the same to get some feedback if it's working
for them too.

ascillato2, 2018-07-19:
Issue deleted by its author (??) Closing...

closed 2018-07-19 by ascillato2