arendst/Tasmota #3262: Ping not possible, except from MQTT host

nappe-karl, 2018-07-18:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used :  Wemos D1 mini and NodeMCU
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
Hi,

I have compiled version 6.1.1b with Arduino IDE 1.8.5, flashed on  Wemos D1 mini and NodeMCU.
Both worked, configuration was possible. After some minutes ping and webserver did not respond, tested from Windows 10 and Rasperry.

But: Ping is always successful  from MQTT host!

Output from Arduino IDE serial monitor seems to be ok:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0318B4, GroupTopic sonoffs) Version 6.1.1b-2_4_1
00:00:00 WIF: Connecting to AP1 myfritz in mode 11N as sonoff-6324...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-6324.local with IP address 192.168.1.56
18:58:43 MQT: Attempting connection...
18:58:43 MQT: Connected
18:58:43 MQT: tele/sonoff/LWT = Online (retained)
18:58:43 MQT: cmnd/sonoff/POWER = 
18:58:43 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.1.1b","FallbackTopic":"DVES_0318B4","GroupTopic":"sonoffs"}
18:58:43 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-6324","IPAddress":"192.168.1.56"}
18:58:43 MQT: tele/sonoff/INFO3 = {"RestartReason":"External System"}
18:58:44 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
18:58:44 MQT: stat/sonoff/POWER = OFF
18:58:51 RSL: tele/sonoff/STATE = {"Time":"2018-07-18T18:58:51","Uptime":"0T00:00:13","Vcc":3.000,"POWER":"OFF","Wifi":{"AP":1,"SSId":"myfritz","RSSI":52,"APMac":"FF:FF:FF:FF:FF:FF"}}

I have tried all changes like disabling //#define USE_ADC_VCC
or setting #define APP_SLEEP 250

Tested several versions of Tasmota and esp8266 core.

Any idea?

Best regards,
Karl

mike2nl, 2018-07-18:
I don't know if it's right and before someone hits me here today after my twp issue posts ;-)...

Had you done the ping with the ip or the hostname?
Can you reach the web page? a.e. http://your.ip.aadr.sonoff/cn = configuration page

What i had done, don't hit me please, in the support.ino at, please take a very good look to find that specific place:

WiFi.disconnect(true);    // Delete SDK wifi config
delay(200);
WiFi.mode(WIFI_STA);      // Disable AP mode
if (Settings.sleep) {
#ifndef ARDUINO_ESP8266_RELEASE_2_4_1     // See https://github.com/arendst/Sonoff-Tasmota/issues/2559
     WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times
#endif

I had added the two slahses '//' before WiFi.setSleepMode(WIFI_LIGHT_SLEEP)
It must look like: //WiFi.setSleepMode(WIFI_LIGHT_SLEEP);

Verify it and and upload it.
For me it was the solution no issues with ping, web server or what ever for the network functions.

But your wifi will not sleep means you need the max power. But you can test it.
Be shure where you have what changed otherwise you have to copy it all again.

nappe-karl, 2018-07-18:
I did not change anything in the Tasmota code, except wifi account.
Host is unreachable for ping with IP address from all computers but is reachable from MQTT host.
Host is visible in the network settings of the fritz!box.

nappe-karl, 2018-07-18:
Changed:
#ifndef ARDUINO_ESP8266_RELEASE_2_4_1     // See https://github.com/arendst/Sonoff-Tasmota/issues/2559 - Sleep bug
//  if (Settings.sleep) { WiFi.setSleepMode(WIFI_LIGHT_SLEEP); }  // Allow light sleep during idle times
#endif

Didn't help.

andrethomas, 2018-07-18:
Some wireless AP's these days disable client to client data - did you change anything on your router settings?

One mine its not allowed by default...

[image]

nappe-karl, 2018-07-18:
No, I didn't change anything. Not on the router (Fritz!Box 7390) and not on the Wemos or NodeMCU.
It worked for a little time (round about an hour) and never again. If you read #2216 or #2178 you can see, that I am not allone with this problem.

Jason2866, 2018-07-18:
Try (compile) a version with arduino core v.2.3.0.
Is there a reason that you want use v.2.4.1?  It is know to make problems!
For some this version works for many not...

nappe-karl, 2018-07-19:
I have already tried:
Tasmota 5.14.0, 6.1.1a, 6.1.1b, each with core 2.3.0 and 2.4.1.
No success.

nappe-karl, 2018-07-19:
Tried another Wemos D1 mini (with "new" MAC) - nothing!
"Played " with arp table - nothing!
Connected my notebook to ethernet instead of WLAN - SUCCESS!

Tasmota 5.14.0, 6.1.1a, 6.1.1b, each with core 2.3.0 or 2.4.1 each test worked like a charm.

Jason2866, 2018-07-19:
There where some posts from Fritzbox user with problems. 
Maybe this combination is a problem

Frogmore42, 2018-07-20:
Are you using name or IP address?
It sounds like you only have a problem when using wireless for both Tasmota and the computer doing the pinging.  Is that correct?

Can you ping other wireless things from you computer when it is wireless?

nappe-karl, 2018-07-21:
I always use the ip address and my other wireless things answer to ping.
Strange.

R4C3R, 2018-07-28:
OK, just found out that the ping response returns after a while. Seems like the device enters some sort of sleep mode where it does not react to pings and the webservice is unavailable.

GregS1000RR, 2018-08-21:
Karl, did you find a solution or root cause to your topic?

I face exactly the same behavior as described, except it is on sonoff devices.
After reading your comments, I also tried to ping and access my Tasmotized SonOff’s from same laptop but from a wired connection.

I just plugged the cable in, and it worked, I could access to the previously lost SonOff’s. 

Except the SonOff Touch, wich since the beginning is the most reluctant to get accessible.

But, then I had to connect to my network via VPN, and per coincidence the previously unreachable web page of the SonOff Touch, appeared like a charm.  

I read many topics about the SonOff Touch got suddenly inaccessible.
So far I can always access it, through the VPN (and hardly from wired internal)

To summarize my test config and behavior
	Router / DHCP / DNS : Fritz Box 7490
	WiFi AP : 3x Ubiquity UniFi 
	MQTT : still the test.mosquitto.org (my RPi 3B+ is almost up and running)
	SonOff 4CH Pro + Tasmota 6.1.1 => Always working, never had trouble
	SonOff 4CH 	+ Tasmota 6.1.1 => Always working, never had trouble
	SonOff RF-Bridge + Tasmota 6.1.1 => Was OK for ~4days, then disappeared from the IP screens
	SonOff Basic + Tasmota 6.1.1 => after few hours, disappeared from IP screens
	SonOff Touch + Tasmota 6.1.1 => Never realy worked, disappeared very quickly, and some times web page not accessible even after power cycle.
	WeMos D1-Mini, didn’t test long enough to observe the disparition

 All above get accessible from VPN (wired)
 Internal wired line, all except the SonOff Touch

Before someone asks, all other WiFi devices (laptops, tablets, phones, SqueezeBox radios, etc.) worked perfectly since month, if not longer. And I always ping with IP address. 

I’ll try to continue to dig into
Greg

Jason2866, 2018-08-21:
@GregS1000RR 
Try the latest development version and compile it with arduino core 2.4.2 / 1.8.0
Maybe it helps. I never had problems with any of my several devices.
I always used arduino core 2.3.0 / 1.5.0  for compiling.
Now i have changend 5 devices to 2.4.2 and these are working since several days without any issue too.

R4C3R, 2018-08-27:
@GregS1000RR I also use a Fritzbox 7490 with FRITZ!OS: 06.93 in my lan as gateway / dns.

My issues were solved with the following Tasmota version

Program Version	6.1.1
Build Date & Time	2018-07-31T17:59:09
Core/SDK Version	2_3_0/1.5.3(aec24ac9)

(tested on sonoff basic, sonoff TH16 and sonoff RF bridge 433)

GregS1000RR, 2018-08-27:
@R4C3R , Thank you for sharing, I'll give it a try and let you know.

btw, did you changed something on the Fritzbox or other WiFi Access Point ?

Somehow I'm trying to figure out, why the SonOff's reply to requests from wired connections and not from WiFi ...

ascillato2, 2018-09-04:
Hi,

Have you managed to solve your issue?

closed 2018-09-05 by ascillato2

GregS1000RR, 2018-09-09:
Hi, 

After a totally fresh, clean and dedicated dev. environment, including all latest lib's, it didn't worked out.

But then I got inspired by @andrethomas post, and tested different network configuration.

And ultimately, after the Upgrade of the UniFi AccessPoints ... It WORKS

And so far it's stable since almost 10 days, with various Devices having different FW (6.1.1.x) and those spread on various AP's, and until now, none of them get "lost"

My conclusion, is that for whatever reason, the IP packets didn't get through from a WiFi to WiFi client, and apparently (I couldn't confirm 100%) when the clients are not on same UniFi AP.

I feel much better now that all those Tasmotized ESP8266 can be reached over the WEB. 
Thank you all for your support.

Regards
Greg

andrethomas, 2018-09-09:
@GregS1000RR Glad you got it sorted - it is unfortunate that most access points default to that setting and in some cases do not even allow users to disable/enable it.

In fact, until I used Tasmota my wifi had that disabled by default and after having a few WTF moments I realized the problem and that fixed it - since then no problems.