arendst/Tasmota #3266: Power State send twice via MQTT

helmar74, 2018-07-19:
Hi,

I'm running a Wemos Mini with a WS2812 connected. The LED Strip works fine.
The MQTT Messages are send to a mosquitto broker, which is connected openhab 2.3.0.
I've got the following problem:
When I turn ON or OFF the LED Strip (with the command POWER ON or POWER OFF) I always get the Power status message vor MQTT twice:
19:55:03 MQT: stat/sonoffWemosMini_5/POWER = {"POWER":"OFF"}
19:55:03 MQT: stat/sonoffWemosMini_5/POWER = OFF

The Problem is that my openhab always throws an Exception becaus one of the Status is invalid.
Is it possible to turn of this double MQTT Messages to get only one status:
19:55:03 MQT: stat/sonoffWemosMini_5/POWER = {"POWER":"OFF"}
                         OR
19:55:03 MQT: stat/sonoffWemosMini_5/POWER = OFF

Thanks a lot

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Wemos Mini and WS2812 LED Strip
- [ ] Provide the output of command ``status 0`` :
```
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoffWemosMini-5"],"Topic":"sonoffWemosMini_5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffWemosMiniWS2812GRBWGruppe","OtaUrl":"http://192.168.xxx.xxx:8080/static/firmware/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:15","StartupUTC":"2018-07-19T08:28:56","Sleep":0,"BootCount":59,"SaveCount":344,"SaveAddress":"F5000"}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018.07.18 13:36:39","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXXX","XXXXXXXXX"],"TelePeriod":300,"SetOption":["00028019","55818000"]}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS4 = {"StatusMEM":{"ProgramSize":539,"Free":464,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000407","0FDAE794","00000000","23B6179E","00000000"]}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-5","IPAddress":"192.168.xxx.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"DC:4F:22:1D:88:E5","Webserver":2,"WifiConfig":3}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xxx.xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D88E5","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS7 = {"StatusTIM":{"UTC":"Thu Jul 19 08:57:11 2018","Local":"Thu Jul 19 3:57:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:27","Sunset":"21:03"}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS10 = {"StatusSNS":{"Time":"2018-07-19T3:57:11","BH1750":{"Illuminance":0}}}
3:57:11 MQT: stat/sonoffWemosMini_5/STATUS11 = {"StatusSTS":{"Time":"2018-07-19T3:57:11","Uptime":"0T00:28:15","Vcc":2.813,"POWER":"OFF","Dimmer":60,"Color":"51,153,54,0","HSBColor":"122,67,60","Channel":[20,60,21,0],"Scheme":6,"Width":1,"Fade":"ON","Speed":2,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXX","RSSI":54,"APMac":"CC:CE:1E:AB:0A:3F"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-07-19:
It seems you have something misconfigured 
It should look like this
`20:34:17 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"ON"}
20:34:17 MQT: stat/sonoff-13D92F/POWER = ON
20:34:23 MQT: stat/sonoff-13D92F/RESULT = {"POWER":"OFF"}
20:34:23 MQT: stat/sonoff-13D92F/POWER = OFF`

helmar74, 2018-07-20:
I configured setoption4 on to get "working" results JSON queries for the connected ws2812. (MQTT Results for setting dimmer, scheme, color, speed, fade....)

Jason2866, 2018-07-20:
SetOption 4 on is the problem!! -> You have two mqqt messages with nearly the same message.
Disable SetOption 4 and post your results. If it looks similar to mine, your OpenHab config is wrong.
Not at TASMOTA issue. In that case close this issue
You can search here https://community.openhab.org/ or ask for help  (i am active there too)
There are already some articles
https://community.openhab.org/t/light-color-rgb-rgbw-rgbcw-to-and-from-hsb-dimmer-conversion/39132/12
Configuration of Color LED (Lamp or WS2812) isnt done out of the box!

helmar74, 2018-07-20:
Not really... If you disable setption4 you get the same "RESULT" for different actions (changing color, scheme, speed, dimmer, fade...)
So you get error in openhab, because you get one JSON Result with different content...
That couldn't be parsed correctly.
I used setoption4 on and as a workaround only the "POWER" Mqtt Message raised an failure, because it was send twice.
As a workaround I used the POWER MQTT Message as a String Item in Openhab. Within a rule I parse the Message, and now i got the correct status (ON, OFF, BLINK) without an error...

Jason2866, 2018-07-20:
Yes, but my answer is correct, not an error in TASMOTA. It is a feature!
I know OpenHab a little :-) You can do nearly everything. It is just the way how.
I did only say your configuration in OpenHab isnt correct ;-)

Please close this issue.

helmar74, 2018-07-20:
Thanks a lot

closed 2018-07-20 by helmar74

Jason2866, 2018-07-20:
Here is my way how is solved that (RegEx to filter unwanted -> causes errors)
`Switch AiLight  "AiLight" <light> (gLight)
    { mqtt=">[broker:cmnd/sonoff-0CB8DA/power:command:*:default], <[broker:stat/sonoff-0CB8DA/RESULT:state:JSONPATH($.POWER):.*POWER.*]"} 
Number AiLight_Dimmer         "AiLight Dimmer"     (gLight)
    {mqtt=">[broker:cmnd/sonoff-0CB8DA/DIMMER:command:*:default],<[broker:stat/sonoff-0CB8DA/RESULT:state:JSONPATH($.Dimmer):.*Dimmer.*]"}
Color AiLight_HSB         "AiLight HSBColor"    <colorwheel> (gLight)
    {mqtt=">[broker:cmnd/sonoff-0CB8DA/HSBCOLOR:command:*:default],<[broker:stat/sonoff-0CB8DA/RESULT:state:JSONPATH($.HSBColor):.*HSBColor.*]"}`

helmar74, 2018-07-21:
Thanks for your help!
For me it's now also working with setoption4 off and the following Openhab Items:

`String i_mqtt_sonoffWemosMini_5_Power "Sonoff Wemos Mini 5 LED: [%s]"  { mqtt=">[mosquitto:cmnd/sonoffWemosMini_5/POWER:command:*:default],<[mosquitto:stat/sonoffWemosMini_5/RESULT:state:JSONPATH($.POWER):.*POWER.*]"}
Dimmer i_mqtt_sonoffWemosMini_5_Dimmer "Sonoff Wemos Mini 5 Dimmer [%.0f %%]" {mqtt=">[mosquitto:cmnd/sonoffWemosMini_5/DIMMER:command:*:default],<[mosquitto:stat/sonoffWemosMini_5/DIMMER:state:JSONPATH($.Dimmer):.*Dimmer.*]"}
Color  i_mqtt_sonoffWemosMini_5_Color "Sonoff B1 1 Color" {mqtt=">[mosquitto:cmnd/sonoffWemosMini_5/HSBCOLOR:command:*:default],<[mosquitto:stat/sonoffWemosMini_5/RESULT:state:JSONPATH($.HSBColor):.*HSBColor.*]"}
Number i_mqtt_sonoffWemosMini_5_Speed "Sonoff Wemos Mini 5 LED Speed: [%d]" {mqtt=">[mosquitto:cmnd/sonoffWemosMini_5/SPEED:command:*:default],<[mosquitto:stat/sonoffWemosMini_5/RESULT:state:JSONPATH($.Speed):.*Speed.*]"}
Number i_mqtt_sonoffWemosMini_5_Pixels   "Sonoff Wemos Mini 5 Pixels [%d]" { mqtt="<[mosquitto:stat/sonoffWemosMini_5/RESULT:state:JSONPATH($.Pixels):.*Pixels.*]"}
Switch i_mqtt_sonoffWemosMini_5_Fade "Sonoff Wemos Mini 5 Fade: [%s]"  { mqtt=">[mosquitto:cmnd/sonoffWemosMini_5/FADE:command:*:default],<[mosquitto:stat/sonoffWemosMini_5/RESULT:state:JSONPATH($.Fade):.*Fade.*]"}
Number i_mqtt_sonoffWemosMini_5_Schema "Sonoff Wemos Mini 5 Schema: [%d]" {mqtt=">[mosquitto:cmnd/sonoffWemosMini_5/SCHEME:command:*:default],<[mosquitto:stat/sonoffWemosMini_5/RESULT:state:JSONPATH($.Scheme):.*Scheme.*]"}
Number i_mqtt_sonoffWemosMini_5_Programm "Programm [MAP(ws2812programme.map):%s]"`