arendst/Tasmota #3267: Support for ARILUX AL-LC04

mrueg, 2018-07-19:
**Have you look for this feature in other issues and in the wiki?**
Yes, this is mainly for tracking progress.
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
Currently this device seems to be unsupported, other LED controller variants tried did not work.
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Working solution for AL-LC04 with IR support
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Looking into configuration and finding out the pins
**Additional context**
https://www.banggood.com/ARILUX-AL-LC04-Super-Mini-LED-WIFI-APP-Controller-Remote-Control-For-RGBW-LED-Strip-DC-9-12V-p-1060231.html
Disclaimer: I have this one to test: https://de.aliexpress.com/item/Mini-Wifi-Controller-With-24Key-Remote-Controller-For-Led-Strip-RGB-RGBW-Timing-Function-DIY-16million/32703897099.html (which seems to be just from a different vendor).

**(Please, remember to close the issue when the problem has been addressed)**

reloxx13, 2018-07-19:
Define the module yourself as generic module

https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

mrueg, 2018-07-21:
@reloxx13 thanks for the link.

Documenting my findings here:

First of all, it's not the AL-LC04, but a custom one. The AC-LC04 supports only 9-12V, while this supports 5-28V.

From https://forum.banggood.com/forum-topic-229190.html I couldn't identify the model:

```
 AL-LC01:WIFI Smart RGB Controller,Working Voltage:DC5-28V,Output Current:RGB,4A*3
AL-LC02:WIFI Smart RGBW Controller,Working Voltage:DC9-12V,Output Current:RGBW,4A*4
AL-LC03:WIFI Smart RGB Controller With 24key IR Romote,Working Voltage:DC5-28V,Output Current:RGB,4A*3
AL-LC04:WIFI Smart RGBW Controller With 24key IR Romote,Working Voltage:DC9-12V,Output Current:RGBW,4A*4
AL-LC05:WIFI Smart RGB Controller,Working Voltage:DC12-24V,Output Current:RGB,4A*3
AL-LC06:WIFI Smart RGBWW Controller,Working Voltage:DC12-24V,Output Current:RGB/WW/CW,4A*5
AL-LC07:Bluetooth Smart RGBW Controller,Working Voltage:DC12-24V,Output Current:GRBW,4A*4
AL-LC08:WIFI Smart RGBWW Controller,Working Voltage:DC5-28V,Output Current:RGBWW,4A*5
AL-LC09:WIFI Smart RGB Controller With 21key RF Remote,Working Voltage:DC9-28V,Output Current:RGB,4A*3
AL-LC10:WIFI Smart RGBW Controller With 21key RF Remote,Working Voltage:DC9-28V,Output Current:RGBW,4A*4
AL-LC11:WIFI Smart RGBWW Controller With 21key RF Remote Working Voltage:DC9-28V,Output Current:RGBWw,4A*5 
```

It looks like it's an AL-LC03 with an additional channel for warmwhite.

```
GPIO02 = Onboard LED LED1i
GPIO04 = IRRecv 
GPIO05 = RED PWM2
GPIO12 = BLUE PWM4
GPIO13 = WARMWHITE PWM1
GPIO14 = GREEN PWM3
```

I could not get the IR to work, I receive a 
``` 11:28:42 RSL: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"FFF807"}} ``` but it doesn't change the color of the LEDs. I assume I have to implement this logic on my own?

ascillato, 2018-07-23:
> I assume I have to implement this logic on my own?

Yes, you can use rules. Please, see the wiki at rules.

ascillato2, 2018-08-18:
Hi,

Closing this issue as there is no feedback. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2