arendst/Tasmota #3268: Wrong SENSOR json string on TH with v6.1.1

chriskmn, 2018-07-19:
Hello,

after flashing to 6.1.1 I experience the following issue on a sonoff TH:

With the mqtt topic set to "sonoff-x" or "sonoff-18" get a correct SENSOR message:
`
19:01:33 MQT: tele/sonoff-x/INFO1 = {"Module":"Sonoff Basic","Version":"6.1.1","FallbackTopic":"DVES_F97095","GroupTopic":"sonoffs"}
19:01:33 MQT: tele/sonoff-x/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-x-4245","IPAddress":"192.168.0.23"}
19:01:33 MQT: tele/sonoff-x/INFO3 = {"RestartReason":"Software/System restart"}
19:01:39 MQT: stat/sonoff-x/RESULT = {"POWER":"OFF"}
19:01:39 MQT: stat/sonoff-x/POWER = OFF
19:02:04 MQT: tele/sonoff-x/UPTIME = {"Time":"2018-07-19T19:02:04","Uptime":"0T00:00:38"}
19:02:05 MQT: Connected
19:02:05 MQT: tele/sonoff-x/LWT = Online (retained)
19:02:05 MQT: cmnd/sonoff-x/POWER = 
19:02:14 MQT: tele/sonoff-x/STATE = {"Time":"2018-07-19T19:02:14","Uptime":"0T00:00:48","Vcc":3.516,"POWER":"OFF","Wifi":{"AP":1,"SSId":"malibu","RSSI":52,"APMac":"C8:0E:14:AD:EE:2E"}}
**19:02:14 MQT: tele/sonoff-x/SENSOR = {"Time":"2018-07-19T19:02:14","AM2301":{"Temperature":31.4,"Humidity":35.2},"TempUnit":"C"}`**

But when I set the mqtt topic to "sonoff-8" or "sonoff-9" I get this buggy SENSOR message:

`19:10:07 MQT: tele/sonoff-9/INFO1 = {"Module":"Sonoff TH","Version":"6.1.1","FallbackTopic":"DVES_F97095","GroupTopic":"sonoffs"}
19:10:07 MQT: tele/sonoff-9/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-9-4245","IPAddress":"192.168.0.23"}
19:10:07 MQT: tele/sonoff-9/INFO3 = {"RestartReason":"Software/System restart"}
19:10:13 MQT: stat/sonoff-9/RESULT = {"POWER":"OFF"}
19:10:13 MQT: stat/sonoff-9/POWER = OFF
19:10:48 MQT: tele/sonoff-9/STATE = {"Time":"2018-07-19T19:10:48","Uptime":"0T00:00:48","Vcc":3.501,"POWER":"OFF","Wifi":{"AP":1,"SSId":"malibu","RSSI":48,"APMac":"C8:0E:14:AD:EE:2E"}}
19:10:48 MQT: stat/sonoff-9/RESULT = {"POWER":"OFF"}
19:10:48 MQT: stat/sonoff-9/POWER = OFF
**19:10:48 MQT: tele/sonoff-9/SENSOR = OFF,"AM2301":{"Temperature":31.2,"Humidity":35.2},"TempUnit":"C"}**`

It seems that a single digit number after the "-" doesn't work.

I tried the cores 2.3.0 and 2.4.1. both with same result.

BUT I am using the Tasmota-MQTT library.

Status 0:
`19:30:40 CMD: status 0
19:30:40 MQT: stat/sonoff-17/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff-x"],"Topic":"sonoff-17","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"SaveData":0,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
19:30:40 MQT: stat/sonoff-17/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.180:81/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:16","StartupUTC":"2018-07-19T18:19:24","Sleep":250,"BootCount":8,"SaveCount":8,"SaveAddress":"F4000"}}
19:30:40 MQT: stat/sonoff-17/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-18T21:48:36","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
19:30:40 MQT: stat/sonoff-17/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.0.180","LogPort":514,"SSId":["malibu","malibu"],"TelePeriod":60,"SetOption":["00008008","55818000"]}}
19:30:40 MQT: stat/sonoff-17/STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F82C880","00000000","02B4009E","00000000"]}}
19:30:40 MQT: stat/sonoff-17/STATUS5 = {"StatusNET":{"Hostname":"sonoff-17-4245","IPAddress":"192.168.0.23","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:F9:70:95","Webserver":2,"WifiConfig":2}}
19:30:40 MQT: stat/sonoff-17/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.180","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F97095","MqttUser":"mosquitto","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:30:40 MQT: stat/sonoff-17/STATUS7 = {"StatusTIM":{"UTC":"Thu Jul 19 18:30:40 2018","Local":"Thu Jul 19 19:30:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:18","Sunset":"20:29"}}
19:30:40 MQT: stat/sonoff-17/STATUS10 = {"StatusSNS":{"Time":"2018-07-19T19:30:40","AM2301":{"Temperature":30.5,"Humidity":36.0},"TempUnit":"C"}}
19:30:40 MQT: stat/sonoff-17/STATUS11 = {"StatusSTS":{"Time":"2018-07-19T19:30:40","Uptime":"0T00:11:16","Vcc":3.499,"POWER":"OFF","Wifi":{"AP":1,"SSId":"malibu","RSSI":50,"APMac":"C8:0E:14:AD:EE:2E"}}}`

chriskmn, 2018-07-19:
Just switched to the MQTT_PUBSUBCLIENT library and it works again:

19:43:56 MQT: tele/sonoff-9/SENSOR = {"Time":"2018-07-19T19:43:56","AM2301":{"Temperature":30.1,"Humidity":36.8},"TempUnit":"C"}

so there seems to be a bug in the MQTT_TASMOTAMQTT library

arendst, 2018-07-20:
Do you always receive this result with tele
```
19:10:48 MQT: tele/sonoff-9/SENSOR = OFF,"AM2301":"Temperature":31.2,"Humidity":35.2},"TempUnit":"C"}
```
or sometimes?
Does it also happen when you execute command ``status 8``

arendst, 2018-07-20:
Do you have rules defined and if so pls show them.

arendst, 2018-07-20:
Cannot reproduce...

chriskmn, 2018-07-20:
I have no rules and I get it allways with tele. I will try status 8 and report.

I also did a reset 2 and got the same result.

chriskmn, 2018-07-20:
so, flashed back to tasmota mqtt and same result:

15:54:53 MQT: tele/sonoff-9/SENSOR = OFF,"AM2301":{"Temperature":28.5,"Humidity":42.9},"TempUnit":"C"}
15:55:04 CMD: status 8
15:55:04 MQT: stat/sonoff-9/STATUS8 = {"StatusSNS":{"Time":"2018-07-20T15:55:04","AM2301":{"Temperature":28.5,"Humidity":43.0},"TempUnit":"C"}}

arendst, 2018-07-20:
And if you change topic from ``sonoff-9`` to ``sonoff-10`` it's ok?

chriskmn, 2018-07-20:
this is my user_config:

/*
  user_config.h - user specific configuration for Sonoff-Tasmota

  Copyright (C) 2018  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_

/*********************************************************************************************\
 * This file consists of TWO sections.
 *
 * SECTION 1:
 * The first section contains PARAMETERS overriding flash settings if define CFG_HOLDER is CHANGED.
 * All parameters can be persistent changed online using commands via MQTT, WebConsole or Serial.
 *   Corresponding MQTT/Serial/Console commands are shown in [brackets]
 *
 * SECTION 2:
 * The second section contains Enabled and Disabled FEATURES allowing different program size.
 *   Changes in this section do NOT need a change of define CFG_HOLDER.
 *
 * ATTENTION:
 * Users are advised to use the user_config_override.h file for most changes.
\*********************************************************************************************/

//#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md

/*********************************************************************************************\
 * SECTION 1
 * - After initial load any change here only take effect if CFG_HOLDER is changed too
\*********************************************************************************************/

// -- Master parameter control --------------------
#define CFG_HOLDER             4617                  // [Reset 1] Change this value to load SECTION1 configuration parameters to flash

//#define BE_MINIMAL                               // Minimal version if upgrade memory gets tight (-45k code, -2k mem)

// -- Project -------------------------------------
#define PROJECT                "sonoff-x"        // PROJECT is used as the default topic delimiter 
#define MODULE                 SONOFF_BASIC      // [Module] Select default model from sonoff_template.h (Should not be changed)
#define FRIENDLY_NAME          "Sonoff-x"        // [FriendlyName] Friendlyname up to 32 characters used by webpages and Alexa
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address  (=14 + Device-#)

#define SAVE_DATA              0                 // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
#define SAVE_STATE             0                 // [SetOption0] Save changed power state to Flash (0 = disable, 1 = enable)

// -- Wifi ----------------------------------------

#define WIFI_GATEWAY           "192.168.0.1"     // {IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.0.1"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "malibu"      	  // [Ssid1] Wifi SSID
#define STA_PASS1              "xxx"  // [Password1] Wifi password
#define STA_SSID2              "malibu"      	  // [Ssid2] Optional alternate AP Wifi SSID
[... 258 lines cut ...]
\*********************************************************************************************/

//#define USE_ALL_SENSORS                          // Create sonoff-allsensors with all sensors enabled (See sonoff_post.h for selected sensors)

/*********************************************************************************************\
 * Select KNX without Emulation to save space
\*********************************************************************************************/

//#define USE_KNX_NO_EMULATION                     // Create sonoff-knx with KNX but without Emulation (See sonoff_post.h)

/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade.
 *   Step 2 is re-compile with option BE_MINIMAL commented out.
\*********************************************************************************************/

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#endif  // _USER_CONFIG_H_

chriskmn, 2018-07-20:
sonoff-10:

16:03:25 MQT: tele/sonoff-10/STATE = {"Time":"2018-07-20T16:03:25","Uptime":"0T00:00:48","Vcc":3.514,"POWER":"OFF","Wifi":{"AP":1,"SSId":"malibu","RSSI":42,"APMac":"C8:0E:14:AD:EE:2E"}}
16:03:25 MQT: tele/sonoff-10/SENSOR = {"Time":"2018-07-20T16:03:25","AM2301":{"Temperature":28.7,"Humidity":41.9},"TempUnit":"C"}
16:03:48 CMD: status 8
16:03:49 MQT: stat/sonoff-10/STATUS8 = {"StatusSNS":{"Time":"2018-07-20T16:03:48","AM2301":{"Temperature":28.7,"Humidity":41.7},"TempUnit":"C"}}

arendst, 2018-07-20:
And what happens with topic ``sonoff-9`` and ``sleep 0``

chriskmn, 2018-07-20:
one sec....

this is the result with sonsoff-09:
16:05:34 MQT: tele/sonoff-09/STATE = {"Time":"2018-07-20T16:05:34","Uptime":"0T00:00:48","Vcc":3.500,"POWER":"OFF","Wifi":{"AP":1,"SSId":"malibu","RSSI":42,"APMac":"C8:0E:14:AD:EE:2E"}}
16:05:34 MQT: tele/sonoff-09/SENSOR = {"Time":"2018-07-20T16:05:34","AM2301":{"Temperature":28.7,"Humidity":41.1},"TempUnit":"C"}

chriskmn, 2018-07-20:
sleep 0:

16:07:08 MQT: tele/sonoff-9/SENSOR = OFF,"AM2301":{"Temperature":28.7,"Humidity":41.2},"TempUnit":"C"}

chriskmn, 2018-07-20:
I do change the value via web interface. do you change it via command ?

arendst, 2018-07-20:
Weird! The problem is that where you see OFF it should have the time as in a correct message. This information has nothing to do with the kind of MQTT driver that is used and I see no releation to topic name too.

This will take some time as I also cannot replicate.....

chriskmn, 2018-07-20:
that is really weird...... I'll investigate more.... changing topic with command leads to same result

chriskmn, 2018-07-20:
some variants:

16:12:25 MQT: tele/sonoff-1/SENSOR = OFF,"AM2301":{"Temperature":28.8,"Humidity":43.0},"TempUnit":"C"}

16:12:58 MQT: tele/sonoff9/SENSOR = {"Time":"2018-07-20T16:12:58","AM2301":{"Temperature":28.8,"Humidity":43.0},"TempUnit":"C"}

16:13:47 MQT: tele/9/SENSOR = {"Time":"2018-07-20T16:13:47","AM2301":{"Temperature":28.9,"Humidity":42.8},"TempUnit":"C"}

16:14:25 MQT: tele/test-9/SENSOR = {"Time":"2018-07-20T16:14:25","AM2301":{"Temperature":28.9,"Humidity":43.4},"TempUnit":"C"}

arendst, 2018-07-20:
What happens is that the string 
```
 {"Time":"2018-07-20T16:03:25","AM2301":{"Temperature":28.7,"Humidity":41.9},"TempUnit":"C"}
```
is built by different subroutines filling it with their local information. So first the time is entered, then the string is supposed to be appended by a sensor, in your case the AM2301 and finally the TempUnit is appended by a third subroutine. It seems the time data is overwritten by a power ON/OFF somewhere and then the string is still passed to the sensor subroutines....

I see currently no relation with topic naming but apparently it exists....

chriskmn, 2018-07-20:
yes and it only happens with Tasmota lib, not with EPS and Pubsub mqtt

arendst, 2018-07-25:
Did some more tests. Still cannot reproduce...
```
11:42:55 MQT: tele/sonoff-9/STATE = {"Time":"2018-07-25T11:42:55","Uptime":"0T00:08:15","Vcc":2.964,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":94,"APMac":"00:22:6B:FE:8E:20"}}
11:42:55 MQT: tele/sonoff-9/SENSOR = {"Time":"2018-07-25T11:42:55","AM2301":{"Temperature":29.2,"Humidity":49.0},"TempUnit":"C"}
11:43:55 MQT: tele/sonoff-9/STATE = {"Time":"2018-07-25T11:43:55","Uptime":"0T00:09:15","Vcc":2.962,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":94,"APMac":"00:22:6B:FE:8E:20"}}
11:43:55 MQT: tele/sonoff-9/SENSOR = {"Time":"2018-07-25T11:43:55","AM2301":{"Temperature":29.2,"Humidity":49.0},"TempUnit":"C"}
```

chriskmn, 2018-07-27:
Well, may be then it miht be related to my aruino IDE environment

fluppie, 2018-08-12:
Mine does this with 6.1.1c:

> 00:00:00 Project sonoff bme680_1_5140 (Topic bme680_1_5140, Fallback DVES_1BAB51, GroupTopic sonoffs) Version 6.1.1c-2_4_1
00:00:00 WIF: Connecting to AP1 WiFi2016AP1 in mode 11N as bme680_1_5140-2897...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on bme680_1_5140-2897.local with IP address 192.168.1.58
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/bme680_1_5140/LWT = Online (retained)
00:00:06 MQT: cmnd/bme680_1_5140/POWER = 
00:00:06 MQT: tele/bme680_1_5140/INFO1 = {"Module":"Generic","Version":"6.1.1c","FallbackTopic":"DVES_1BAB51","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/bme680_1_5140/INFO2 = {"WebServerMode":"Admin","Hostname":"bme680_1_5140-2897","IPAddress":"192.168.1.58"}
00:00:06 MQT: tele/bme680_1_5140/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: homeassistant/light/bme680_1_5140_1/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_1/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_2/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_2/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_3/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_3/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_4/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_4/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_5/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_5/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_6/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_6/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_7/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_7/config =  (retained)
00:00:06 MQT: homeassistant/light/bme680_1_5140_8/config =  (retained)
00:00:06 MQT: homeassistant/switch/bme680_1_5140_8/config =  (retained)
13:47:57 MQT: stat/bme680_1_5140/RESULT = {"Command":"Unknown"}
13:47:59 MQT: tele/bme680_1_5140/STATE = {"Time":"2018-08-12T13:47:59","Uptime":"0T00:00:14","Vcc":2.973,"Wifi":{"AP":1,"SSId":"WiFi2016AP1","RSSI":86,"APMac":"78:8A:20:5A:41:76"}}
13:47:59 MQT: tele/bme680_1_5140/SENSOR = {"Time":"2018-08-12T13:47:59","BME680":{"Temperature":25.8,"Humidity":43.6,"Pressure":1012.3,"Gas":868.30},"TempUnit":"C"}

> 

Will there be support for mqtt sensor discovery by Home Assistant? So far only switch and light work.

chriskmn, 2018-08-15:
So after evaluating a couple of things I finally found out, that there is a strong connection between the homebridge module „homebridge-mqtt-switch-tasmota“ and the „homebridge-fhem“ module of homebridge.

In this special case I declared  the temperature and humidity feature of the sonoff-th in homebridge (via config.json) and the switch feature in fhem (via fhem.cfg). So two different instances accessed the sonoff TH device.

After putting all three features (temp., hum. and switch) to homebridge (-> config.json) everything is working fine again. So no homebridgemapping in fhem.

So sharing features between fhem and homebridge seems to lead to issues.
This also explains why changing the sonoff topic seems to solve the issue because the homebridge instance then can‘t access the device anymore....

ascillato2, 2018-08-22:
Hi,

So, have you managed to solve your issue?

chriskmn, 2018-08-22:
Hi ascillato

Yes, as described I have moved the three features completely to homebridge - no mix between fhem and homebridge. And now it works fine. But I don‘t know what causes the problem....

ascillato, 2018-08-22:
ok,

Great! Good to know. Thanks for share.

So, if the issue is solved, please close it. Thanks! :+1:

closed 2018-08-22 by chriskmn