arendst/Tasmota #3270: Switch ON all timers with one request

roguestreak, 2018-07-19:
**Have you look for this feature in other issues and in the wiki?**  Yes

**Is your feature request related to a problem? Please describe.**
Following on from issue/solution #2962, "switch off all timers with one request", can/has the RE-enabling of all timers be instated?

**Describe the solution you'd like**
As per witek1308's suggestion, allow both disarming AND arming of all timers (or more accurately the timing SYSTEM) from a single 0/1 command.
User cases;

- going on holidays, disable timing system, get back, enable timing system.  No need to re-arm all timers individually

- manual bypass of timers; eg. a swimming pool pump is setup with several timers automatically operating each day, setup a rule on the manual switch that, as well as turning the pump on or off manually, toggles the timer system so it doesn't turn the pump back off or on automatically.  Auto timers for day-to-day operation, manual bypass for when you want to go for a swim or perform maintenance and not have the pump turn on or off automatically.

roguestreak, 2018-07-20:
Quick note/thought further to that last "maintenance" part; this would be a "disable" sent via mqtt central contoller before eg. pulling apart the pool filter for cleaning, and/or a dedicated button on a GPIO on the device (maybe with a dedicated LED to indicate timer enablement state).

ascillato, 2018-07-20:
So, like a master enable/disable option for Timers?

:+1:

ascillato, 2018-07-20:
Tested Ok!!!!

Great!!! 

Thanks :+1: 

( Wiki updated )

ascillato, 2018-07-20:
Hi @roguestreak 

Please, test last changes and if it also works for you, close this issue. Thanks :+1:

roguestreak, 2018-07-21:
Thanks for the quick inclusion, I haven't had a chance to try it yet but from a quick glance through the changed code I have one more suggestion for completeness which I don't think's been included; can I suggest on the web interface on the Configure Timer page having a checkbox for "enable timing system", perhaps on a line either at the very top or very bottom?

ascillato, 2018-07-21:
Hi @roguestreak

Last added features tested ok.

Please, test last changes and if it also works for you, close this issue. Thanks +1

roguestreak, 2018-07-22:
Looks like it's all there and working
I will however have to figre out a cleverer rule for disabling the timers when manually turned on, as the following one using "Power1#state" also gets triggered when a timer kicks in?  Switch1 doesn't seem to respond, any other options I'm missing?

rule2 on Power1#state=1 do Timers 0 endon on Power1#state=0 do Timers 1 endon`

ascillato, 2018-07-22:
That rule should work. So we have to see what is happening with your configuration. For that we need more information. So, for not mixing different subjects and make it easy for other people to find the issue/solution, please close this issue and open a new issue with all the troubleshooting template complete. Thanks.