arendst/Tasmota #3271: Convert dual sonoff into sonoff 4ch for blind motor?

yonaesp, 2018-07-20:
Shutter motors use only rise and fall (output phase cables).

From the tasmota configuration (module type)
I have managed to have 3 buttons (2 reals and one virtual button that activates a real button (switch)) selecting different models from mine.

What has made me think that we could emulate 2 extra buttons that activate the same switches but with the advantage of being able to configure different pulse times.

I will be very clear, in this photo i show my current configuration with a sonoff 4ch R2.
https://i.imgur.com/TmQt22K.jpg
And in this photo I show what I want to do with a dual sonoff r2
https://i.imgur.com/GHhOceS.jpg

Frogmore42, 2018-07-20:
This doesn't seem to be an issue.  None of the template is present and there is nothing clear about what is written.  Images should be inserted, not linked.

Jason2866, 2018-07-20:
You dont have enough GPIOs on a Dual for this. Pulsetime is only possible for relays.
What you want is logic with "higher level" -> Job for Home Automation
Maybe somehow possible with rules take a look in wiki
Just my 2 cents: If it works with a 4ch got get some of them.

localhost61, 2018-07-20:
@yonaesp the title of your post is confusing, in fact you restrict a CH4 R2 to 2 channels and you want to be able to do the same with a Dual R2. It's a shame to spend hardware resources to solve a software issue and yes it's achievable with a Dual R2. All the software material is available, we only need to gather it the right way.
@lobocobra is working on a [fork](https://github.com/lobocobra/sonoff-shutter) that stores the current position of the shutter and that can reach any other position with a MQTT or console command.

curzon01, 2018-07-21:
Regarding basic request from @yonaesp why not using the powerful [Rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) instead of restricted buildin PulseTime? e.g. somthing like that

`rule on switch1#state do Backlog Power1 ON; Delay 210;Power1 OFF endon on switch2#state do Backlog Power1 ON; Delay 30;Power1 OFF endon
`

Same for switch 3 and 4 using second relay (Power2)

May be you have to adapt your needs for button and switches but the principle must clear using powerfull rules.
I'm excessive using commands like `Backlog Powerx ON;Delay xxx;Powerx OFF` for position my shutters also from a home automation system (I'm using FHEM) because the Sonoff Tasmota do the exact work with the delay.

To shadow my shutter I positon it to a basic position (e.g. open) and then do the shadow position:
For example Power1 = Open and Power2 = Close and the full runtime from open to close may be 21 sec, the shadow position from open position may be 9 sec. so the "macro"

`Backlog Power1 ON;Delay 215;Power1 OFF;Power2 ON;Delay 90;Power2 OFF`

do the job. If you are using interlock mode (`SetOption14 ON`) you can remove the _Power1 OFF_:

`Backlog Power1 ON;Delay 215;Power2 ON;Delay 90;Power2 OFF`

Link such with rule and your needed trigger

ascillato2, 2018-07-24:
Hi

@yonaesp 

Have you managed to solve your issue?

ascillato2, 2018-07-28:
No feedback. Closing this issue. Please, ask for reopening if you need anything else.

closed 2018-07-28 by ascillato2