arendst/Tasmota #3272: What counter really counts?

kajmaj, 2018-07-20:
I have two counters connected to sonoff basic,
Both of them are counting pulses.
One (C1) is measuring per teleperiod and works quite well.
Second (C2) should measure number of pulses within a time period, and it is what C2 really counts, but different number is provided if I want to publish it (counter2)
Value of C2 might be anything between 0 and 500. And a change (if any) is usually very slow.
```
08:09:52 RSL: tele/sonoff_4/SENSOR = {"Time":"2018-07-20T08:09:52","COUNTER":{"C1":61,"C2":50}}
08:09:52 RUL: TELE-COUNTER#C1>0 performs "publish cmnd/co1/counter1 61"
08:09:52 RSL: cmnd/wind/counter1 = 61
08:10:00 RUL: TIME#MINUTE performs "publish cmnd/co2/counter2 490"
08:10:00 RSL: cmnd/rain/counter2 = 490
.
.
.
08:21:36 MQT: tele/sonoff_4/SENSOR = {"Time":"2018-07-20T08:21:36","COUNTER":{"C1":91,"C2":50}}
08:21:36 RUL: TELE-COUNTER#C1>0 performs "publish cmnd/co1/counter1 91"
08:21:36 MQT: cmnd/wind/counter1 = 91
08:21:36 RUL: TELE-COUNTER#C1>0 performs "counter1 0"
08:21:36 MQT: stat/sonoff_4/RESULT = {"Counter1":0}
08:22:00 RUL: TIME#MINUTE performs "publish cmnd/co2/counter2 501"
08:22:00 MQT: cmnd/rain/counter2 = 501

```
Rule for C1:
`on tele-counter#c1>0 do publish cmnd/co1/counter1 %value% endon on tele-counter#c1>0 do counter1 0 endon`
Rule for C2:
`on Time#Minute do publish cmnd/co2/counter2 %value% endon`

So C2 counts well, I expects published value as is in `tele/sonoff_4/SENSOR`  but really published is another number (it's like amount of published messages from beginning/resetting).

Is it a bug or my misunderstanding how counter works?

ascillato, 2018-07-20:
Hi

For C2,

%value% is replaced by the value contained in the trigger. So, it is giving the Time#minutes value, not the counter

For C1, as you are using as trigger the tele-counter#C1>0 it is giving you the value of the counter C1

ascillato, 2018-07-20:
You can change the teleperiod to 1 minute and set rules like:

Rule for C1:
`on tele-counter#c1>0 do publish cmnd/co1/counter1 %value% endon on tele-counter#c1>0 do counter1 0 endon`
Rules for C2:
```
on tele-counter#c2>0 do publish cmnd/co2/counter1 %value% endon
on Time#Minute=1439 do counter2 0 endon
```

(remember that the last part of your rule will only works once per day)

kajmaj, 2018-07-20:
????
So Time#minutes and clock#timer is not suitable for what I need?
And if I need to publish daily or hourly (does not matter) value of C2 and delete it afterwards? Principally the same as C1 but longer period. Value of C2 might be sometimes increased often, sometimes not (depending on pulses coming)
If it publishes under teleperiod it shows number of pulses (C2), If under Time#minutes or clock#timer it is trigger content?
Value contained in the trigger? What does it mean? Every use of trigger increase value by 1? 
I am confused.

ascillato, 2018-07-20:
> Value contained in the trigger? What does it mean?

Rules:

on [trigger] do [command] endon

if a [trigger] is time#minute, your [command] will be executed every minute
if a [trigger] is time#minute and your [command] is [command %value%], %value will be replaced by the value that your trigger has, in this case the **_minutes_**, so [command] will be [command **_minutes_**]

Another example:

if a [trigger] is tele-counter#c1>0 and your [command] is [command %value%], %value will be replaced by the value that your trigger has, in this case the **_COUNTS_**, so [command] will be [command **_COUNTS_**]

kajmaj, 2018-07-20:
If I understand well, used to publishing Time#Minute and Clock#Timer is related to the value of trigger.
How to reset this value?
Are those details described somewhere?

ascillato, 2018-07-20:
Hi,

Extracted from rules in the wiki:

```
Time#Initialized once when NTP is initialized and time is in sync
Time#Initialized>120 once when NTP is initialized and time is in sync after 02:00
Time#Set every hour when NTP makes time in sync
Time#Minute every minute
Time#Minute=241 every day once at 04:01
Clock#Timer=3 when global Timer3 is activated
Rules#Timer=1 when countdown RuleTimer1 expires
```

So, Time#Minute is the actual time of the system (minutes from midnight) and can not be reseted. The time is synced with the NTP time servers, those are not counters.

If you need to count, use a variable.

See commands VAR, ADD in the wiki.

ascillato, 2018-07-20:
your rule

on Time#Minute do publish cmnd/co2/counter2 %value% endon

is being translated as

on Time#Minute do publish cmnd/co2/counter2 TIME_MINUTES_FROM_MIDNIGHT endon

So, you are publishing minutes from midnight

------
Example:

if your rule is

on Sensor#Temperature do publish cmnd/co2/counter2 %value% endon

your rule will do

on Sensor#Temperature do publish cmnd/co2/counter2 TEMPERATURE_VALUE endon

( %value% IS the value of the trigger )

ascillato, 2018-07-20:
So,

In words. What is what do you want that C1 and C2 does?

kajmaj, 2018-07-20:
(In words) I would like to have following scheme:
C1 counts pulses per teleperiod (frequency) -> home automation calculates wind speed
 - this rule `on tele-counter#c1>0 do publish cmnd/co1/counter1 %value% endon on tele-counter#c1>0 do counter1 0 endon` does what I need
C2 continuously counts pulses , publish total daily, reset -> home automation calculates precipitation.
- so far I am able to process continuous count (it is within `tele/sonoff_4/SENSOR` ), reset daily but not publish proper C2 value once a day.

ascillato, 2018-07-20:
Ok,

So, C2 will count pulses and once per day will send its value by MQTT and reset itself. For example at midday (12 hours * 60 min/hour = 720 min).

One solution could be:

```
on tele-counter#c2>0 do var1 %value% endon
on Time#Minute=720 do counter2 0; publish cmnd/co2/counter1 %var1% endon
```

kajmaj, 2018-07-20:
Thanks for guiding me.
I will try it. But need rain first.

ascillato, 2018-07-21:
Great!

If your question has been addressed, please close this issue. Thanks :+1:

closed 2018-07-21 by ascillato2