arendst/Tasmota #3273: More than 1 FRIENDLY_NAME ?

DarkDragon23, 2018-07-20:
Hi,
is it possible to add more than one Friendly Name to the user_config_override.h ?
In my case I use a Sonoff 4CHPro and i would give the device 4 different names (Aquarium, Pumpe, Box, Ventilator).
I tried different things like "#define FRIENDLY_NAME2", but it doesn´t work.

Regards
Dark Dragon

Jason2866, 2018-07-20:
It is there
[image]

DarkDragon23, 2018-07-20:
Yes, I now.
But I would add it into the user_config_override.h in Atom, so that all the data after compiling with Atom is complete in the firmware.bin

Jason2866, 2018-07-20:
command: 
FriendlyName1 Aquarium
FriendlyName2 Pumpe

Jason2866, 2018-07-20:
Yes this is correct, not every option or command is possible to configure at compile time.
Not an error, design decision.

Frogmore42, 2018-07-20:
Since, generally, one has more than one device and Tasmota remembers its settings even when new firmware is flashed, it is a good idea to keep the changes to user_config_override.h to the minimum necessary for your location.  That would generally by:
WiFi settings 
MQTT settings
Topic prefix

Configure the rest with serial, web, or MQTT command after you have flashed each one.  I keep a list of the commands to use and cut and paste them into the command window.  This way I can see each device is working and be reasonably sure it will work when I deploy it somewhere.

Then, perhaps months later, I don't have to remember exactly which firmware I need to build to update it.  I just have one flavor to build and deploy to all devices.

ascillato2, 2018-07-24:
Hi,

Closing this issue now that has the answer https://github.com/arendst/Sonoff-Tasmota/issues/3273#issuecomment-406536315

Please, do not hesitate on ask anything else or to ask to reopen this issue. Thanks

closed 2018-07-24 by ascillato2