arendst/Tasmota #3278: How to reliably know how much time device has been ON/OFF?

demian85, 2018-07-20:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : Sonoff basic
- [ x] Provide the output of command ``status 0`` :
```
13:17:50 CMD: status 0
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff-desk-lamp"],"Topic":"sonoff-desk-lamp","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T16:51:13","StartupUTC":"2018-07-19T23:26:37","Sleep":0,"BootCount":1124,"SaveCount":1224,"SaveAddress":"F4000"}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-27T19:32:50","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garchatel WiFi",""],"TelePeriod":300,"SetOption":["00008029","55818000"]}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS5 = {"StatusNET":{"Hostname":"sonoff-desk-lamp-1472","IPAddress":"192.168.0.17","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:10:E5:C0","Webserver":2,"WifiConfig":3}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"sonoff-desk-lamp","MqttClient":"sonoff-desk-lamp","MqttUser":"saeyhxpw","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS7 = {"StatusTIM":{"UTC":"Fri Jul 20 16:17:50 2018","Local":"Fri Jul 20 13:17:50 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"01:08","Sunset":"16:44"}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS10 = {"StatusSNS":{"Time":"2018-07-20T13:17:50"}}
13:17:50 MQT: stat/sonoff-desk-lamp/STATUS11 = {"StatusSTS":{"Time":"2018-07-20T13:17:50","Uptime":"0T16:51:13","Vcc":3.197,"POWER":"ON","Wifi":{"AP":1,"SSId":"Garchatel WiFi","RSSI":84,"APMac":"CC:2D:E0:A6:11:71"}}}
```

Hello,

Can anyone please tell me how to know for sure how much time the device has been on/off? When changing state, no time is being published. All I know is the uptime, which is not useful.
I may subscribe to the stat topic, but what if my server goes down for some time? I won't know what happened in the mean time when it comes up again.

Thanks.

Frogmore42, 2018-07-21:
By default, the device doesn't keep track.  You might be able to use rules to do something, but it is not really clear what you want and over what period.

I would and do have infrastructure that stays up long enough to cover my needs, so I have it, NodeRED and emonCMS keep track.

What are you really trying to do?

demian85, 2018-07-21:
I don't want the device to keep track. I just want it to publish the time when the status changes, is it possible?
I built my own custom home-automation server and web app with Node.js and I want to come up with a better logic for turning on/off my room heater based on the time it's been on.
I could also do it by measuring outside temp and calculating diff. I don't want to turn off the heater if it's freezing outside because it takes a lot of time for it to heat (electric panel)

Frogmore42, 2018-07-21:
Still not clear what you want.  The device does publish when the status changes at the instant it changes, so you can know the time by checking the local time and remembering the time of the last change with whatever you are using to control things.

Have you seen this
https://github.com/arendst/Sonoff-Tasmota/wiki/PID-Control-with-Sonoff-Devices

ascillato, 2018-07-21:
>  I just want it to publish the time when the status changes, is it possible?

Use the syslog feature. Read the wiki at commands.

ascillato2, 2018-07-24:
Hi,

Could you manage to solve your issue?

Frogmore42, 2018-07-25:
@demian85 the device publishes when a relay changes state.  Since that publish event occurs at the moment (+ a small amount of time) that the change happened, if you subscribe to that topic you can know within a sec or so of when the state change occurred.

demian85, 2018-07-26:
I will take a look at syslog feature. 
@Frogmore42 What if my server is down at that moment? I won't know the exact time it changed state...

Frogmore42, 2018-07-26:
You need to decide which failures your larger system can tolerate and how much uptime is enough.  Getting one 9 (90%) is pretty easy, getting five 9's with a distrubted system will be incredibly hard and probably not possible with esp8266 Arduino currently.

If you are concerned about your server not being available, syslog won't help.  It requires a syslog server to be available at the moment the log message is sent.  This is no different than if the message is sent via MQTT.

Data loss and systems being down are a normal part of life.  If your application can't tolerate that, you will need to invest a LOT of time and money to create a system that does not have that.

You said you are controlling a room heater.  This is a relatively simple task.  I am not sure why you are making it so complicated.  The thermal dynamics of the room and your heater do not require a fast control system.  A simple bang-bang control would be more than adequate.

demian85, 2018-07-30:
I thought syslog feature could be used to print useful data about the device, such as "last state change time".
Okay then, I managed to accomplish my goal using a different approach.

closed 2018-07-30 by demian85

Frogmore42, 2018-07-30:
The syslog feature allows sending the information sent to the console to a syslog server.  You can query the syslog server (using its tools) to get back messages that it stored.  This means that it has to have captured the message in the first place.

Since people sometimes search the logs for answers to their questions, how did you solve the problem you had?  That might help someone else in the future.

demian85, 2018-07-30:
Instead of relying in the time the device has been on, I went for a totally
different approach by taking into account the temperature difference
between the room and outside (using weather API for now).
Thanks anyway.

Frogmore42, 2018-07-31:
Yes, that sounds like a much better approach.  If you don't want to rely on a weather API, you can deploy another device with Tasmota on it to measure the outside temperature directly.  Of course, then you have to decide where exactly outside to put it and do you want it in the shade or the sun.  Normal outside temperature is supposed to be the air temperature, without the additional gain from solar, but you might want something different.