arendst/Tasmota #3280: Artnet / Dmx für RGBW Wifi Bulbs

mode2k, 2018-07-20:
**Have you look for this feature in other issues and in the wiki?**
I found this prototype for WS2801:
https://github.com/arendst/Sonoff-Tasmota/pull/395

**Describe the solution you'd like**
Bulb as a artnet client to controll the 4 channels via dmx over artnet (udp)

ascillato, 2018-07-20:
Hi,

Why not using MQTT?

mode2k, 2018-07-20:
Because artnet is made for low overhead 25hz output to allow smooth fade. I do not think this will be possible with mqtt?!

ascillato, 2018-07-20:
Ok, So you are planning to do some effects with DMX.

That is very cool. I like that. :+1: 

But, please, be aware that Tasmota is an alternative firmware for Sonoff Devices that enables MQTT support so as to be used with home automation softwares for smart homes.

A Sonoff device with Tasmota can be controlled over MQTT, KNX, WEB, SERIAL or HTTP REST API.

So, what other feature of Tasmota do you need besides DMX? Because, for direct control of effects over UDP you will need all the power of the Esp8266 so as to not having glitches. Please, see in the Pull Request you have pointed, in the proposed change of the file _sonoff.ino_ in line 2476, there is a block to stop all Tasmota processings and just use the ARTNET/DMX mode.

I ran some experiments with the CPU_LOAD in Tasmota and I found that with this firmware the Esp8266 have some extra free processing time but that time is not constant. It can be used for processing rules and several other stuffs that do not require Real Time Processing.

So, I think that adding DMX as a polling soubroutine will make some glitches in your effects. That's why I think that for something so specific as DMX Effects, you should try an specific firmware. I search and found a lot of them here in github.

If you find any way of implementing DMX in Tasmota, so it will have no glitches and all the other Tasmota features continue working, please share it.

ascillato, 2018-07-20:
If you need a smooth fade, please read the wiki at commands. There is the fade and speed commands for that

ascillato, 2018-07-20:
Hi,

Have you solve your issue with these commands?

closed 2018-07-21 by ascillato2

arnaud-aub, 2021-04-15:
bonjour, où trouver le protocole MQTT pour SONOFF, et sur quelle pin faut-il brancher le MAX-485 ? Merci

ascillato, 2021-04-15:
@arnaud-aub 

DMX is not supported in Tasmota. You should use another firmware specific for DMX.

DMX n'est pas pris en charge dans Tasmota. Vous devez utiliser un autre firmware spécifique pour DMX. Ce référentiel Github est en anglais pour un public plus large.