arendst/Tasmota #3282: Wifi not working correctly on Electrodragon modules since Version 6

canis-ferrin, 2018-07-21:
Tested with Electrodragon SPDT module. 
Flasched ESPEasy: working.
Flasched Tasmota 5.12.0c: working.
Flashed Tasmota sonoff6.1.1c: Wifi only working for short times, mostly no ping answer.
Ping successful at approx. 10%

ascillato, 2018-07-21:
Insufficient information to help you. Please, complete all the troubleshooting template

canis-ferrin, 2018-07-21:
Where I find the all the troubleshooting template?

I found this:
"By releasing the firmware 5.12, the new ExpressIf library 2.4.0 is being used by default.
"
Possibly it is not working with the Electrodragon Boards.

ascillato, 2018-07-21:
When you have clicked on new issue button, it shows you 3 options, bug report, feature request and troubleshooting. If you click on troubleshooting, you will have a template to complete some information that is very helpful for us.

Now you can copy and paste that, and complete all the information required. That will help us a lot.

oriolism, 2018-07-22:
This wifi issue is affecting the sonoff basic too.

Frogmore42, 2018-07-22:
If you want help, you really need to provide the requested information from the template.

Having said that, the WiFi in the underlying code, not Tasmota, has issues in some environments and on some devices.  You may be using a combination that is known to be frequently bad in which case switching to something that is known to be generally mostly okay would be good.  If you are already using that, your environment/device might be particularly sensitive to the issue.

canis-ferrin, 2018-07-22:
This wifi issue is affecting the Wemos D1 mini (LOLIN) too.

Frogmore42, 2018-07-22:
WiFi is provided by the underlying libraries, not Tasmota.  Core 2.3.0 has been providing the best results for most people, but it clearly doesn't work for everyone.  There have been some cases where some of the non English binaries have had issues.  It probably won't help, but you can try using the English one.

ascillato, 2018-07-23:
Hi, please check https://github.com/arendst/Sonoff-Tasmota/issues/3289. Thanks

canis-ferrin, 2018-07-24:
Problem is solved since Tasmota 6.1.1

canis-ferrin, 2018-07-24:
but not completely, there are still several drops in WIFI connection
and bad ping times.

ascillato2, 2018-07-25:
Hi,

please, share your `status 0` output

ascillato2, 2018-08-18:
Hi,

Closing this issue as there is no feedback. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2