arendst/Tasmota #3283: MCP23xxx Config Menu not work

dr-apple, 2018-07-21:
**Describe the bug**

_A clear and concise description of what the bug is._

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues and in the wiki
- [x] Hardware used :  Wemos D1 mini MCP23017
- [ ] Provide the output of command ``status 0`` :
```
13:05:49 CMD: status 0
13:05:49 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:05:49 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:01:17","StartupUTC":"2018-07-21T12:04:32","Sleep":0,"BootCount":4,"SaveCount":8,"SaveAddress":"3F4000"}}
13:05:49 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018.07.20 20:15:54","Boot":31,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
13:05:49 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mega_Wlan_2.4GHZ_FB",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
13:05:49 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":517,"Free":504,"Heap":9,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000407","0FDAE794","00000000","F7B61E9E","00000000"]}}
13:05:49 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1885","IPAddress":"192.168.0.142","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:34:47:5D","Webserver":2,"WifiConfig":5}}
13:05:49 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.21","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_34475D","MqttUser":"drapple","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:05:49 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 21 12:05:49 2018","Local":"Sat Jul 21 13:05:49 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:09","Sunset":"20:43"}}
13:05:49 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-07-21T13:05:49","APDS9960O":{"Red":0,"Green":0,"Blue":0,"Ambient":0,"Proximity":255},"MCP23017":{"D0":0,"D1":0,"D2":0,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"D8":0,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":0,"D15":0}}}
13:05:49 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-07-21T13:05:49","Uptime":"0T00:01:17","Vcc":3.014,"Wifi":{"AP":1,"SSId":"Mega_Wlan_2.4GHZ_FB","RSSI":78,"APMac":"34:FC:B9:7C:6E:61"}}}

```

**To Reproduce**
_Steps to reproduce the behavior: The Bug is every time 

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
https://drive.google.com/file/d/1zOsbHF-EOBYXZLzqmgVo0mgn8K5h0ahD/view?usp=sharing

**Additional context**
The problem is in Safari, Firefox and Chrome

Thanks Danny

andrethomas, 2018-07-21:
Hi

The web output looks weird... can you upgrade your arduino esp8266 core to 2.4.1 and try again?

Thanks

ascillato, 2018-07-21:
changing the core will not change the webmenu.

Use 2.4.1 if you want but remember that sleep are not working with that library and also have some wifi bugs.

andrethomas, 2018-07-21:
@ascillato True - the web output just looks wrong though... something else causing that.

@dr-apple Do you get errors when configuring the device using sensor29 commands as documented for Option 1 in the Wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017

dr-apple, 2018-07-21:
Yes thank you, i have update the esp8266 core to 2.4.1 and the webmenu work fine now :-)

Can you put the function for the mcp23xx for output ?

andrethomas, 2018-07-21:
`Yes thank you, i have update the esp8266 core to 2.4.1 and the webmenu work fine now :-)`

Good - had me worried there :)

`Can you put the function for the mcp23xx for output ?`

For now output is not supported in development branch - I am awaiting response from @arenst if I can add it as an optional define. The current test code for the output capable version you can replace the xsns_29_mcp230xx.ino file from my test branch at https://github.com/andrethomas/Sonoff-Tasmota/tree/mcp230xx_output

Just note for output there is no interrupt and you need to use sensor29 commands to turn a pin on and off

To configure pin 0 do in web interface or command:
`sensor29 0,5,0`

To turn pin 0 on do command:
`sensor29 0,ON`

To turn pin 0 off do command:
`sensor29 0,OFF`

Thanks for testing, appreciated.

arendst, 2018-07-21:
The wrong webpage is typical for lack of RAM. That's why I had to make a lot of changes to get the module webpage as small as possible removing redundancy and using javascript to render the page on the client side.

Considering the amount of redundant field entries for the MCP I was expecting this kind of behavior.

@andrethomas to solve this issue for 2.3.0 and future memory degradation I would strongly suggest you start removing rudundancy from your MCP webpage and move it to the client side using javascript.

andrethomas, 2018-07-21:
@arendst 

> The wrong webpage is typical for lack of RAM. That's why I had to make a lot of changes to get the module webpage as small as possible removing redundancy and using javascript to render the page on the client side.

Did cross my mind when I saw some replace statements in the web page which was displayed...

> Considering the amount of redundant field entries for the MCP I was expecting this kind of behavior.
>
> @andrethomas to solve this issue for 2.3.0 and future memory degradation I would strongly [...]

Or move away completely from the web config option and only use sensor29 commands for configuration - it's easy enough. I'm confident that we can remove the web interface given that everything can be done using the sensor29 command - if you really want I can move to javascript but I've not done this before - do you have an example of one of them? That is, if we want to keep web config option.

arendst, 2018-07-21:
Using the javascript option is messy! Example is my module configuration webpage.

I prefer NOT to use a webpage. While users love it it's a pain to provide.

andrethomas, 2018-07-21:
@arenst HandleMCP230xxConfiguration = 8817 bytes of RAM :| That's too much

dr-apple, 2018-07-21:
@andrethomas thank you for testcode :-) I tested in the next days....

andrethomas, 2018-07-21:
@dr-apple No problem - In light of above discussion please do not use web config define - rather configure using sensor29 commands - I believe web config option will be deprecated in favour of sensor29 commands.

Please close this issue for now.

Thanks

dr-apple, 2018-07-21:
Ok thank for your help !

closed 2018-07-21 by dr-apple