arendst/Tasmota #3286: Wemos D1 mini, VCC read too low

Claus55, 2018-07-21:
Please bear with me if I overlooked something, but I'm still quite new to Tasmota.
I have a couple of ESP8266 (ESP-12F) which I've flashed with Tasmota sonoff.bin respectively version 5.14.0 and 6.1.1, and when I issue the cmd. status 11 they all report the VCC to be around 3.3 volts.
After purchasing a D1 Mini NodeMcu 4M bytes Lua WIFI Development Boards ESP8266 and flashing it with 6.1.1., I tried the same cmd. (status 11), but it always reports a voltage around 2.8 volts.
(5.14.0 also tested with same result).
Am I missing something or is it working as designed?
/Claus

ascillato, 2018-07-21:
It is Ok. That is the voltage in wemos and in nodemcu.

No issue. It works that way.

If your question has been addressed, please, close this issue. Thanks :+1:

Claus55, 2018-07-21:
I thought it was me, who should close the issue when I felt my question was answered.
But ascilato thinks otherwise! :-|
Isn't there no way to get it to report the VCC correct? Via a shield?
Doesn't any the Wemos D1 boards report the VCC correct?
/Claus

ascillato2, 2018-07-21:
Hi,

Sorry. If you think you need more help, you are able to reopen the issue if you want. No problem at all.

Tasmota can not do anything to the VCC value of the power regulator of the WEMOS nor NODEMCU. Sorry.

Claus55, 2018-07-21:
you cannot re-open your own issues if a repo collaborator closed them.

ascillato2, 2018-07-21:
Reopened as asked.

What else we can do for solve your issue?

Claus55, 2018-07-21:
Isn't there no way to get it to report the VCC correct? 
Via a shield? 
Doesn't any the Wemos D1 boards report the VCC correct?
/Claus

ascillato, 2018-07-21:
No,

The Hardware was made like that.

ascillato, 2018-07-21:
That reading is not a Tasmota Software Issue.

Please, remember that Tasmota is mainly aimed for Sonoff devices to be integrated in Home Automation.

Also, please do not get upset when closing issues. This Github issue tracker is for tracking Tasmota software bugs and for feature requests. If there are many issues opened, we have an extra difficulty to follow all the issues and solve the bugs as there are a lot of issues being left behind by their authors or stuck in troubleshooting.

If an issue is closed and you think you need more time, more attention, etc. Just ask to be reopen. No problem at all. The issues are for track the problem.

What else we can do for you?

Claus55, 2018-07-21:
I'll modify the board similar to: https://arduinodiy.wordpress.com/2016/12/25/monitoring-lipo-battery-voltage-with-wemos-d1-minibattery-shield-and-thingspeak/
Therefore I ask about shield!

I never claimed it to be a Tasmota bug!
I only aks some questions, and the shield part was not answered.
To me it looked like you were more interested in closing that answering.
Have a nice day.
Claus

closed 2018-07-21 by Claus55