arendst/Tasmota #3288: Sonoff 4CH Pro, web server unavailable.

noisufnoc, 2018-07-21:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff 4CH Pro 
- [x] Provide the output of command ``status 0`` :

I'm uploading my known good sonoff basic configs, that i've used on numerous basic devices before.  I can get the arduino console to show some status, which for whatever reason i cannot copy/paste here, but when i try to visit the ip address in the logs the web ui never loads.  are there extra steps i need to take for the 4ch pro model over a basic?

noisufnoc, 2018-07-21:
I'm also using the latest release, uploaded to the sonoff via the Arduino ide on Linux.

meingraham, 2018-07-21:
FYI... I compile my own 6.1.1b and flashed four 4CH Pro (R1) yesterday. I am able to connect to the admin web pages and configure w/o issue. I know this isn't much help, but I wanted to give a data point that your issue is not associated with all 4CH Pro devices.

Have you tried connecting to the web server from another PC or mobile device?

Question though - is yours a newer R2 or the original 4CH Pro? Mine are all rev. 1.

Mike

noisufnoc, 2018-07-22:
Yes, this is a rev 1 board.  i'm going to give it another try tonight and see what i come up with.  Did you leave the module defined in user_config.h as SONOFF_BASIC and then change later in the web ui?

noisufnoc, 2018-07-22:
Okay, I'm getting closer... I changed the module manually in user_config.h to SONOFF_4CHPRO and was able to get it on the wifi...

```
04:40:50 CMD: status 0
04:40:50 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
04:40:50 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:03:09","StartupUTC":"2018-07-22T03:37:41","Sleep":0,"BootCount":20,"SaveCount":106,"SaveAddress":"FA000"}}
04:40:50 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-21T23:14:39","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
04:40:50 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"10.0.0.61","LogPort":514,"SSId":["technodrome",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
04:40:50 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":542,"Free":460,"Heap":3,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
04:40:50 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0343","IPAddress":"10.0.0.197","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.61","Mac":"60:01:94:AC:E1:57","Webserver":2,"WifiConfig":3}}

```

I can toggle 2 channels at a time, while powered via AC.  I cannot, however, view the module configuration page $IPADDRESS/md?, which just displays a blank page.

noisufnoc, 2018-07-22:
I also changed the dip switches per:

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro#4ch-pro-dipswitch-configuration

right now, pings are intermittent...seems like it keeps falling off the wifi or something

meingraham, 2018-07-22:
Yes, I left the module defined as Basic in the user_config and changed it via the web interface after it booted on the new firmware.

I am running my 4CH Pro devices in self-locking (K5 1111, S6 1) mode. I would be floored if these settings have any impact on any part of how the firmware behaves, especially Wi-Fi or the web admin.

Mike

noisufnoc, 2018-07-22:
I thought about it last night, and just for grins I flashed an older version of Tasmota that I had laying around on my machine.  I have my 4CH Pro up and running on v 4.1.3, which doesn't have the correct module setting for the Pro.  This old version was compiled with 1M (64K SPIFFS) so I can't OTA a newer version.  I'm going to keep tinkering and find the most recent version that works for me.  I'm suspecting that I'm running into the Wifi Issue that I've been reading on versions >6.X

meingraham, 2018-07-22:
The fact that setting the module type allows you to get it on Wi-Fi is strange though. Going back to the version that does not allow you to connect to the web admin, if you look in your Wi-Fi router, does the device show up there? If so, can you see MQTT traffic - e.g., if you invoke a command such as:

`mosquitto_sub -v -u {broker username} -P '{broker password}' -t +/{4CHSonoff Topic}/#`

And yes, I was having problems with TASMOTA 6.1.x and ended up reverting to the ESP 2.3.0 core library. I use a Sonoff Basic for testing new firmware (only costs me $5 if I brick it <grin>). Once I got 6.1.1x stable operation (i.e., using 2.3.0), I haven't had any issues since. Then I upgraded my 4CH Pro devices... and also have stable behavior as well.

noisufnoc, 2018-07-23:
switched to ESP 2.3.0 and re-flashed 6.1.1 just now

```
02:10:06 CMD: status 0
02:10:06 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
02:10:06 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:41","StartupUTC":"2018-07-23T01:07:25","Sleep":0,"BootCount":27,"SaveCount":135,"SaveAddress":"F5000"}}
02:10:06 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-22T21:07:46","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
02:10:06 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"10.0.0.61","LogPort":514,"SSId":["technodrome",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
02:10:06 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
02:10:06 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0343","IPAddress":"10.0.0.197","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.61","Mac":"60:01:94:AC:E1:57","Webserver":2,"WifiConfig":3}}
02:10:06 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.noisufnoc.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ACE157","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:10:06 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 23 01:10:06 2018","Local":"Mon Jul 23 02:10:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:12","Sunset":"20:41"}}
02:10:06 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-07-23T02:10:06"}}
02:10:06 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-07-23T02:10:06","Uptime":"0T00:02:41","Vcc":3.186,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"technodrome","RSSI":74,"APMac":"AC:A3:1E:AE:F6:A1"}}}
```

Seems happier...going to test more.  The configuration page works w/o issue, that's the first thing I noticed.

noisufnoc, 2018-07-23:
Just switched it to mains power rather than my FTDI adapter, web server still responding and working as expected.  I'm going to play for a few more days but I think that was the issue.  Thanks @meingraham for your help, my landscape lighting project was saved!

ascillato2, 2018-07-24:
Hi,

Closing this issue. Please, if you think is necessary, just ask and we can reopen this issue. Thanks

closed 2018-07-24 by ascillato2