arendst/Tasmota #3290: Sonoff basic switch turns off and on again after WiFi restart

demian85, 2018-07-21:
Hello,

I commented about this in other issues but didn't get any response.
The issue is that when a device is ON and the router restarts or WiFi config changes somehow, the device will turn off and on again in matter of milliseconds. 
I cannot make sense of that behavior. If the device never loses power, why would it toggle state in that way? Is there a config value to fix that?

Thanks.

- [ ] Searched the problem in issues and in the wiki
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
11:01:11 CMD: status 0
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff-desk-lamp"],"Topic":"sonoff-desk-lamp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T14:34:34","StartupUTC":"2018-07-19T23:26:37","Sleep":0,"BootCount":1124,"SaveCount":1227,"SaveAddress":"F9000"}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-27T19:32:50","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garchatel WiFi",""],"TelePeriod":300,"SetOption":["00008029","55818000"]}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS5 = {"StatusNET":{"Hostname":"sonoff-desk-lamp-1472","IPAddress":"192.168.0.17","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:10:E5:C0","Webserver":2,"WifiConfig":3}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"sonoff-desk-lamp","MqttClient":"sonoff-desk-lamp","MqttUser":"saeyhxpw","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 21 14:01:11 2018","Local":"Sat Jul 21 11:01:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"01:09","Sunset":"16:43"}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS10 = {"StatusSNS":{"Time":"2018-07-21T11:01:11"}}
11:01:11 MQT: stat/sonoff-desk-lamp/STATUS11 = {"StatusSTS":{"Time":"2018-07-21T11:01:11","Uptime":"1T14:34:34","Vcc":3.196,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Garchatel WiFi","RSSI":92,"APMac":"CC:2D:E0:A6:11:71"}}}
```

ascillato, 2018-07-21:
Hi. Sorry. There is no solution by now. That very anoying blink is because of the esp8266 restart. On the mcu boot up, it turn off some pins.

demian85, 2018-07-21:
Okay, but... if power is never lost, why is the device booting up again? it's already up! it just needs to connect to the WiFi again! I don't get it 😕

arendst, 2018-07-21:
Change wificonfig to 5 to stop restart.

demian85, 2018-07-21:
Great. thanks!

closed 2018-07-21 by demian85