arendst/Tasmota #3292: Using a separate 3.3v power supply to flash the iFan02

finity69x2, 2018-07-21:
I see that is necessary to supply AC main power to the iFan02 in order to provide enough power to the chip to flash it correctly.

Would it be feasible to use a higher rated external 3.3vdc power supply to the 3.3v and GND pins on J3 while the UART is connected to the Tx & Rx pins? Would it be necessary to connect the ground pins of both the UART and the power supply together for it to work correctly?

I was thinking of using a 19vdc @ 1.58A power supply regulated down to 3.3v with a buck converter regulator.

Kind of like this:

[image]

Or am I completely off base here and it won't work at all? I figured that it would be a safer way to flash it if it worked.

Frogmore42, 2018-07-22:
Yes, that should work fine.  I use a buck converter on the 5V from USB, but I have USB ports that can supply 2A.  A linear buck converter might get hot  dropping from 19V so using something closer to 5V would be better.

finity69x2, 2018-07-22:
I'll look to see if I have something closer to 5v that puts out that much current but off the top of my head I don't think I do. thanks for the tip!.

OTOH, I'm using the same buck converter dropping from 24v to 5v supplying a NodeMCU and so far there haven't been any negative outcomes. I'll try it and see how it goes.

I'm glad to know I'm on the right track.

Frogmore42, 2018-07-22:
If it is a switching buck converter instead of a linear one that will be fine generally.

Frogmore42, 2018-07-22:
For a linear regulator, just use a USB charger that can supply an amp or more.  They also work for the switching regulators.

finity69x2, 2018-07-22:
It's a MP1584EN switching converter.

So far I've got a prototyping board soldered/wired up and can communicate via the serial monitor on the ArduinoIDE with a sonoff basic I had laying around.

And I've got the iFan02 powered up with my "programming device" and I see gibberish on the ArduinoIDE serial interface and the little blue LED inside the can is flashing about once a second. Hopefully that's normal.

If so then I guess I'm on to trying to flash it!

digiblur, 2018-07-24:
The ifan02 module I just got in worked with the SonOTA process to flash it. Kinda shocked it worked. Flashed the latest 6.1 I compiled a few days ago after the process and I did a reset config to start fresh and all is working from what I could tell. Haven't hooked it up to the fan yet. Remote tested as well.

finity69x2, 2018-07-24:
How do you program the remote? Do you have to program it before you flash it or can you do it afterward?

ascillato2, 2018-08-29:
Hi,

Have you managed to solve your issue?

finity69x2, 2018-08-30:
@ascillato 

Which issue?

If it's the flashing part, yes, I just used the set up that I posted above with the separate power supply. I did modify one part tho. I used a 2.4a 5VDC cell phone power supply instead of the 19VDC one. It worked perfectly.

If it's the remote pairing part, yes, you pair it exactly the way it says in the instructions that comes with the unit.

See the other support thread for further information: 

https://github.com/arendst/Sonoff-Tasmota/issues/2839

ascillato, 2018-08-30:
Thanks a lot for the summary. Very appreciated.

Please, can you add this useful information to the wiki?

ascillato2, 2018-08-31:
Thanks everyone for sharing this information and added it to the wiki.

closed 2018-08-31 by ascillato2