arendst/Tasmota #3293: Cant flash Sonoff 1 Channel 5/12v

Againstyou, 2018-07-22:
Hello,
today i try to flash my Sonoff 1 Channel 5/12v. I found many guides on the internet and when i flash it the first time it looks like it work. (got 100%). But the relay was constantly switching after this. So i flashed it twice again and had the same issue. After that i realised that i was flashing in DIO mode instead of DOUT. So i changed to DOUT and try to flash again. Since this change i cant flash anymore.

The Sonoff just blinking and switching relay every second after bootup, when i hit button 2 the relay stop switching and 2 LED turns on.  (When the internet is right this is the flashmode?)

When i flash first i dont enter any special flash mode, but it send all data anyway.

I try the everything i found on the internet:

- Change to ESP8285
- Try all Baudrates
- Switch TX and RX
- Connect GPIO 0 to GND
- Different Flash Sizes
- Power up via 3.3v onboard
- Power up via USB Port

I think i've bricked my device. Is there anything i can do to flash tasmota properly?

Thanks for you help in advance!

Frogmore42, 2018-07-22:
There is a difference between flashing and the device working after it has been flashed.  I get that you are frustrated with the experience, but you have provided none of the necessary information from the template to easily help you.

First you can't brick an esp8266 by flashing it.  It has a fail safe bootloader that will always allow you to flash it again.  Having said that, there are (probably many) ways that you can break/damage an esp8266.

Tasmota provides lots of useful information about what it thinks is happening over the serial port.  Read the tutorials on how to see that.  If you can flash it, you should be able to figure it out easily.  When you power it up with the serial port connected to a terminal emulator, you should see debug information that will tell you what is going on.

When you attempt to flash what happens?  If there was no error listed, it is likely that the flashing part succeeded.  If it didn't, the error and when the error occurred will give a clue as to what the problem might be.

But first see if you can see anything on the serial port.

Againstyou, 2018-07-22:
Oh sorry, i had added more infos, but delete accidentally.

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Do you mean this? Or another log? This is the log in Arduino IDE.
This error is 90% of the time. Sometimes i get another error when i play around with different tools.
I wonder that i can flash it without bringing it into any flash mode before and now i cant flash it anymore, even with the old DIO Setup.

Frogmore42, 2018-07-22:
That says the flashing is not communicating with the esp8266.  That is usually for one or more of many reasons.
It doesn't have power.
It is not in flashing mode.
It is not connected correctly.

Arduino has a serial port viewer.  It has been years since I have used it so I don't know what they call it.  There should be basic tutorials on how to see Serial print output.  Find one and see what you can see.

Frogmore42, 2018-07-22:
https://www.instructables.com/id/HOW-TO-use-the-ARDUINO-SERIAL-MONITOR/

Tasmota uses 115200 baud

Frogmore42, 2018-07-22:
If you can get it to print output, that means you have at least one of the serial lines connected correctly, which means the other one probably is too.  It also means it is getting some power and you have the correct port.  That just leaves, you probably haven't enabled bootloader mode.  You have to hold GPIO 0 low, find the right button to press for this.  Then while holding the button apply power to the esp8266.  Keep holding the button for at least three seconds, but you can keep it pressed longer.  Then you can try flashing again.

Againstyou, 2018-07-22:
Unfortunately i cant copy the Serial Output, but there a just strange signs. Some mirrored "?" and a arrow.
Again, while i flashed the first time i dont press any buttons or connect Gpio0 to gnd and it works. After that the relay is clicking constantly after boot up. When i press the 2nd button the relay stop clicking and the LEDs shine constantly. This should be the flashing mode. So it looks like its boot in flashmode anyway because i get the same result no matter if i connect gpio0 to gnd or press both buttons while start up. Its allways the same: clicking relay and when i press btn2 it stops and shine constantly.

I have the bigger 1Channel 5/12V not 5V but this video shows how the Sonoff act. But like i said, it act like these everytime when i boot up no matter if connect any pins or press any buttons.

Maybe 2 more things to help troubleshooting:
When i power the Sonoff with the FTDI so Relay LED flashing but the relay dont clicking, when i use USB power the LED flashing and the relay flashing. I think 3.3 v is to low to switch the relay?

And while i try to flash, i once "brigde" or "connect" a TX or RX pin accidentally with some part of the the Sonoff board and the sonoff turns off and on instantly. And because i have to connect TX and RX to the chip directly it may happend that i touched other pins than RX and TX at the ESP. But i have to say that it dont work too before i hit any "wrong connections" on the sonoff. But i fear that i destroy my device. :(

Thanks for your help so far.

Frogmore42, 2018-07-22:
You HAVE to connect GPIO 0 go ground to put it into flashing mode.  There is no other way to do it on an esp8266.  If you are getting strange characters that means the baud rate is not correct.  If see some then it means that you have at least figured out how to get it power and how to hook up one of the serial lines.

As far as cutting and pasting goes, what are you using?  Cutting and pasting text is a basic skill.  If you don't know how to do that, I'm not sure how successful you will be with Tasmota and it might be best to stick with the default firmware.

Pictures of your setup would probably be a good idea if you want to continue.  Flashing an esp8266 is not especially hard once you understand ALL of the things that you have to correctly.  Until then, it can be very frustrating.

Againstyou, 2018-07-22:
I have flashed several ESP8266 yet with tasmota. So it isnt my first time. Anyway somehow i got it worked right now. I dont know why, i try it like 60 times in different setups and these time it worked, but i tried the same method like 6 times before. I connect GPIO0 to GND on FTDI, connecting FDTI GND to Sonoff, power the Sonoff with micro usb instead of using the 3.3v FTDI output and try to flash and it works...like i said i try these and other ways so often today and yesterday i really dont know what the problem is.

Because the pasting issue: it looks like not possible to copy text out of the Terminal from Arduino, you can mark the text, but rightclick and/or cmd+c for copy dont work. Sure i can screenshot the log, but it was only inverted "?" , so nothing interested.

Thanks for your time and help anyway. :)

closed 2018-07-22 by Againstyou

Frogmore42, 2018-07-22:
The problem is that the FTDI is not rated to supple enough power to flash an esp8266.  Reason number one from my list.

That says the flashing is not communicating with the esp8266. That is usually for one or more of many reasons.
 It doesn't have power.
 It is not in flashing mode.
 It is not connected correctly.

Use a terminal emulator which supports copy.  I use Teraterm
http://ttssh2.osdn.jp/

Againstyou, 2018-07-22:
Yes that the ESP dont get enough power was an early idea i had. Thats why im trying to support it with micro-usb in 50% of my tests. But like i said, try it this way like 7-8 times in different setups and it never works...i dont know why it works this time.