arendst/Tasmota #3294: Sonoff stop searching for Wi-Fi:

shayshtern, 2018-07-22:
Every time my Insternet get lost or My router gets off and I get esp Wi-Fi for every device I have that anyone can login to it. How can I change it to have password to that configuration page or just set the Sonoff `OFF` and not search for Wi-Fi at all until u press sonoff manually? 
Thanks 💪🏾

Frogmore42, 2018-07-22:
Read the wiki, search for commands
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi

Wificonfig 5

Is probably what you want.

ascillato2, 2018-07-22:
Hi,

If your question has been answered, please close this issue. Thanks :+1:

ascillato2, 2018-07-24:
Hi,

Closing the issue. Please, if you still have problems, just give us some feedback and we can reopen your issue if you want. Thanks.

closed 2018-07-24 by ascillato2