arendst/Tasmota #3297: S20 gone for good after night

SSkull995, 2018-07-22:
Hi everybody, I am opening a new issue hoping that someone could help. Yesterday I connected my sonoff s20 to the raspberry homebridge and everything was perfect. I switched it on/off couple of times from the iPhone and it was working.
This morning I woke up and tried to switch it up, no response. I went to my router ip and I found out it was not even connected to the router. I looked at it, it is under my table, and found out it was off, no led, nothing.
I tried to connect it via ftdi, no response but my Mac restarted due to a problem twice, as soon as I connected the gnd to the sonoff.
I really don't understand what is going on, I would try to connect to the AC and measure the different volts but I don't know where to look on the board. May it be the little fuse on the board that is gone?
I connected it to my 3D printer and I don't think it takes more that 2200Watt of power per second ahah.
To anyone who answers a big thank you for your time, really appreciated.

Ah, no smell of burnt things on the board.

Frogmore42, 2018-07-22:
If you don't know what you are doing, working on a Sonoff connected to the AC line is a recipe for "horror and inconvenience" and possibly death.

Use the method of powering and connection that you used to flash it and see if it is powering up and you can see any activity on the serial lines.  If not it is dead and you need a new one, probably.

SSkull995, 2018-07-22:
Actually I do know electric things but I just don't know how to investigate on it. Anyway, I tried with the cables w/o results. If I connect the gnd to one of the leds, it switches up, otherwise no. Honestly, don't know why, yesterday it was totally fine.

Frogmore42, 2018-07-22:
Apply 3V3 to the VCC pin and connect the gnd pin.  Use a DMM to measure on the VCC pin on the chip and gnd on the chip (esp8266).  Use a logic probe to see if there is any activity on any of the pins.  Use the current meter function of your DMM to measure the current in the VCC line.  These are some of the things that I would do to investigate the issue.

You said you provide the circuit with power.  Verify the power is getting to the chip.  If it is and there is no other activity, the chip is dead.  That can happen.  I have found most electronic stuff to be remarkably tolerant of bad things, but sometimes really bad things happen or parts are just bad.

SSkull995, 2018-07-22:
first of all thanks @Frogmore42. Before putting controlling the voltage, I decided to investigate the output on the serial monitor. Shitty as expected. 
ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3380
~ld

Exception (9):
epc1=0x40256ff1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000030b depc=0x00000000

ctx: sys
sp: 3ffff830 end: 3fffffb0 offset: 01a0

This is a part of the code, the only normal one. The other part is like the last line (3ffff...) many many many things, so it seems to work but not to start

SSkull995, 2018-07-22:
ok I checked the esp8266 no power at all. I check the flash chip (the one just on the right to the esp) and the current arrives from the bottom but does not go out from the top of the chip. It seems that the chip itself blocks the current

SSkull995, 2018-07-22:
I can flash it, I am re-flashing it right now

Jason2866, 2018-07-22:
Do a full erase with esptool

SSkull995, 2018-07-22:
ok I got it. I changed the firmware to the 5.x.x and now it works without problems. Many thanks for the replies!

SSkull995, 2018-07-23:
ok no I was too happy. I managed to erase the chip, I changed the version to 5.14, I flashed again the son off. All good.
Now, when I power up the s20, it does nothing, no led but just the same code as above spitted in the serial monitor.

Jason2866, 2018-07-23:
Sometimes hardware dies... It looks like

Frogmore42, 2018-07-24:
I would check the 3V3 regulator.  I have a couple of devices that I believe have bad ones.  I haven't taken the time to replace the regulator because I have enough spare devices.  These things are really inexpensive even if they have a failure rate of 20%.  I just buy a replacement one and that takes less time/money than it would to try and repair one.  If I get really bored, I might try repairing one for fun.

ascillato2, 2018-07-25:
Hi,

Have you managed to solve your issue?

SSkull995, 2018-07-26:
Hi @ascillato2 I solved it and I am sorry but I forgot to update the post. I uploaded the 5.14 using the esptool to erase and flash it.
I really don't know what happened using atom, but esptool saved me. Thanks to all of you!

closed 2018-07-26 by SSkull995