arendst/Tasmota #3298: Send WOL packet to MQTT broker host on failed MQTT client connecion attempt

ivanpavlina, 2018-07-22:
It would be nice if sensors could wake the host up after power failure.
Example of code to send magic packet is here https://github.com/snargledorf/WakeOnLan-Arduino/blob/master/WakeOnLan.cpp

Could this be implemented?

arendst, 2018-07-23:
Why on earth would you have a crappy MQTT environment needing WOL to wake the MQTT broker? The MQTT broker should be up 24/7.

Sorry, won't implement.

ivanpavlina, 2018-07-23:
Well, okay.. It would be nice for me because I have some problems with AC network, so on power failure my UPS is scheduling shutdown of MQTT broker, but i need something that will trigger WOL when the  power is restored (UPS can't do that unfortunately).
Nevermind, I'm closing this request, best regards.

closed 2018-07-23 by ivanpavlina