arendst/Tasmota #3299: A bunch of sonoff T1 to drive relays

FabrizioRomanoGenovese, 2018-07-22:
Hello everyone, 

First of all thanks for everything, this project is fantastic and the community is amazing.

# Desiderata
I have some lights that have to be controlled by different switches. All my switches are sonoff T1s. What I want to achieve is:
- Having all these sonoffs to control a relay that switches the light on and off, so that the light can be physically controlled by many different points;
- I want the sonoff to consistenly inform MQTT about this in a nice way, e.g. by sending a `toggle` message;
- I want to be able to trigger other events by pressing a button on the sonoffs for, say, n seconds.

# Current setup
I have defined `pulsetime 2` on each button (so for instance `pulsetime1 2`, `pulsetime2 2`, `pulsetime3 2` to have it on all the gangs of a T1 3CH). This allows me to touch the button and send a brief signal to the relay to toggle the lightbulb. This solves desiderata 1.

With respect to points 2-3, at the moment every time I touch a button on the sonoff MQTT receives two messages, a `POWER ON` when the internal relay of the T1 turns on and a `POWER OFF` shortly after when it switches back off.

# What I want
I'd like to have a couple of things:
- When a button is pressed I'd like MQTT to see only a "toggle" message. I read around that this can be achieved by fiddling with `switchmode` but I don't know how to tell T1 that I want to change the switchmode of a physical button. I read also that this may involve changing the firmware code, or fiddling with GPIO in a way or another, but these are things that I don't really understand. Any coherent, detailed explanation would be appreciated. Also, since I am flashing my stuff with `sonOTA` I'd like to know if there is a simple way to achieve this.
- I'd like to implement an hold function: If a button on the T1 is pressed for more than x seconds, I'd like to have a command sent out. This may be useful to trigger scenes and the like. This seems again to be possible by fiddling with `switchmode` (setting it to `5`, for instance), but then again I don't know how to apply `switchmode` to buttons. Surely `switchmode1 5` didn't work.

Sorry if these requests seem trivial, I just started looking into this recently.

Frogmore42, 2018-07-22:
This is not really an issue with Tasmota.  The requests are not trivial.  They are asking to explain things that will be very difficult to do to someone who does not understand the most basic things of the environment.  Spend some time reading the Wiki.  You should be able to experiment with sending commands and getting responses.  You have discovered switchmode.  Have you tried to use it?
Have you figured out how to send commands?
There are at least three ways to send commands and see responses.  You can use serial, web console, web commands, MQTT.  You need to be proficient in at least two methods.  By proficient I mean you have done it more than once and if asked to do it again a week later you can remember (or have written down the method and can find, read, understand) how to do it again.

You also need to fill in all the information of the template.

FabrizioRomanoGenovese, 2018-07-22:
Hey @Frogmore42, 

First of all thanks for your time, I appreciate it.

I read the wiki quite a lot, but I have to admit that it is not an easy thing to navigate.
As I said in my first post, I am already able to send and receive MQTT messages to/from the T1, so yes, I am able to send commands and get responses. 
I am able to send/receive commands both via the T1 web console or via another MQTT client subscribing to the same topic.

I clearly tried to use `switchmode`. If you read my first post carefully, I already said "Surely `switchmode1 5` didn't work". I have read around in the issues that `switchmode` doesn't really apply to buttons, but to switches. What I am looking for is exactly what are the relevant commands (hopefully to be given via MQTT, or in any way without having to manually recompile the code and flash it), to enforce `switchmode` on buttons, that is, to force tasmota to consider the physical buttons on the T1 as switches as well (if this makes sense).

Also, notice that in the event of enforcing `switchmode` on buttons is achievable all my desiderata become trivial, since `switchmode 5` does exactly what I want. At that point the desired behavior when pressing and holding a button can be achieved by a combination of `switchmode` and `switchtopic`.

I admit that understanding serial has been the most difficult thing for me up to now. Probably I missed some relevant page in the wiki. I'll search again, but I'd really appreciate if you could point me to something informative in this sense.

Frogmore42, 2018-07-22:
No button mode, but see this 
https://github.com/arendst/Sonoff-Tasmota/issues/1017

Frogmore42, 2018-07-22:
"Surely" doesn't have as clear a meaning as you might think.  If you had said, "I sent ... and it still behaved the same" it would have been clearer that you were actually able to send the command and it didn't do what you wanted.  The way you used "surely", I thought you meant "I can't believe" it is so simple.

Frogmore42, 2018-07-22:
Switches and buttons are similar, but not the same.  Buttons get multipess options.  Switches do not.  You can definitely a GPIO as either a button or a switch and attach either a button or a switch, but not all combinations are good ideas.

Frogmore42, 2018-07-22:
I would use the custom module type and define the buttons to be switches.

I would use switchmode 1 or 2 and switch topic 1 or 2.  That will give you most of what you need.  The rest will need to be handled by some logic probably in NodeRED.  You will get messages when you press and also when you release the button.

You might be able to do it with rules, but I have not taken the time to learn what they can do, since I normally use NodeRED and it has more than enough capability to do what I need.

FabrizioRomanoGenovese, 2018-07-22:
My bad, I am not a native speaker. :P What I meant is that the T1 changes `switchmode` successfully if I issue the command, but the behavior doesn't change. I inferred that, `switchmode1` doesn't refer to physical button1 as I thought (what you linked confirms this). 

The issue you referenced is very useful, and fundamentally says that what I want to do is not doable without fiddling with the module libraries. Problem with that is that I don't know how to flash said custom module via sonOTA atm :/

Moreover, if I declare buttons to be switches, I suppose that the buttons would basically end up being disconnected from the relays. Relay actuation would have to be managed by MQTT. This is bad because it entails that a network failure results in my whole house to be unusable, while as things stand now my T1 still behaves as a standard switch when net is down. :)

Anyway, I am surprised that stuff like `buttonmode` is not natively implemented. I supposed (maybe incorrectly) that fiddling with the features of physical buttons would be the main interest in using a device like the T1: I have three physical buttons and I want to use them to do whatever I want. Having an `hold` function on the physical buttons would actually give you 6 buttons (three buttons that can be operated in two different ways each) instead of 3, extending the device capabilities by an order of 2.

BTW: The toggle function is achievable for buttons even without using `buttonmode`. The idea is that when the button is used to drive a relay (hence with `pulsetime` set to, say, `2`), each time a button is pressed this is what happens:
- An `on` state gets sent to the broker
- `pulsetime 2` seconds or whatever pass
- An `off` state gets sent to the broker. 

If we write our rules such that the `off` state is simply ignored, then `on` can be used as toggle. The idea is that in an evironment like Node-red or openHAB the `off` messages on the button topic are ignored and the `on` ones are parsed as they were toggles. It's hacky and dirty (and I suppose you have to write a simple parsing function that transforms an `on` to be used as a `toggle`) but it works.

So, all in all, it's not totally bad. Pity for the "hold" mode on buttons tho, that would have been dope! :/

Frogmore42, 2018-07-22:
You can easily do the first two things you requested.  It is the third thing that complicates things.

You did not include any wiring diagrams, so exactly how you are connecting the T1s is not clear.  Normally a T1 controls the hot/live wire of the load.  You can send it a MQTT message to turn the load on or off.  You can have the T1 send a MQTT command whenever the button is pressed and released.  The T1 will always send a MQTT message when it switches the load.

The last thing is where you are trying to overload the definition of the behavior of the button.  There is a definition for a very long press, but that does a complete reset, which is not what you want.  Generally, people find multiple button presses confusing.  If you don't, you might be able to modify the software to do what you want.  That will not be easy.  You will need to become much more familiar with the existing code and will need to learn how to flash over serial.  Using a Wemos mini or NodeMCU will make it easy to experiment.  You might be able to do it, but it will require some investment of time and energy.

FabrizioRomanoGenovese, 2018-07-22:
So, I am using a ratchet relay: This is a relay that modifies its state according to a series of pulses of the same kind, as follows:
- First pulse: Turns on
- Second pulse: Turns off
- Third pulse: Turns on
- ...
All the pulses can be given on the same port. This is very useful when you want to control an appliance from many different physical points: You deploy n pushbuttons, each of them operating on a wire that brings the live to the relay port. If any of the pushbuttons is pressed, the relay toggles its state. This results in a relatively simple topology for an appliance that has to be controlled from many different points.

You can use sonoff T1s instead of the pushbuttons in the previous example by setting `pulsetime != 0`: Each time a button on any of the T1s connected to the ratchet relay is pressed, the T1 internal relay stays closed for an amount of time specified by `pulsetime`, then opens again. This simulates the behavior of the pushbuttons in the previous example.

In this scenario, the MQTT message telling me that the internal relay has went off is totally useless: I know it will since this is what `pulsetime` tells me. Similarly, if I have to control the ratchet relay remotely, I just have to send an `on` command to any of the T1s connected to it. Again, thanks to `pulsetime`, the internal relay of the T1 will open again automatically after the set time has passed.

I agree that the third thing is the hard one to do, since I want to overload a button with the properties of a switch. Again, this is the only difficult part: After this using `switchmode 7` on a button would give me the `hold` function by default. As I said, tho, I am surprised that an equivalent of `switchmode` has not been implemented for buttons by default, since it seems to me to be a very, very natural feature to have! I want to creatively control the physical buttons before thinking about how to creatively control the GPIO pins!

Frogmore42, 2018-07-22:
Buttons are already complex given the number of issues raised by people not understanding how they work.  They also have to support multipess, so I see why Theo didn't support button mode.

I didn't define the functionality or write the code for Tasmota.  I don't even claim to know it very well.  But, I have been using it for months and understand how to do most basic things with it and some more advanced things.  If you want to do something that Tasmota was not designed to do, the source is available to make it do whatever you can imagine and implement.  To do that will require a decent understanding of what/how it does things now.  It already has defined behavior for multi-press, perhaps you can repurpose the code to do what you want.

As far as Toggle goes, I find that to be okay if a human is controlling it but not really suitable for automation.  When I want to control something from multiple locations, I want to know that the command is going to work.  In a prior system I use some 415MHz switches that are controlled by an MCU.  The switches have defined RF commands for on and for off.  This makes it easy for me to send the command multiple times and be reasonably certain of the outcome.  If it only supported toggle, like some do, it would be much harder to deal with the lossy communication channel.

So, my point is that you need to look at the system as a whole and figure out where to put which part of the logic.  Sometimes by changing that, you can convert an impossible to solve problem into an easy or at least a solvable one.

If you want button mode, define your buttons as switches and you will get it.  But you still don't get the long press function.

If you are adept at programming, you should be able to adapt Theo's code to include long button press functionality.  If you do a good job on it, submit a pull request and it might even get added as an option to the base code.

Or, convince Theo this is a REALLY, REALLY, REALLY great idea and maybe he will implement it for you and everyone else who might want it in the future.

Frogmore42, 2018-07-22:
You can find the button handling code here:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff.ino

Look for ButtonHandler.

FabrizioRomanoGenovese, 2018-07-23:
`If you want button mode, define your buttons as switches and you will get it. But you still don't get the long press function.`

...As I said already, you do! If buttons are defined as switches you get `switchmode 5` which does exactly this. :)

ascillato, 2018-07-23:
Hi,

Have you read the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices ?

There is explained the HOLD feature and the TEXT change to make a MQTT Message with TOGGLE

Frogmore42, 2018-07-23:
It looks like everything you need is there.  All you need to do is configure the module type as generic and then set the pin definition as would be one for the T1, but make the buttons switches.  Then you can use switchmode and switchtopic to do what you want.  You will need something external to do something for hold, probably.

ascillato2, 2018-07-25:
Hi,

Have you managed to solve your issue?

FabrizioRomanoGenovese, 2018-07-25:
@ascillato, unfortunately I forgot my sonoff modules on the construction site, so I wasn't able to try it out. You can close it for now tho, I'll comment back if it turns out it didn't work :)

Thanks!

ascillato2, 2018-07-25:
Please, If you want to close your issue, do it yourself. If I close your issue, then you can not reopen it. If you close it, you can reopen it if you need something else.

FabrizioRomanoGenovese, 2018-08-04:
Hello all,

I can confirm that commands found on this page:
https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

effectively solve my issue. In particular:

- Point one of my desiderata is solved by fiddling with `pulsetime` makes buttons able to drive relays (they turn on and then, when the amount of time specified by `pulsetime` has passed, they turn off, simulating a pushbutton);

- Point two of my desiderata is solved just ignoring the `off` messages in MQTT, and considering only the `on` ones. See the discussion in this issue for more;

- Point three of my desiderata is solved by setting `buttontopic` to something and `option1` to `1`. This enable the `HOLD` command when a button is pressed and held for an amount of time specified by `option32`. Setting `option11` to `1` enables double press functionality. `option1` and `option11` can be enabled at the same time, enabling to use the same button in three different ways.

Thanks!