arendst/Tasmota #3300: Wemo+PIR motion Strange behavior

hwarang098tkd, 2018-07-22:
Hi all!!!
II connect wemo d1 with pir motion sensor. G->G,VCC->5v,Out->D5. I configure the module in generic mode and d5 to switch1. 
When I start the module and trigger for the first time the sensor with motion I noticed that it cant un-trigger when there isn't motion. I have a voltmeter connect to d5 pin and stays with no motion at 2,73 volt and when it is trigger in 3,2 V. It cant return in the un-trigger condition with 0v.
I tested with the same configuration with I switch and it is working fine. I also tested my pir motion and it is OK.
what is going wrong? I followed the instructions
[https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR501-PIR-Motion-Sensor](

reloxx13, 2018-07-22:
what is your Switchmode1 ?

should be `Switchmode1 1`

hwarang098tkd, 2018-07-22:
I did it. I changed anoder sensor and work fine. but it is functional the sensor I used to use.

closed 2018-07-22 by hwarang098tkd