arendst/Tasmota #3301: SwitchTopic for Switch5, 6, 7 not right

brainnex, 2018-07-22:
I configured 2 switches: switch4 and switch5.
Switch4 is sending the right MQTT message:
`15:15:36 MQT: cmnd/sonoff_test1/POWER4 = ON`

but Switch5, 6, 7 are sending by identical configuration to Switch4:
`15:15:36 MQT: domoticz/in = {"command":"switchlight","idx":0,"switchcmd":"On"}`

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : Sonoff RF
- [x] Provide the output of command ``status 0`` :
```
15:31:19 CMD: status 0
15:31:19 MQT: stat/sonoff_test1/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2","Sonoff3"],"Topic":"sonoff_test1","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":1}}
15:31:19 MQT: stat/sonoff_test1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:47:10","StartupUTC":"2018-07-22T13:44:09","Sleep":50,"BootCount":27,"SaveCount":58,"SaveAddress":"F6000"}}
15:31:19 MQT: stat/sonoff_test1/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-14T15:57:03","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:31:19 MQT: stat/sonoff_test1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Machete_braucht_kein_WLAN","indebuurt2"],"TelePeriod":300,"SetOption":["00000028","55818000"]}}
15:31:20 MQT: stat/sonoff_test1/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
15:31:20 MQT: stat/sonoff_test1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_test1-4717","IPAddress":"192.168.178.73","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.76","Mac":"2C:3A:E8:05:D2:6D","Webserver":2,"WifiConfig":5}}
15:31:20 MQT: stat/sonoff_test1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.76","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_05D26D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:31:20 MQT: stat/sonoff_test1/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 22 14:31:20 2018","Local":"Sun Jul 22 15:31:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:11","Sunset":"20:42"}}
15:31:20 MQT: stat/sonoff_test1/STATUS10 = {"StatusSNS":{"Time":"2018-07-22T15:31:20","Switch4":"ON","Switch5":"OFF","BME280":{"Temperature":27.9,"Humidity":49.4,"Pressure":1003.4},"BH1750":{"Illuminance":94},"TempUnit":"C"}}
15:31:20 MQT: stat/sonoff_test1/STATUS11 = {"StatusSTS":{"Time":"2018-07-22T15:31:20","Uptime":"0T00:47:11","Vcc":3.159,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Machete_braucht_kein_WLAN","RSSI":100,"APMac":"34:81:C4:25:31:3D"}}}
15:31:26 MQT: domoticz/in = {"command":"switchlight","idx":0,"switchcmd":"On"}
```

**To Reproduce**
Configure Switch4 and 5.
set: 
```
SwitchMode4 1
SwitchMode5 1
SwitchTopic 1
```
Toggle Switch4 and 5 an watch different output in console

**Expected behavior**
same output as switch4

tonyjobson, 2018-07-22:
Jumping on the back of this issue.
I also have the same / similar problem.

NodeMCU board. Tasmota 6.1.1c flashed.
I have the board in a housing with 8 push to make switches. I want to use this device to trigger HA actions.

I've set D1 to D4 as switch 1-4
I've set D5-D7 as switch 5-7
I've set D0 to switch 8.

I can see the MQTT messages coming through for every switch but every message has the same messages content.
stat/remote01/POWER OFF
stat/remote01/RESULT {"POWER":"OFF"}

Nowhere in my MQTT messages is it indicating which switch the message is about.

I've compared this to my Sonos Duals and this allows me to set an MQTT friendly name.
Is there anyway to set this for a generic device?

brainnex, 2018-07-27:
thanks for sharing your solution.

brainnex, 2018-08-23:
thanks @arendst for fixing this and the others for help

closed 2018-08-23 by brainnex