arendst/Tasmota #3302: ESP-01 TAsmota

ssinseeme, 2018-07-22:
I would like to get my ESP-01 flashed with Tasmota. I use IDE arduino to flash my NodeMCU and it works fine but not on ESP-01. 
I read in here that some of you was able to do that but the string was closed to post in, so I reopen this one. Hoping to find an answer. Thank you.

Frogmore42, 2018-07-22:
The esp-01 only has 512MB of flash.  The default size for Tasmota is 1MB.  You might be able to reduce the included code yo make it fit in 512MB, but you would also need to create a custom loader definition for it.  You would not be able to use OTA for sure.

Sounds like a LOT of work to support something that will never really work right.  Spend the extra money for one that has at least 1MB.  You can use the esp-01 for something else.

curzon01, 2018-07-22:
All current esp-01 do have 1 MB (also mine ordered 6 month ago) so it can be the issue but I don't think so.
 May be your USB converter used for esp-01 flashing does not provide enough power. On your NodeMCU it works because the RS232 interface is on board.

ascillato2, 2018-07-22:
Is the ESP-01, but Tasmota is aimed for sonoff devices. So for that, you should make some compilation adaptations.

curzon01, 2018-07-23:
If you use esptool.py you can readout the size of flash of your device 

`esptool.py --port /dev/ttyUSB0 flash_id`

or under windows

`esptool.py --port COM1 flash_id`

adapt ports to your environment

Examples:

**my NodeMCU**
```
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting...

```
**my ESP-01**
```
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Manufacturer: e0
Device: 4014
Detected flash size: 1MB
Hard resetting...

```

Unfortunately if you have an very old ESP-01 device with only 512kB flash, you have to adapt some compilation settings. But if your ESP-01 has already 1MB, flash is possible without any changes (all my six ESP-01 devices having Tasmota running unchanged).

ssinseeme, 2018-07-23:
@curzon01. Thanks for the information I was about to guive up. Yes I tried the esptool and the size is 1M. But yet it wouldn’t flash with tasmota using Arduino IDE. Could you please share the Tasmota file you are using and what are you using to flash with. I was looking for Tasmota.bin so I can try Easy flash. Let me know.

curzon01, 2018-07-23:
Don't give up, it's not the binary. Which USB adapter do you use for flashing?

Use one of the ready compiled binary from [Release page](https://github.com/arendst/Sonoff-Tasmota/releases) and it have to start, otherwise you have one of the following issues:
- the flash is damaged
- your USB adapter is garbage. There exists cheap adapters on the market which can not provide enough power during flash procedure and therefore the flash procedure fails.
- you exchanged RX and TX - try flashing exchange RX/TX line on ESP

Advantage of using esptool is an easy way to test if your USB adapter works and is connected in correct way. If you don't get results for esptool commands read_mac, flash_id your adapter is wrong or connection to ESP is wrong.

Try flash one of the release binary, let the USB adapter connected and observer what the serial line (115200 baud) displays after rebooting the device. The ESP boot loader and Tasmota are sending data out of the serial line when you start the device.

curzon01, 2018-07-23:
Sorry, forgot your question about flashing. I simply use the esptool.py too, e.g.

`esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin`

will do the job (use `--port COMx` on windows).

Jason2866, 2018-07-23:
Just my 2 cents. Flashing an ESP-01 as an beginner can be frustrating.
You have to take care for this: 
Is Chip Enable pulled up to high?
Is GPIO 2 pulled up to high (internal weak pull doesnt work always as it should)?
IS GPIO 0 pulled to GND?
Are the connections really reliable?
Is the USB adapter the "right choice" for this use case?
Is your PC setup correct?

Many possibilitys for failing.... Helping out of the distance isnt easy either

Jason2866, 2018-07-23:
I have seen cheap chinese USB adapters for flashing an ESP 01 which can and will never work without changes in design.

Simply wrong designed!!!

vMeph, 2018-07-23:
Flashed esp 01 with usb adapter using atom with out doing any changes, worked perfect

ssinseeme, 2018-07-23:
Thank again everybody for the replies. 
I don’t think the problem is the power or the  USB adapter I am using because I was able to flash a couple of things to test it. I flashed LED blink and 2 other MQTT IDE file I found in YouTube. 
When I flash Tasmota on ESP-01 with IDE “ that is what I used for my NodeMCU” it return with an error. I don’t remember now I am at work ( I will post the error later” 
The. I tried easy flash and it completed th flash but I could find the ESP-01 Tasmota IP address on the router or using the Tasmota scanner. 

I Believe the problem lies on the Tasmota file on ESP-01 should be different either size or content than the file I use to flash my Nod MCU. 

Finally, to confirm please provide me with the Tasmota file you used on ESP-01.  Either .bin or IDE. Thanks again.

Jason2866, 2018-07-23:
@ssinseeme 
I have a relais board with a ESP01 (1024 kbyte) and it works with the normal Tasmota version
which i use for every other of my devices. This is v.6.1.1b development (arduino core 2.3.0)
There is nothing special, just defined as generic and select Gpio2 as relais ....

curzon01, 2018-07-24:
@ssinseeme As I wrote above go to [Release page](https://github.com/arendst/Sonoff-Tasmota/releases) - here are the ready compiled binarys I used for the tests.

Try start using the previous [Tasmota 5.12.0](https://github.com/arendst/Sonoff-Tasmota/tree/v5.12.0) and if succeded try an up-to-date one.

To reprodue (I just did the steps below):

- Dowload this [sonoff.bin](https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff.bin) (5.12.0)

- Flash using commands (Linux):

```
esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
```

Then I restartet the ESP-01 and start monitor serial line using `screen /dev/ttyUSB0 115200,cs8` and got the following:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_6930ED, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-4333...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-4333...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 3 minutes
```

curzon01, 2018-07-24:
Just upgraded the same ESP-01 to current 6.1.1:

- download 6.1.1 [sonoff-minimal.bin](https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.1.1/sonoff-minimal.bin) and 6.1.1 [sonoff.bin](https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.1.1/sonoff.bin)
- goto web gui "Main Menu" / "Firmware Upgrade"
- upgrade to sonoff-minimal.bin from above
- after restart again goto web gui "Main Menu" / "Firmware Upgrade"
- upgrade to sonoff.bin from above

voilà:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_6930ED, GroupTopic sonoffs) Version 6.1.1-2_3_0
...

```

ascillato2, 2018-07-25:
Hi,

Have you managed to solve your issue?

ssinseeme, 2018-07-25:
No. I will keep trying. Tried curzon’s method. Having hard time may be due to my inexperience. Tried Atom for the first time it looked like it was flashing ok 100% but it stock in a loop where it was flashing again and again without final message being successful. 
Then I tried Atom with NodeMCU just to confirm and it didn’t work tried old version and new and it the same. Not sure yet. I will try again in the weekend. Thanks for the support

ssinseeme, 2018-07-25:
Just a quick question @Curzon. I used 5.12.0 Sonof.bin you provided, I used easy flash instead and it looks like it worked completing the flash successfully. But ! as a newbie since with the .bin file I didn’t configure my IP and MQTT. How do I connect to the Tasmota. I used the serial software ( I forget the name) to chthe check what in th ESP-01 and find the IP then I can configure the MQTT. But I couldn’t. 
So the question now it easy flash works. How do I find the IP to connect to the chip. Thanks.

curzon01, 2018-07-26:
When using binaries from release page use the powerful [Button functionality](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage).

You have several options to connect to your brand new flashed device, use one you prefer.

ssinseeme, 2018-07-28:
Just a quick update. A Little improvement but not success yet. I had sometime  to play with Sonoff today. I am more familiarize with Arduino IDE so I figured out I will keep using that since it works with my Node MCUs, I was able to figure out what was wrong when flashing ESP-01. the setting should be different when setting the IDE with ESP-01 I changed the board to generic of course that was done before but what was important compare to NodeMCU was IWIP variante has to be V1.4 prebuilt and flash size to 1M 64 SPIFFS. anyhow this completed and flashed but again the question is my Routed could not fine the IP address. I have 2 Node MCU showing but not anextra IP for ESP-01!!
I am powering the ESP-01 only with VCC and ground using USB, is this correct. or is there anything else should i do to figure out the IP.
@curzon01 I dont know what you mean with "(using it in Acess Point mode, WPS, Smartconfig App...)." could you elaborate please thanks again

Frogmore42, 2018-07-28:
You really should read the wiki again
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
You have the wrong flash size.  This will cause problems if you try and do an OTA update later.

In order to see what is going on, you have to get serial output.  There is an Arduino way of doing this and several other choices.  There are tutorials on the web for it to.  With the ability to see the serial output, you have to get lucky that everything is configured correctly and that it just works.  That has not been your experience so far, so you will need to get more knowledge about the tools you are using.  Using custom FW on esp8266 devices is not just plug and play.

https://www.instructables.com/id/HOW-TO-use-the-ARDUINO-SERIAL-MONITOR/

The ESP-01 devices are VERY tricky to get working correctly.  I wouldn't waste your time with them until you are very knowledgeable about their hardware and software requirements.  The hardware requirements for pull-ups pull-downs and decoupling are documented on various sites.  Some people consider it worth it to same a little money.  I consider it a waste of time to use them as it is really hard and frustrating to give them everything they require.  There is a reason the NodeMCU and Wemos boards are so popular.  For only a little bit more, you get something that is generally very easy to use.  You just plug it in to USB and then it generally works.

https://github.com/jandelgado/NodeMCU/wiki/ESP8266-01-overview-and-flashing-instructions

https://www.esp8266.com/viewtopic.php?f=13&t=2839

Frogmore42, 2018-07-28:
Should say without the a ability to see serial

pmknowles, 2018-07-29:
If you look at the Button usage page of the wiki you can ground GPIO0 quickly 5 times to put the ESP-01 into WPS mode so that it will connect to your router - need to put the router into WPS mode too. 
There are 2 versions of ESP-01 - ESP-01 (with 512k) and ESP-01S (with 1MB)

ssinseeme, 2018-07-31:
To all who contributed to this string thank you. I didn't give up, I think now i am a master .lol.

After reading and trying multiple things. what I figure was ( I hope this will help newbies) when flashing ESP-01 use GPIO 0 to ground, but as soon as it complete flashing reboot and dont forget to remove GPIO 0 from the ground this will allow ESP-01 to connect to WIFI.
well it took me a while but it now working.

ascillato, 2018-07-31:
Great to know that now it is working for you!

Please, close this issue. Thanks :+1:

closed 2018-08-18 by ascillato2