arendst/Tasmota #3305: Group Topic & Sonoff R2

smnino, 2018-07-23:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used : Sonoff Dual R2
- [ X] Provide the output of command ``status 0`` :
````
01:13:01 CMD: status 0
01:13:01 MQT: stat/kitchen_oh/STATUS = {"Status":{"Module":39,"FriendlyName":["K Light 1","K Light 2"],"Topic":"kitchen_oh","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":1}}
01:13:01 MQT: stat/kitchen_oh/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"kitchen","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:45:42","StartupUTC":"2018-07-22T22:27:19","Sleep":0,"BootCount":31,"SaveCount":885,"SaveAddress":"F6000"}}
01:13:01 MQT: stat/kitchen_oh/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-07-06T19:35:51","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
01:13:01 MQT: stat/kitchen_oh/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bigbrotherswatching_nomap",""],"TelePeriod":300,"SetOption":["000A80E9","55818000"]}}
01:13:01 MQT: stat/kitchen_oh/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":388,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
01:13:01 MQT: stat/kitchen_oh/STATUS5 = {"StatusNET":{"Hostname":"kitchen_oh","IPAddress":"10.0.1.14","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"60:01:94:9E:90:27","Webserver":2,"WifiConfig":2}}
01:13:01 MQT: stat/kitchen_oh/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.26","MqttPort":1883,"MqttClientMask":"kitchen_oh","MqttClient":"kitchen_oh","MqttUser":"DEVuser","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:13:01 MQT: stat/kitchen_oh/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 23 00:13:01 2018","Local":"Mon Jul 23 01:13:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:12","Sunset":"20:41"}}
01:13:01 MQT: stat/kitchen_oh/STATUS10 = {"StatusSNS":{"Time":"2018-07-23T01:13:01","Switch1":"OFF","Switch2":"OFF"}}
01:13:01 MQT: stat/kitchen_oh/STATUS11 = {"StatusSTS":{"Time":"2018-07-23T01:13:01","Uptime":"0T01:45:42","Vcc":3.411,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"WiFi","RSSI":88,"APMac":"00:00:00:00:00:00"}}}
````

Commands published to the set group topic of 'kitchen' are ignored by the second relay. So, if I publish "cmnd/kitchen/power OFF", only the first relay will respond. I'm sure I missed something along the way but, I assumed, any device with the group topic would respond. Apparently the sonoff dual behaves differently? 

Any help would be appreciated.

ascillato, 2018-07-23:
Hi,

For the second relay you have to send `cmnd/kitchen/power2 OFF`

The _grouptopic_ is for a group of sonoffs.

So, writing to a group of sonoffs, you are already specifying with POWER that is the POWER1. When ommiting the index, Tasmota uses 1.

So, with `cmnd/kitchen/power OFF` you are telling all your sonoffs subscribed to `/kitchen` to POWER OFF RELAY number 1

There is not a "POWER ALL OFF" MQTT message

But you can send to the grouptopic a backlog command (several commands at once)

`cmnd/kitchen/backlog power1 0; power2 0`

more information in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features

smnino, 2018-07-23:
Well, that's disappointing.....

I am using Node-Red to send commands. If I understand what you're saying and the quick testing I've done, I cannot specify the command as 'cmnd/kitchen/backlog power1; power2' and the payload of on or off from a separate node as everything after /backlog is the payload, correct?

ascillato, 2018-07-23:
You can use all `cmnd/kitchen/backlog power1 0; power2 0` as a command without payload.

and another `cmnd/kitchen/backlog power1 1; power2 1` to turn all on

ascillato, 2018-07-23:
or set a rule in TASMOTA to receive the MQTT messages:

```
cmnd/kitchen/event OFF
cmnd/kitchen/event ON
```

So, for the rule (type in the console):

```
rule 1
rule on event#OFF do backlog power1 0; power2 0 endon on event#ON do backlog power1 1; power2 1 endon
```

ascillato, 2018-07-23:
Hope this un-disappoint yourself :wink:

smnino, 2018-07-23:
Well, you've given me a lot to explore. I'm sure one solution or another will work. Thank you for taking the time to share your knowledge and help me out. 

However, it seems to me that both relays should respond to the group topic. I realize that is not the way the firmware is written but, seems logical that since both relays are subscribed to the topic that they should both respond.

ascillato, 2018-07-23:
Yes, both realys are subscribed to that topic (kitchen), but the command for each is different:

**POWER** ON is equal to **POWER1** ON    _( no index is equal to have index = 1 )_
**POWER2** ON is different to **POWER** ON

This is because you need to have different commands for different outputs.

Hope this helps.

Please, remember to close this issue when you think is done. Thanks.

smnino, 2018-07-23:
Makes perfect sense. I suppose. :-)

Thank you very much. I sincerely appreciate your time.

closed 2018-07-23 by smnino